MRPT/mrpt #770: Investigate wrong results for graph-slam optimizer (EDGE_SE2)

jlblancoc, 2018-05-18:
Problem: expected graph for the LevMarq optimizer in mrpt-graphslam doesn't match correct optimal (e.g. from G2O).

Test datasets are attached. Should be added as a new unit test.

[graphs.zip](https://github.com/MRPT/mrpt/files/2015052/graphs.zip)

jlblancoc, 2018-05-24:
Probably the toughest-to-fix and subtlest bug in a while... I've been struggling for 1 entire week 😠 
A bug that has been around for, probably, 8 years (!!). 

I'll have to re-apply this fix to all kind of graphslam optimizers and to bundle adjustment. 
It'll be fixed in both, master and mrpt-1.5. 

@bergercookie : After this, all your graphslam code should work **amazingly** better. Testing against G2O now gives exactly the same results (optimal graphs up to machine precission).

bergercookie, 2018-05-24:
> Probably the toughest-to-fix and subtlest bug in a while... I've been struggling for 1 entire week angry
A bug that has been around for, probably, 8 years (!!).

Alright!!! Quite impressive, gonna test it out!

So where was the actual bug?

https://github.com/MRPT/mrpt/commit/8786bd6f76076ce667b1808bd92efab9718f67c1#diff-b2e9b4a71fbc6b6406855b6a2f1684f2L435 ?

jlblancoc, 2018-05-25:
That one, together with the use of H_map as a variable OUTSIDE the while(iter) loop... so all Hessian blocks were actually **accumulating** over iterations 🤕 

Will summarize the changes in a clean PR anyway.

jlblancoc, 2018-05-27:
Reopening after https://github.com/MRPT/mrpt/commit/609e607a8f27ad6969d854896b000f32ae944449. 

It turned out more robust formulation for graphslam Jacobians was needed to make things really robust.

To-do: https://github.com/jlblancoc/tutorial-se3-manifold/issues/5

jlblancoc, 2018-06-27:
Status of this issue: 
* SE(2) stuff is done (pending documentation in tech. report)
* SE(3) still to be done: `jacobian_dDinvP1invP2_depsilon()`, etc. 

WIP is in branch `test-alternative-slam2d-jacobs` for mrpt 2.0, mrpt-1.5 itself for 1.5.

PR #904 (2019-03-18): New SE(3) & SE(2) jacobians
## Changed apps/libraries
* mrpt-poses
* mrpt-graphslam

## PR Description

* Implement more robust graphSLAM formulation for SE(2) and SE(3). See updated technical report https://github.com/jlblancoc/tutorial-se3-manifold/issues/5
* Refactor all Lie Algebra stuff into new namespace mrpt::poses::Lie
* Remove an old, now useless graphSLAM Jacobian.
* Many new unit tests.

Closes #770.
commit 21c4771d (2019-03-18): Merge pull request #904 from MRPT/new-se3-jacobs

* Implement more robust graphSLAM formulation for SE(2) and SE(3). See updated technical report https://github.com/jlblancoc/tutorial-se3-manifold/issues/5
* Refactor all Lie Algebra stuff into new namespace mrpt::poses::Lie
* Remove an old, now useless graphSLAM Jacobian.
* Many new unit tests.

Closes #770.
closed 2019-03-18 by jlblancoc