MRPT/mrpt #772: MinGW x64 building issue

JekaS178RUS, 2018-05-23:
Hello!

I got issue compiling mrpt wint MinGW64:

cmake .. -G "MSYS Makefiles" - ok
mingw32-make:
`$ make
Scanning dependencies of target EP_octomap
[  0%] Creating directories for 'EP_octomap'
[  0%] Performing download step (download, verify and extract) for 'EP_octomap'
-- Downloading...
   dst='D:/Programming/libs/mrpt-1.5.6/build/EP_octomap-prefix/src/devel.zip'
   timeout='none'
-- Using src='https://github.com/MRPT/octomap/archive/devel.zip'
-- Downloading... done
-- extracting...
     src='D:/Programming/libs/mrpt-1.5.6/build/EP_octomap-prefix/src/devel.zip'
     dst='D:/Programming/libs/mrpt-1.5.6/build/otherlibs/octomap'
-- extracting... [tar xfz]
-- extracting... [analysis]
-- extracting... [rename]
-- extracting... [clean up]
-- extracting... done
[  0%] No patch step for 'EP_octomap'
[  0%] No update step for 'EP_octomap'
[  1%] Performing configure step for 'EP_octomap'
-- The C compiler identification is GNU 7.2.0
-- The CXX compiler identification is GNU 7.2.0
-- Check for working C compiler: D:/Programming/tools/msys2/mingw64/bin/gcc.exe
-- Check for working C compiler: D:/Programming/tools/msys2/mingw64/bin/gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: D:/Programming/tools/msys2/mingw64/bin/g++.exe
-- Check for working CXX compiler: D:/Programming/tools/msys2/mingw64/bin/g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done

-- octomap building as Release
-- Found Doxygen: C:/Program Files/doxygen/bin/doxygen.exe (found version "1.8.10") found components:  doxygen missing components:  dot

-- Compile octomap using: make
-- Install octomap using: make install
--     (be sure to set the correct CMAKE_INSTALL_PREFIX before)
-- Compile API-documentation using: make docs

-- Configuring done
-- Generating done
-- Build files have been written to: D:/Programming/libs/mrpt-1.5.6/build/EP_octomap-prefix/src/EP_octomap-build
[  1%] Performing build step for 'EP_octomap'
Scanning dependencies of target octomath-static
[  7%] Building CXX object octomap/src/math/CMakeFiles/octomath-static.dir/Vector3.cpp.obj
[ 14%] Building CXX object octomap/src/math/CMakeFiles/octomath-static.dir/Quaternion.cpp.obj
[ 21%] Building CXX object octomap/src/math/CMakeFiles/octomath-static.dir/Pose6D.cpp.obj
[ 28%] Linking CXX static library /D/Programming/libs/mrpt-1.5.6/build/otherlibs/octomap/lib/liboctomath.a
[ 28%] Built target octomath-static
Scanning dependencies of target octomap-static
[ 35%] Building CXX object octomap/src/CMakeFiles/octomap-static.dir/AbstractOcTree.cpp.obj
[ 42%] Building CXX object octomap/src/CMakeFiles/octomap-static.dir/AbstractOccupancyOcTree.cpp.obj
[ 50%] Building CXX object octomap/src/CMakeFiles/octomap-static.dir/Pointcloud.cpp.obj
[ 57%] Building CXX object octomap/src/CMakeFiles/octomap-static.dir/ScanGraph.cpp.obj
[ 64%] Building CXX object octomap/src/CMakeFiles/octomap-static.dir/CountingOcTree.cpp.obj
[ 71%] Building CXX object octomap/src/CMakeFiles/octomap-static.dir/OcTree.cpp.obj
[ 78%] Building CXX object octomap/src/CMakeFiles/octomap-static.dir/OcTreeNode.cpp.obj
[ 85%] Building CXX object octomap/src/CMakeFiles/octomap-static.dir/OcTreeStamped.cpp.obj
[ 92%] Building CXX object octomap/src/CMakeFiles/octomap-static.dir/ColorOcTree.cpp.obj
[100%] Linking CXX static library /D/Programming/libs/mrpt-1.5.6/build/otherlibs/octomap/lib/liboctomap.a
[100%] Built target octomap-static
[  1%] No install step for 'EP_octomap'
[  1%] No test step for 'EP_octomap'
[  1%] Completed 'EP_octomap'
[  1%] Built target EP_octomap
Scanning dependencies of target DocumentationFiles
[  1%] Built target DocumentationFiles
[  1%] Generating C unity source libs/base/cotire/mrpt-base_C_unity.c
[  2%] Generating CXX prefix source libs/base/cotire/mrpt-base_CXX_prefix.cxx
[  2%] Generating CXX prefix header libs/base/cotire/mrpt-base_CXX_prefix.hxx
[  2%] Building CXX precompiled header libs/base/cotire/mrpt-base_CXX_prefix.hxx.gch
[  2%] Generating C prefix source libs/base/cotire/mrpt-base_C_prefix.c
[  2%] Generating C prefix header libs/base/cotire/mrpt-base_C_prefix.h
[  2%] Building C precompiled header libs/base/cotire/mrpt-base_C_prefix.h.gch
Scanning dependencies of target mrpt-base
[  2%] Building CXX object libs/base/CMakeFiles/mrpt-base.dir/src/bayes/CParticleFilter.cpp.obj
[  2%] Building CXX object libs/base/CMakeFiles/mrpt-base.dir/src/bayes/CParticleFilterCapable.cpp.obj
[  2%] Building CXX object libs/base/CMakeFiles/mrpt-base.dir/src/compress/zip.cpp.obj
[  2%] Building CXX object libs/base/CMakeFiles/mrpt-base.dir/src/math/CAtan2LookUpTable.cpp.obj
[  2%] Building CXX object libs/base/CMakeFiles/mrpt-base.dir/src/math/CHistogram.cpp.obj
[  2%] Building CXX object libs/base/CMakeFiles/mrpt-base.dir/src/math/CMatrix.cpp.obj
[  2%] Building CXX object libs/base/CMakeFiles/mrpt-base.dir/src/math/CMatrixB.cpp.obj
[  2%] Building CXX object libs/base/CMakeFiles/mrpt-base.dir/src/math/CMatrixD.cpp.obj
[  3%] Building CXX object libs/base/CMakeFiles/mrpt-base.dir/src/math/CPolygon.cpp.obj
[  3%] Building CXX object libs/base/CMakeFiles/mrpt-base.dir/src/math/CRuntimeCompiledExpression.cpp.obj
D:/Programming/tools/msys2/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.2.0/../../../../x86_64-w64-mingw32/bin/as.exe: CMakeFiles/mrpt-base.dir/src/math/CRuntimeCompiledExpression.cpp.obj: too many sections (49113)
D:\Programming\tools\msys2\tmp\ccPuqE9n.s: Assembler messages:
D:\Programming\tools\msys2\tmp\ccPuqE9n.s: Fatal error: can't write 19 bytes to section .text of CMakeFiles/mrpt-base.dir/src/math/CRuntimeCompiledExpression.cpp.obj because: 'File too big'
D:/Programming/tools/msys2/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.2.0/../../../../x86_64-w64-mingw32/bin/as.exe: CMakeFiles/mrpt-base.dir/src/math/CRuntimeCompiledExpression.cpp.obj: too many sections (49113)
D:\Programming\tools\msys2\tmp\ccPuqE9n.s: Fatal error: can't close CMakeFiles/mrpt-base.dir/src/math/CRuntimeCompiledExpression.cpp.obj: File too big
make[2]: *** [libs/base/CMakeFiles/mrpt-base.dir/build.make:325: libs/base/CMakeFiles/mrpt-base.dir/src/math/CRuntimeCompiledExpression.cpp.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:332: libs/base/CMakeFiles/mrpt-base.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
`

jolting, 2018-05-24:
I honestly have never compiled using mingw
I think the gcc variant of this code might work for mingw.
https://github.com/MRPT/mrpt/blob/235cb25ad01f977cefd57e519e8cba8765b12a3c/libs/base/src/synch/atomic_incr.cpp#L90

@jlblancoc 
https://github.com/MRPT/mrpt/blob/235cb25ad01f977cefd57e519e8cba8765b12a3c/libs/base/src/synch/atomic_incr.cpp#L44
P.S. This needs a read barrier. I think the windows intrinsic is called `_ReadBarrier`

I'm so glad we're using `std::atomic` now.

jlblancoc, 2018-05-28:
This should be solved after https://github.com/MRPT/mrpt/commit/863c4ba2ba2915662f0294b08fd125b1549d290c

If not (can't test directly with MinGW), let us know and re-open.

JekaS178RUS, 2018-05-29:
Problem with atomic_increment solved, but got new troubles with glew o_0 which is strange because I used it many times.

[log.txt](https://github.com/MRPT/mrpt/files/2048239/log.txt)

jlblancoc, 2018-05-30:
On efficiency... I've observed **huge** gains in speed in GCC (under Linux) vs. MSVC binaries in the past. 
I used to publish some performance charts comparing precisely this point. 
For example, this one: http://mrpt.ual.es/reference/1.4.0/perf-html/comparison_1.2.0dev-GCC-64bit_vs_1.2.0dev-MSVC11-64bit.html

More recent versions of the documentation doesn't include these tables (it's a long story... basically it was inconvenient to build them as part of the .deb generation process, etc.). It would be a good idea to generate them in a separate subdomain under mrpt.org for more modern versions of MRPT.... when time permits ;-)

Anyhow: yes, GCC normally *used* to generate code much faster then MSVC, but it was years ago, it should be re-evaluated after the huge progress made by MSVC in recent versions.

JekaS178RUS, 2018-05-30:
@jolting Didn't see any options for doing this...

JekaS178RUS, 2018-05-31:
Yes you are right there is no such thing in Windows, but mingw is something in between, so it can has "system" glut and glew. For example: in my case using MSYS2 pacman package manager I install glew and can use it without adding any additional include directories and so on.

Unfortunatly I didn't have much experience with CMAKE, so I couldn't help you much with a PR, sorry.
What I think you should try (pseudocode):

    IF WIN32 
        IF CMAKE_CXX_COMPILER_ID == GNU    // Or any better way for detecting mingw/msys environment
            use_system_glew()
        ELSE
            use_embedded_glew()
        ENDIF
    ENDIF

jlblancoc, 2018-05-31:
I see... 

However, I followed the involved headers and couldn't find a reason of this error... there must be something I miss. 

Just in case, I have updated the version of glew to the latest release (7 years newer...). 
Could you give it a try to see if it somehow fixes the problem? It's in the branch `newer-glew`. 
Commit: https://github.com/MRPT/mrpt/commit/188ef763b3a2fcc4f852bbc3f4859178ee452949

It may also help to uninstall the glew package with pacman, if you don't need it for any other project...

jolting, 2018-05-31:
There must be some sort of weird MinGW implicit precompiled header or `#define` leaking into the compiler. I don't understand it myself.

jlblancoc, 2018-06-19:
The commits that were intended to fix the "atomic" issue:
* 863c4ba2ba2915662f0294b08fd125b1549d290c
* and today's attempt to fix it with: a9337660e0427e365ecbd9d5b691a77f9ae00eca

caused a **serious regression**, in the form of thousands of memory leaks. E.g:

```
Direct leak of 16 byte(s) in 1 object(s) allocated from:
    #0 0x7f36f289e532 in operator new(unsigned long) (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x99532)
    #1 0x7f36ee79bed1 in stlplus::smart_ptr_base<mrpt::utils::CObject, stlplus::clone_copy<mrpt::utils::CObject>, mrpt::synch::CAtomicCounter>::smart_ptr_base(mrpt::utils::CObject*) (/usr/lib/x86_64-linux-gnu/libmrpt-nav.so.1.5+0x4ced1)
    #2 0x60200022ffaf  (<unknown module>) 
```

I'm trying to see what's wrong here, without luck for now. A missing implementation of an = operator, move operators,etc.??

jlblancoc, 2018-06-19:
For now, I'm just undoing 863c4ba2ba2915662f0294b08fd125b1549d290c even if that means we break builds in MinGW64... the leaks are a much major issue!

jlblancoc, 2018-06-25:
This should be fixed now after #788 . Please, reopen if the issue subsists.

commit 863c4ba2 (2018-05-28): atomic_incr now based on std::atomic_long

We can safely do this for mrpt-1.5 too, since old versions of compilers support this C++11 type and it's not exposed to the user API (.h).

Closes #772
PR #788 (2018-06-25): Simplify and robustify mrpt-1.5 by using minimal subset of C++11
(W.I.P. / Request for comments)

The idea is to get rid of:
* Custom thread implementation.
* stlplus for smart pointers
* custom atomic counters

The need to enable C++11 would be propagated to user CMake scripts via `MRPTConfig.cmake`.
While some users may find the C++17 requirement for mrpt-master a bit too "on the edge", I think C++11 is now old enough to be imposed without problems.

I think it's about time for this, thinking of the next release mrpt-1.5.7. It could make sense to rename it mrpt-1.6 but it would break so many [... cut]
commit 714a58b5 (2018-06-25): Merge branch 'mrpt-1.5-cpp11' into mrpt-1.5

Closes #772
closed 2018-06-25 by jlblancoc