MRPT/mrpt #786: CDisplayWindow3D crash, using windows binaries.

antithing, 2018-06-17:
I am linking the windows binaries in a c++ project, visual studio 2015, 64 bit.
When I run this code, to create a window:

//.h
```
#include <mrpt/opengl.h>
#include <mrpt/opengl/CPointCloudColoured.h>
#include <mrpt/gui.h>
#include <mrpt/utils/CConfigFile.h>
#include <mrpt/utils/CConfigFileBase.h>
using namespace mrpt;
using namespace mrpt::gui;
using namespace mrpt::poses;
using namespace mrpt::utils;
using namespace mrpt::math;
using namespace mrpt::opengl;

```

 CDisplayWindow3D*           win;

//.cpp
`win  = new CDisplayWindow3D("Scene",1920,1080);`

I get a crash, with this:

`Exception thrown at 0x00007FFA6F012356 (libmrpt-gui156_msvc140_x64.dll) in dataset.exe: 0xC0000005: Access violation reading location 0x000000000000AE18.`

I have these include paths:
```

C:\Program Files\mrpt-1.5.6\libs\hwdrivers\include
C:\Program Files\mrpt-1.5.6\libs\graphs\include
C:\Program Files\mrpt-1.5.6\libs\vision\include
C:\Program Files\mrpt-1.5.6\libs\obs\include
C:\Program Files\mrpt-1.5.6\libs\maps\include
C:\Program Files\mrpt-1.5.6\libs\gui\include
C:\Program Files\mrpt-1.5.6\libs\opengl\include
C:\Program Files\mrpt-1.5.6\libs\base\include
C:\Program Files\mrpt-1.5.6\include\mrpt\mrpt-config
```

and these libs linked:
```

libmrpt-base156_msvc140_x64.lib
libmrpt-opengl156_msvc140_x64.lib
libmrpt-gui156_msvc140_x64.lib
libmrpt-maps156_msvc140_x64.lib
libmrpt-obs156_msvc140_x64.lib
libmrpt-vision156_msvc140_x64.lib
libmrpt-graphs156_msvc140_x64.lib
libmrpt-hwdrivers156_msvc140_x64.lib
```

Why is this crashing? What am I doing wrong here?

Thank you.

jlblancoc, 2018-06-20:
Closing this one, let's continue discussing on SO https://stackoverflow.com/questions/50915476/mrpt-cdisplaywindow3d-crash-using-windows-binaries

closed 2018-06-20 by jlblancoc

XiongWei1012, 2020-03-10:
I have the same problem.  Could you please tell me how to solve it?