MRPT/mrpt #796: Error in RawlogViewer Code

rachit173, 2018-07-05:
[reproduction of error](https://github.com/rachit173/bug-mrpt-web)
However it gives a different error message.
I tried it with the following rawlog,
mrpt/datasets/localization_demo.rawlog
It gave error on index 2, 6 etc of the rawlog, those indices were associated with CActionRobotMovement2D.
Check for the following code in rawlog-viewer/main_display_processor.cpp
```cpp
if (classID == CLASS_ID(CActionRobotMovement2D))
{...
}

```

jlblancoc, 2018-07-07:
What version of mrpt did you build this against?

rachit173, 2018-07-08:
```
commit 6a4569c348d13acd16495bef560d96e992cc6366 (HEAD -> master, origin/master)
Author: rachit173 <rachittibrewal@outlook.com>
Date:   Fri Jun 8 16:38:55 2018 +0530

    PIMPL idiom for CSchemeArchiveBase + unittest
    
    +mocking class

commit 82a4c747f78a1cbac8a3f80b5d1a9e2900d3f9a7
Author: rachit173 <rachittibrewal@outlook.com>
Date:   Tue Jun 5 11:48:43 2018 +0530

    add bool,(u)int**_t explicit operator overload

commit 68667ab2bc819c89d60b45050656f84568a24a2f
Author: rachit173 <rachittibrewal@outlook.com>
Date:   Tue Jun 5 11:08:57 2018 +0530

    type conversion explicit overloading in base class+sample serializations

commit e55347e5b0eb070dbf00ab670e635be7396f2e33
Author: rachit173 <rachittibrewal@outlook.com>
Date:   Sun Jun 3 13:33:16 2018 +0530

    json serial. using CSchemeArchiveBase wo pimpl
    
    +CPoint2D example

commit 3be29d52c36a613b0ff2c9853088f1d2b8f0b5ab
Author: rachit173 <rachittibrewal@outlook.com>
Date:   Sat Jun 2 09:45:17 2018 +0530

    add CSchemeArchiveBase virtual methods

commit bde58bd1309b5c5cf3e7a46ae65983ce529a83e4
Author: jgmonroy <javier.gmonroy@gmail.com>
Date:   Tue May 29 11:00:49 2018 +0200

    Including missing <mutex> header

commit 8cbec87cbf75985e172109cc3d034d3af359a8e0
Author: Hunter Laux <hunterlaux@gmail.com>
Date:   Sun May 27 23:28:03 2018 -0700

    Fix build stage variant suffix
    
    Underscores are part of the variable name.
    Use dashes instead.

commit e417d538c4305c1740be337b03938b4c5a8ed4e6
Author: Hunter Laux <hunterlaux@gmail.com>
Date:   Sun May 27 21:36:04 2018 -0700

    Unique stage image names in travis_docker.sh
```
I am building against my own fork, earlier it was giving no error after I commented some lines(drawManySamples). But now I am getting this error, which can also be found in the error reproduction above.
```
rawlog-viewer: /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:408: Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator[](Eigen::Index) [with Derived = Eigen::Matrix<double, -1, 1>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = double; Eigen::Index = long int]: Assertion `index >= 0 && index < size()' failed.
Aborted (core dumped)
```

jlblancoc, 2018-07-16:
My first attempt to reproduce the crash under MSVC (Windows): It works. 

Well, in fact it crashes at first, but only because you commented out the line:

```
      act->poseChange->drawManySamples(N, samples);
```
which then leaves the `      vector<CVectorDouble> samples(N);` container uninitialized, so accessing their contents afterwards leads to an assert. 

Commenting out that line, all worked...

rachit173, 2018-07-18:
I tried uncommenting the line and running code, I have pushed it to the [repo](https://github.com/rachit173/bug-mrpt-web). Could you pull the repo again and try to run it with the rawlog /usr/share/mrpt/datasets/localization_demo.rawlog. I am getting this error, 
```
darthvaddr@darthvaddr-Lenovo-ideapad-Y700-15ISK:~/Desktop/bug/build/rawlog-viewer$ ./rawlog-viewer /usr/share/mrpt/datasets/localization_demo.rawlog
0
1
2
CActionRobotMovement2D
double free or corruption (out)Aborted (core dumped)
```
I am using Ubuntu 18.04 and built the repo using the latest mrpt commit.

jlblancoc, 2018-07-31:
Any news on this?

rachit173, 2018-07-31:
Tried gdb backtrace on the executable, built on latest commit from mrpt master branch
```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff4eb8801 in __GI_abort () at abort.c:79
#2  0x00007ffff4f01897 in __libc_message (action=action@entry=do_abort, 
    fmt=fmt@entry=0x7ffff502eb9a "%s\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff4f0890a in malloc_printerr (
    str=str@entry=0x7ffff5030870 "double free or corruption (out)") at malloc.c:5350
#4  0x00007ffff4f0fe75 in _int_free (have_lock=0, p=0x555555d8cdb0, av=0x7ffff5263c40 <main_arena>)
    at malloc.c:4278
#5  __GI___libc_free (mem=0x555555d8cdc0) at malloc.c:3124
#6  0x0000555555567e91 in Eigen::internal::aligned_free(void*) ()
#7  0x000055555556fa64 in void Eigen::internal::conditional_aligned_free<true>(void*) ()
#8  0x000055555556e08c in void Eigen::internal::conditional_aligned_delete_auto<double, true>(double*, unsigned long) ()
#9  0x0000555555576c1d in Eigen::DenseStorage<double, -1, -1, 1, 0>::~DenseStorage() ()
#10 0x0000555555575786 in Eigen::PlainObjectBase<Eigen::Matrix<double, -1, 1, 0, -1, 1> >::~PlainObjectBase() ()
#11 0x00005555555757a2 in Eigen::Matrix<double, -1, 1, 0, -1, 1>::~Matrix() ()
#12 0x00005555555757be in mrpt::math::dynamic_vector<double>::~dynamic_vector() ()
#13 0x00005555555757d9 in void std::_Destroy<mrpt::math::dynamic_vector<double> >(mrpt::math::dynamic_vector<double>*) ()
#14 0x0000555555573f18 in void std::_Destroy_aux<false>::__destroy<mrpt::math::dynamic_vector<double>*>(mrpt::math::dynamic_vector<double>*, mrpt::math::dynamic_vector<double>*) ()
#15 0x0000555555571c1d in void std::_Destroy<mrpt::math::dynamic_vector<double>*>(mrpt::math::dynamic_vector<double>*, mrpt::math::dynamic_vector<double>*) ()
---Type <return> to continue, or q <return> to quit---
#16 0x000055555556e62b in void std::_Destroy<mrpt::math::dynamic_vector<double>*, mrpt::math::dynamic_vector<double> >(mrpt::math::dynamic_vector<double>*, mrpt::math::dynamic_vector<double>*, std::allocator<mrpt::math::dynamic_vector<double> >&) ()
#17 0x000055555556a895 in std::vector<mrpt::math::dynamic_vector<double>, std::allocator<mrpt::math::dynamic_vector<double> > >::~vector() ()
#18 0x00005555555667ca in mrpt::webapps::mainDisplayProcessor(std::shared_ptr<mrpt::serialization::CSerializable> const&, mrpt::obs::CRawlog const*) ()
#19 0x0000555555563094 in CRawlogTreeProcessor::getTreeDataPoint(unsigned long) const ()
#20 0x000055555555d3ce in main ()
```

jlblancoc, 2018-07-31:
Hmmm... a guess: please check the gcc optimization flags used for: 
- mrpt itself
- your app

As you may know, they can be seen by building with `VERBOSE=1 make -j1` instead of `make`. 
Look for differences like the presence/absence of `-mtune=native`, `-O3` vs other optimization levels, etc.

Also, it's important, try with `valgrind` to see what it finds!

rachit173, 2018-08-01:
Did you use windows or linux for last build?
http://eigen.tuxfamily.org/bz/show_bug.cgi?id=554#c11
I think the above link indicates our problem, trying to find a solution there. Also installed the latest eigen version from git repo.
Currently trying different optimzation levels,
Valgrind gives this
```
==20796== Memcheck, a memory error detector
==20796== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==20796== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==20796== Command: ./rawlog-viewer /usr/share/mrpt/datasets/localization_demo.rawlog
==20796== 
0
1
2
CActionRobotMovement2D
==20796== Invalid free() / delete / delete[] / realloc()
==20796==    at 0x4C30D3B: free (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==20796==    by 0x11C082: Eigen::internal::aligned_free(void*) (in /home/darthvaddr/Desktop/bug/build/rawlog-viewer/rawlog-viewer)
==20796==    by 0x123C3E: void Eigen::internal::conditional_aligned_free<true>(void*) (in /home/darthvaddr/Desktop/bug/build/rawlog-viewer/rawlog-viewer)
==20796==    by 0x12222B: void Eigen::internal::conditional_aligned_delete_auto<double, true>(double*, unsigned long) (in /home/darthvaddr/Desktop/bug/build/rawlog-viewer/rawlog-viewer)
==20796==    by 0x12AE38: Eigen::DenseStorage<double, -1, -1, 1, 0>::~DenseStorage() (in /home/darthvaddr/Desktop/bug/build/rawlog-viewer/rawlog-viewer)
==20796==    by 0x12991D: Eigen::PlainObjectBase<Eigen::Matrix<double, -1, 1, 0, -1, 1> >::~PlainObjectBase() (in /home/darthvaddr/Desktop/bug/build/rawlog-viewer/rawlog-viewer)
==20796==    by 0x129939: Eigen::Matrix<double, -1, 1, 0, -1, 1>::~Matrix() (in /home/darthvaddr/Desktop/bug/build/rawlog-viewer/rawlog-viewer)
==20796==    by 0x129955: mrpt::math::dynamic_vector<double>::~dynamic_vector() (in /home/darthvaddr/Desktop/bug/build/rawlog-viewer/rawlog-viewer)
==20796==    by 0x129970: void std::_Destroy<mrpt::math::dynamic_vector<double> >(mrpt::math::dynamic_vector<double>*) (in /home/darthvaddr/Desktop/bug/build/rawlog-viewer/rawlog-viewer)
==20796==    by 0x128043: void std::_Destroy_aux<false>::__destroy<mrpt::math::dynamic_vector<double>*>(mrpt::math::dynamic_vector<double>*, mrpt::math::dynamic_vector<double>*) (in /home/darthvaddr/Desktop/bug/build/rawlog-viewer/rawlog-viewer)
==20796==    by 0x125DF6: void std::_Destroy<mrpt::math::dynamic_vector<double>*>(mrpt::math::dynamic_vector<double>*, mrpt::math::dynamic_vector<double>*) (in /home/darthvaddr/Desktop/bug/build/rawlog-viewer/rawlog-viewer)
==20796==    by 0x1227CA: void std::_Destroy<mrpt::math::dynamic_vector<double>*, mrpt::math::dynamic_vector<double> >(mrpt::math::dynamic_vector<double>*, mrpt::math::dynamic_vector<double>*, std::allocator<mrpt::math::dynamic_vector<double> >&) (in /home/darthvaddr/Desktop/bug/build/rawlog-viewer/rawlog-viewer)
==20796==  Address 0x2d46bde0 is 32 bytes inside a block of size 56 alloc'd
==20796==    at 0x4C2FB0F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==20796==    by 0x5153982: mrpt::poses::CPosePDFGaussian::drawManySamples(unsigned long, std::vector<mrpt::math::dynamic_vector<double>, std::allocator<mrpt::math::dynamic_vector<double> > >&) const (in /home/darthvaddr/mrpt/build/lib/libmrpt-poses.so.1.9.9)
==20796==    by 0x11A7D4: mrpt::webapps::mainDisplayProcessor(std::shared_ptr<mrpt::serialization::CSerializable> const&, mrpt::obs::CRawlog const*) (in /home/darthvaddr/Desktop/bug/build/rawlog-viewer/rawlog-viewer)
==20796==    by 0x1171C3: CRawlogTreeProcessor::getTreeDataPoint(unsigned long) const (in /home/darthvaddr/Desktop/bug/build/rawlog-viewer/rawlog-viewer)
==20796==    by 0x1113CD: main (in /home/darthvaddr/Desktop/bug/build/rawlog-viewer/rawlog-viewer)
==20796== 
3
4
5
6
CActionRobotMovement2D
7
8
9
10
CActionRobotMovement2D
11
12
13
14
CActionRobotMovement2D
15
16
17
18
CActionRobotMovement2D
19
20
21
22
CActionRobotMovement2D
23
24
25
26
CActionRobotMovement2D
27
28
29
30
CActionRobotMovement2D
31
32
33
34
CActionRobotMovement2D
35
36
37
38
CActionRobotMovement2D
39
40
41
42
CActionRobotMovement2D
43
44
45
46
CActionRobotMovement2D
47
48
49
50
CActionRobotMovement2D
51
52
53
54
CActionRobotMovement2D
55
56
57
58
CActionRobotMovement2D
59
60
61
62
CActionRobotMovement2D
63
64
65
66
CActionRobotMovement2D
67
68
69
70
CActionRobotMovement2D
71
72
73
==20796== 
==20796== HEAP SUMMARY:
==20796==     in use at exit: 1,064,632 bytes in 18,811 blocks
==20796==   total heap usage: 163,731 allocs, 162,920 frees, 53,139,987 bytes allocated
==20796== 
==20796== LEAK SUMMARY:
==20796==    definitely lost: 1,008,000 bytes in 18,000 blocks
==20796==    indirectly lost: 0 bytes in 0 blocks
==20796==      possibly lost: 0 bytes in 0 blocks
==20796==    still reachable: 56,632 bytes in 811 blocks
==20796==         suppressed: 0 bytes in 0 blocks
==20796== Rerun with --leak-check=full to see details of leaked memory
==20796== 
==20796== For counts of detected and suppressed errors, rerun with: -v
==20796== ERROR SUMMARY: 18000 errors from 1 contexts (suppressed: 0 from 0)

```

jlblancoc, 2018-08-01:
> Did you use windows or linux for last build?

Linux, Ubuntu 16.04.  Thanks for the tests!

jlblancoc, 2018-08-15:
I think we can close this one, although we never found the actual cause of your problem, right?

closed 2018-08-15 by jlblancoc