MRPT/mrpt #799: /usr/bin/ld: cannot find -llas_c

wendaoyuchen, 2018-07-16:
Hi, when I make this project, this problem came out , what should I do to slove it ?

jolting, 2018-07-16:
Which version of Linux are you running?
Are you using the MRPT version from the PPA?

wendaoyuchen, 2018-07-17:
@jolting ,Thanks for your reply !
My Linux version is Ubuntu 16.04 LTS.
I need to install the version of MRPT([https://github.com/MRPT/mrpt/tree/0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3]), so I download the source from the link above, and
 `cd mrpt`
`mkdir build`
`cd build`
`cmake ..`
`make`
and then the error message is shown in the terminal as below:
`/usr/bin/ld: cannot find -llas_c
/usr/bin/ld: cannot find -llas_c
/usr/bin/ld: cannot find -llas_c
collect2: error: ld returned 1 exit status
libs/maps/CMakeFiles/mrpt-maps.dir/build.make:1400: recipe for target 'lib/libmrpt-maps.so.1.5.0' failed
make[2]: *** [lib/libmrpt-maps.so.1.5.0] Error 1
CMakeFiles/Makefile2:323: recipe for target 'libs/maps/CMakeFiles/mrpt-maps.dir/all' failed
make[1]: *** [libs/maps/CMakeFiles/mrpt-maps.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
`
I also install MRPT from PPA, but it dosen't work.

jolting, 2018-07-17:
Looks like you have liblas-dev installed but not liblas-c-dev.
`sudo apt install liblas-c-dev`

wendaoyuchen, 2018-07-17:
Thank you very much !
I tried liblas-dev and liblas_c-dev yesterday, I should have tried this.
Thank you again !

closed 2018-07-17 by jolting