MRPT/mrpt #803: MRPT runs on MacOS10.14

alibenD, 2018-07-17:
Emmm... I have tried to build mrpt on MacOS10.14, but some requirements can not be fullfilled.

Is there any solutions for running MRPT on MacOS? Or does anyone try to fix it? I would like to join this fixed work.

jolting, 2018-07-17:
Please post a log of what is missing.

alibenD, 2018-07-20:
Sorry for my late reply.

There is the error info below.
```
/Users/aliben/project/source_code/mrpt/libs/core/src/aligned_malloc.cpp: In function 'void* mrpt::aligned_malloc(size_t, size_t)':
/Users/aliben/project/source_code/mrpt/libs/core/src/aligned_malloc.cpp:28:11: error: '::aligned_alloc' has not been declared
  return ::aligned_alloc(alignment, size);
           ^~~~~~~~~~~~~
make[2]: *** [libs/core/CMakeFiles/mrpt-core.dir/src/aligned_malloc.cpp.o] Error 1
make[1]: *** [libs/core/CMakeFiles/mrpt-core.dir/all] Error 2
make: *** [all] Error 2
```

alibenD, 2018-07-20:
I have tried to use the default compiler(Apple provided) and gnu-gcc-8, but the problem is the same.

There is the error info using complier Apple one.

```
Scanning dependencies of target DocumentationFiles
[  0%] Built target DocumentationFiles
Scanning dependencies of target mrpt-random
[  0%] Building CXX object libs/random/CMakeFiles/mrpt-random.dir/src/RandomGenerator.cpp.o
[  0%] Building CXX object libs/random/CMakeFiles/mrpt-random.dir/src/random-precomp.cpp.o
[  0%] Linking CXX shared library ../../lib/libmrpt-random.dylib
[  0%] Built target mrpt-random
Scanning dependencies of target mrpt-core
[  0%] Building CXX object libs/core/CMakeFiles/mrpt-core.dir/src/aligned_malloc.cpp.o
/Users/aliben/project/source_code/mrpt/libs/core/src/aligned_malloc.cpp:28:11: error: 
      no member named 'aligned_alloc' in the global namespace
        return ::aligned_alloc(alignment, size);
               ~~^
1 error generated.
make[2]: *** [libs/core/CMakeFiles/mrpt-core.dir/src/aligned_malloc.cpp.o] Error 1
make[1]: *** [libs/core/CMakeFiles/mrpt-core.dir/all] Error 2
make: *** [all] Error 2
```

jolting, 2018-07-20:
Yep. That's because aligned_alloc is in the std namespace.
https://en.cppreference.com/w/cpp/memory/c/aligned_alloc

We're using the C++17 standard, not C11. Do you want to add std and try compiling?

alibenD, 2018-07-20:
Yes, I would try to build again. However it would change mrpt source code, and it would be not convenient to keep upstream repo.

One more question, besides to add std, is there any way to be compatible C++17 and C++11? Can mrpt deal this case according to different compilers? I am not very sure if condition compiling worked for that. 

Thank you for your solution. :) 👍

jlblancoc, 2018-07-20:
> Yes, I would try to build again. However it would change mrpt source code, and it would be not convenient to keep upstream repo if develop didn't use c++ 17.

No problem at all! If you are using a cloned repository of mrpt, just send us the output of `git diff` at the end of all the required changes, if you don't feel confident on pushing to a fork and creating a pull request, and we'll merge the changes.

alibenD, 2018-07-20:
Thanks for your comment.

There is a new error info while I added `std` beyond `::aligned_alloc`. It seems like that the std library doesn't have the function `std::aligned_alloc`. It is the same result on gcc(Apple) and gcc-8(gnu). 
```
/Users/aliben/project/source_code/mrpt/libs/core/src/aligned_malloc.cpp:28:14: error: 'aligned_alloc' is not a member of 'std'
  return std::aligned_alloc(alignment, size);
              ^~~~~~~~~~~~~
/Users/aliben/project/source_code/mrpt/libs/core/src/aligned_malloc.cpp:28:14: note: suggested alternative: 'aligned_union'
  return std::aligned_alloc(alignment, size);
              ^~~~~~~~~~~~~
              aligned_union
make[2]: *** [libs/core/CMakeFiles/mrpt-core.dir/src/aligned_malloc.cpp.o] Error 1
make[1]: *** [libs/core/CMakeFiles/mrpt-core.dir/all] Error 2
make: *** [all] Error 2
```

alibenD, 2018-07-20:
I am trying to use gcc-7(7.3 brew provided) to build again, the build result I will post in here later if the error appears.

jolting, 2018-07-20:
Code can be added to handle OS X, but I was under the naive assumption that OS X fully supported c++17. That's disappointing.

Does OS X use libc++ or do they have their own c++ library? This seems like a bug with the OS X c++ library not supporting c++17.

alibenD, 2018-07-21:
I found the path of cstdlib.h by the following command.
```
$ echo | cpp -xc++ -Wp,-v
clang -cc1 version 9.1.0 (clang-902.0.39.2) default target x86_64-apple-darwin17.7.0
ignoring nonexistent directory "/usr/include/c++/v1"
ignoring nonexistent directory "/usr/local/include"
ignoring duplicate directory "/usr/include"
  as it is a non-system directory that duplicates a system directory
#include "..." search starts here:
#include <...> search starts here:
 /Users/xxx/dev/env/homebrew/include/c++/8.1.0
 /Users/xxx/dev/env/homebrew/include
 /Users/xxx/dev/env/include
 .
 /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1
 /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/9.1.0/include
 /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include
 /usr/include
 /System/Library/Frameworks (framework directory)
 /Library/Frameworks (framework directory)
```

When I opened  `/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/cstdlib`, I found `void *aligned_alloc(size_t alignment, size_t size); ` was defined at this header file. 
```
namespace std
{
...
void* calloc(size_t nmemb, size_t size);
void free(void* ptr);
void* malloc(size_t size);
void* realloc(void* ptr, size_t size);
void abort(void);
...
void *aligned_alloc(size_t alignment, size_t size);                       // C11

}  // std
```

And also I checked complier options in Xcode, there is a c++17 support option. It may work in Xcode, but I haven't tried. However, I found an answer http://www.cplusplus.com/forum/beginner/229382/, where said it works in xcode but not terminal using g++ command.

alibenD, 2018-07-21:
> Does OS X use libc++ or do they have their own c++ library? This seems like a bug with the OS X c++ library not supporting c++17.

About this question, I found an answer from https://stackoverflow.com/questions/20848341/where-are-libc-headers-located-on-mac-os-x-mavericks
```
Mac OS X 10.9+ no longer uses GCC/libstdc++ but uses libc++ and Clang.
```

alibenD, 2018-07-21:
https://stackoverflow.com/questions/47478894/how-to-get-a-c17-environment-on-macos-high-sierra

There is a 'bad new'. Like @jolting said, this seems like a bug with MacOS10.13.6(Apple LLVM version 9.1.0 (clang-902.0.39.2)). 

I would try to compiling GCC 8 on macOS following https://solarianprogrammer.com/2017/05/21/compiling-gcc-macos/, since I am not sure if gcc-8 which brew provided miss something.

------------------

Update:

> The compiler shipped with Xcode supports C++17 language features but not C++17 standard library > features. Looking at your screenshot you will see the standard library support goes up to C++11, and > Apple has not yet shipped a version of clang that has stdlib support for either C++14 or C++17.
from https://stackoverflow.com/questions/49577343/filesystem-with-c17-doesnt-work-on-my-mac-os-x-high-sierra

alibenD, 2018-08-03:
> libc++ is also a 100% complete C++14 implementation. A list of new features and changes for C++14 can be found here.
> libc++'s C++17 implementation is not yet complete. A list of features and changes for C++17 can be found here.

Thus, it has to wait the completion of libc++'s C++17 implementation to build MRPT on MacOS

from https://libcxx.llvm.org/

jolting, 2018-08-03:
When MRPT 2.0 is released officially it probably won't be a problem. On the other hand, it's a shame to leave OS X users out in the cold. 

It looks like their C11 stuff is marked as complete.

Also, the missing functionality in question has been in libcxx for about 6 years.

https://github.com/llvm-mirror/libcxx/blob/9a13b433e32679309eb51c7d873acc0829154593/include/cstdlib#L158

It looks like it's not enabled for that platform though.
https://github.com/llvm-mirror/libcxx/blob/db9b3d81a40eaf854e6fa34b2aedcf2885e97983/include/__config#L433

I'm not sure if there is a way to enable it.

It's best to ask the libcxx devs about this.

jlblancoc, 2018-08-25:
@alibenD any news on this? Did you still want to contribute back some patch, to solve (at least partially) OSX problems?

alibenD, 2018-08-28:
@jlblancoc  Yes, I still want to solve this problem. However, no news on this topic with clang(llvm), so far as I found about clang(Apple LLVM version). It would be appreciated that if you have any suggestion action to do.

I have tried using gcc-8(GNU GCC) to compile, the problem is still, I guess that gcc use clang(llvm) include header(libc++) but not (libstd++), this should be problem with my environment setting. I am still looking for solution.

alibenD, 2018-09-18:
https://github.com/marian-nmt/marian-dev/issues/227
> The problem here is that macos has a bad implementation of C11, and doesn't implement aligned_alloc. I've confirm this with LLVM developers.

I found an alternative solution for `::aligned_malloc`, which is to replace `::aligned_alloc` with `boost::alignment::aligned_alloc`. I am trying now, the result would be posted later.

alibenD, 2018-09-18:
UPDATE:
-------
`#include "core-precomp.h"  // Precompiled headers
#include <mrpt/core/aligned_allocator.h>
#include <cstdlib>  // free, realloc, C++17 aligned_alloc
#include <cstring>  // memset
#ifdef __APPLE__
#include <boost/align/aligned_alloc.hpp>
#endif  // END of __APPLE__

void* mrpt::aligned_calloc(size_t bytes, size_t alignment)
{
	void* ptr = mrpt::aligned_malloc(bytes, alignment);
	if (ptr) ::memset(ptr, 0, bytes);
	return ptr;
}
void* mrpt::aligned_malloc(size_t size, size_t alignment)
{
	// size must be an integral multiple of alignment:
	if ((size % alignment) != 0) size = ((size / alignment) + 1) * alignment;
#ifdef _MSC_VER
	return _aligned_malloc(size, alignment);
#elif __APPLE__
  return boost::alignment::aligned_alloc(alignment, size);
#else
	return ::aligned_alloc(alignment, size);
#endif
}
void mrpt::aligned_free(void* ptr)
{
#ifdef _MSC_VER
	return _aligned_free(ptr);
#else
	return ::free(ptr);
#endif
`
 Error Info

`[  3%] Building CXX object libs/system/CMakeFiles/mrpt-system.dir/src/os.cpp.o
/Users/xxxx/project/source_code/mrpt/libs/system/src/os.cpp:692:3: error: 
      unknown type name 'HANDLE'
                HANDLE g_hChildStd_IN_Rd = NULL;
                ^
/Users/xxxx/project/source_code/mrpt/libs/system/src/os.cpp:693:3: error: 
      unknown type name 'HANDLE'
                HANDLE g_hChildStd_IN_Wr = NULL;
                ^
/Users/xxxx/project/source_code/mrpt/libs/system/src/os.cpp:694:3: error: 
      unknown type name 'HANDLE'
                HANDLE g_hChildStd_OUT_Rd = NULL;
                ^
/Users/xxxx/project/source_code/mrpt/libs/system/src/os.cpp:695:3: error: 
      unknown type name 'HANDLE'
                HANDLE g_hChildStd_OUT_Wr = NULL;
                ^
/Users/xxxx/project/source_code/mrpt/libs/system/src/os.cpp:697:3: error: 
      unknown type name 'HANDLE'
                HANDLE g_hInputFile = NULL;
                ^
/Users/xxxx/project/source_code/mrpt/libs/system/src/os.cpp:698:3: error: 
      unknown type name 'SECURITY_ATTRIBUTES'
                SECURITY_ATTRIBUTES saAttr;
                ^
/Users/xxxx/project/source_code/mrpt/libs/system/src/os.cpp:700:27: error: use of
      undeclared identifier 'SECURITY_ATTRIBUTES'
                saAttr.nLength = sizeof(SECURITY_ATTRIBUTES);
                                        ^
/Users/xxxx/project/source_code/mrpt/libs/system/src/os.cpp:701:27: error: use of
      undeclared identifier 'TRUE'
                saAttr.bInheritHandle = TRUE;
                                        ^
/Users/xxxx/project/source_code/mrpt/libs/system/src/os.cpp:705:10: error: use of
      undeclared identifier 'winerror2str'
                        throw winerror2str("StdoutRd CreatePipe");
                              ^
/Users/xxxx/project/source_code/mrpt/libs/system/src/os.cpp:709:49: error: use of
      undeclared identifier 'HANDLE_FLAG_INHERIT'
                if (!SetHandleInformation(g_hChildStd_OUT_Rd, HANDLE_FLAG_I...
                                                              ^
/Users/xxxx/project/source_code/mrpt/libs/system/src/os.cpp:710:10: error: use of
      undeclared identifier 'winerror2str'
                        throw winerror2str("Stdout SetHandleInformation");
                              ^
/Users/xxxx/project/source_code/mrpt/libs/system/src/os.cpp:715:10: error: use of
      undeclared identifier 'winerror2str'
                        throw winerror2str("Stdin CreatePipe");
                              ^
/Users/xxxx/project/source_code/mrpt/libs/system/src/os.cpp:719:48: error: use of
      undeclared identifier 'HANDLE_FLAG_INHERIT'
                if (!SetHandleInformation(g_hChildStd_IN_Wr, HANDLE_FLAG_IN...
                                                             ^
/Users/xxxx/project/source_code/mrpt/libs/system/src/os.cpp:720:10: error: use of
      undeclared identifier 'winerror2str'
                        throw winerror2str("Stdin SetHandleInformation");
                              ^
/Users/xxxx/project/source_code/mrpt/libs/system/src/os.cpp:723:3: error: 
      unknown type name 'PROCESS_INFORMATION'
                PROCESS_INFORMATION piProcInfo;
                ^
/Users/xxxx/project/source_code/mrpt/libs/system/src/os.cpp:724:3: error: 
      unknown type name 'STARTUPINFOA'
                STARTUPINFOA siStartInfo;
                ^
/Users/xxxx/project/source_code/mrpt/libs/system/src/os.cpp:725:3: error: 
      unknown type name 'BOOL'
                BOOL bSuccess = FALSE;
                ^
/Users/xxxx/project/source_code/mrpt/libs/system/src/os.cpp:725:19: error: use of
      undeclared identifier 'FALSE'
                BOOL bSuccess = FALSE;
                                ^
/Users/xxxx/project/source_code/mrpt/libs/system/src/os.cpp:729:34: error: use of
      undeclared identifier 'PROCESS_INFORMATION'
                ZeroMemory(&piProcInfo, sizeof(PROCESS_INFORMATION));
                                               ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [libs/system/CMakeFiles/mrpt-system.dir/src/os.cpp.o] Error 1
make[1]: *** [libs/system/CMakeFiles/mrpt-system.dir/all] Error 2
make: *** [all] Error 2
`

jolting, 2018-09-22:
This branch needs a thorough review:
https://github.com/MRPT/mrpt/tree/xcodebuild

I'm not sure if boost is the best thing to use for aligned_alloc.

Looks like we are stuck with 10.13 osx for now. Unfortunately it's missing support for std::any and some chrono stuff. Hopefully circleci will upgrade to 10.14 soon.

jolting, 2018-09-22:
10.14 launches September 24. Hopefully, CircleCI enables that stuff soon. BTW, CircleCI guys have been awesome so far.

jlblancoc, 2018-09-22:
Great! Having a more modern version of macos on CI will indeed help to solve problems like the ones in this thread...

alibenD, 2018-10-09:
@jolting Sorry for my late reply, I am just back from my vacation. I would try on 10.14.

BTW, using boost is not a best solution for it, I just tried a potential way. Anyway, thank you @jolting @jlblancoc for discussing with the problem I met.

alibenD, 2018-10-09:
UPDATE:

System: `macOS Mojave 10.14`
XCode Setting: 
[image]
----------------------------------
Error info:

[image]

PR #903 (2019-03-11): Fixes OSX build
closes #803
closed 2019-03-11 by jolting