MRPT/mrpt #804: libcvd will not build on windows

antithing, 2018-07-17:
I have pulled the source, and using cmake, created the build files. In Visual studio 2017, libcvd will not build, giving 100+ errors. is this a required dependency?

Thanks!

jolting, 2018-07-17:
It is. It was moved to the otherlibs to better comply with bsd and make it easier to upgrade. It's actually been a part of mrpt for some time. 

Appveyor isn't reporting a problem. Can you provide more details on yours?

antithing, 2018-07-17:
Sure! Some of the errors below: 

Steps to recreate: Pull the latest repo, build in cmake, open in visual studio. Have I missed something? Thanks!

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	C2556	'int CVD::ImageRef::operator [](int) const': overloaded function differs only by return type from 'int &CVD::ImageRef::operator [](int) const' [D:\thirdParty\mrpt-master\build\libcvd-prefix\src\libcvd-build\CVD.vcxproj]	libcvd (ExternalProjectTargets\libcvd\libcvd)	D:\thirdParty\mrpt-master\otherlibs\libcvd\cvd\image_ref.h	158	
Error	C2373	'CVD::ImageRef::operator []': redefinition; different type modifiers [D:\thirdParty\mrpt-master\build\libcvd-prefix\src\libcvd-build\CVD.vcxproj]	libcvd (ExternalProjectTargets\libcvd\libcvd)	D:\thirdParty\mrpt-master\otherlibs\libcvd\cvd\image_ref.h	158	
Error	C2511	'CVD::ImageRef &CVD::ImageRef::operator =(const CVD::ImageRef &)': overloaded member function not found in 'CVD::ImageRef' [D:\thirdParty\mrpt-master\build\libcvd-prefix\src\libcvd-build\CVD.vcxproj]	libcvd (ExternalProjectTargets\libcvd\libcvd)	D:\thirdParty\mrpt-master\otherlibs\libcvd\cvd\internal\image_ref_implementation.hh	55	
Error	C2597	illegal reference to non-static member 'CVD::ImageRef::x' [D:\thirdParty\mrpt-master\build\libcvd-prefix\src\libcvd-build\CVD.vcxproj]	libcvd (ExternalProjectTargets\libcvd\libcvd)	D:\thirdParty\mrpt-master\otherlibs\libcvd\cvd\internal\image_ref_implementation.hh	56	
Error	C2597	illegal reference to non-static member 'CVD::ImageRef::y' [D:\thirdParty\mrpt-master\build\libcvd-prefix\src\libcvd-build\CVD.vcxproj]	libcvd (ExternalProjectTargets\libcvd\libcvd)	D:\thirdParty\mrpt-master\otherlibs\libcvd\cvd\internal\image_ref_implementation.hh	57	
Error	C2671	'CVD::ImageRef::=': static member functions do not have 'this' pointers [D:\thirdParty\mrpt-master\build\libcvd-prefix\src\libcvd-build\CVD.vcxproj]	libcvd (ExternalProjectTargets\libcvd\libcvd)	D:\thirdParty\mrpt-master\otherlibs\libcvd\cvd\internal\image_ref_implementation.hh	58	
Error	C3250	'v': declaration is not allowed in 'constexpr' function body [D:\thirdParty\mrpt-master\build\libcvd-prefix\src\libcvd-build\CVD.vcxproj]	libcvd (ExternalProjectTargets\libcvd\libcvd)	D:\thirdParty\mrpt-master\otherlibs\libcvd\cvd\internal\image_ref_implementation.hh	73	
Error	C3249	illegal statement or sub-expression for 'constexpr' function [D:\thirdParty\mrpt-master\build\libcvd-prefix\src\libcvd-build\CVD.vcxproj]	libcvd (ExternalProjectTargets\libcvd\libcvd)	D:\thirdParty\mrpt-master\otherlibs\libcvd\cvd\internal\image_ref_implementation.hh	73	
Error	C2511	'CVD::ImageRef &CVD::ImageRef::operator *=(const double)': overloaded member function not found in 'CVD::ImageRef' [D:\thirdParty\mrpt-master\build\libcvd-prefix\src\libcvd-build\CVD.vcxproj]	libcvd (ExternalProjectTargets\libcvd\libcvd)	D:\thirdParty\mrpt-master\otherlibs\libcvd\cvd\internal\image_ref_implementation.hh	78	
Error	C2597	illegal reference to non-static member 'CVD::ImageRef::x' [D:\thirdParty\mrpt-master\build\libcvd-prefix\src\libcvd-build\CVD.vcxproj]	libcvd (ExternalProjectTargets\libcvd\libcvd)	D:\thirdParty\mrpt-master\otherlibs\libcvd\cvd\internal\image_ref_implementation.hh	79	
Error	C3867	'CVD::ImageRef::x': non-standard syntax; use '&' to create a pointer to member [D:\thirdParty\mrpt-master\build\libcvd-prefix\src\libcvd-build\CVD.vcxproj]	libcvd (ExternalProjectTargets\libcvd\libcvd)	D:\thirdParty\mrpt-master\otherlibs\libcvd\cvd\internal\image_ref_implementation.hh	79	
Error	C2597	illegal reference to non-static member 'CVD::ImageRef::y' [D:\thirdParty\mrpt-master\build\libcvd-prefix\src\libcvd-build\CVD.vcxproj]	libcvd (ExternalProjectTargets\libcvd\libcvd)	D:\thirdParty\mrpt-master\otherlibs\libcvd\cvd\internal\image_ref_implementation.hh	80	
Error	C3867	'CVD::ImageRef::y': non-standard syntax; use '&' to create a pointer to member [D:\thirdParty\mrpt-master\build\libcvd-prefix\src\libcvd-build\CVD.vcxproj]	libcvd (ExternalProjectTargets\libcvd\libcvd)	D:\thirdParty\mrpt-master\otherlibs\libcvd\cvd\internal\image_ref_implementation.hh	80	
Error	C2671	'CVD::ImageRef::*=': static member functions do not have 'this' pointers [D:\thirdParty\mrpt-master\build\libcvd-prefix\src\libcvd-build\CVD.vcxproj]	libcvd (ExternalProjectTargets\libcvd\libcvd)	D:\thirdParty\mrpt-master\otherlibs\libcvd\cvd\internal\image_ref_implementation.hh	81	
Error	C2511	'CVD::ImageRef &CVD::ImageRef::operator /=(const double)': overloaded member function not found in 'CVD::ImageRef' [D:\thirdParty\mrpt-master\build\libcvd-prefix\src\libcvd-build\CVD.vcxproj]	libcvd (ExternalProjectTargets\libcvd\libcvd)	D:\thirdParty\mrpt-master\otherlibs\libcvd\cvd\internal\image_ref_implementation.hh	85	
Error	C2597	illegal reference to non-static member 'CVD::ImageRef::x' [D:\thirdParty\mrpt-master\build\libcvd-prefix\src\libcvd-build\CVD.vcxproj]	libcvd (ExternalProjectTargets\libcvd\libcvd)	D:\thirdParty\mrpt-master\otherlibs\libcvd\cvd\internal\image_ref_implementation.hh	86	
Error	C3867	'CVD::ImageRef::x': non-standard syntax; use '&' to create a pointer to member [D:\thirdParty\mrpt-master\build\libcvd-prefix\src\libcvd-build\CVD.vcxproj]	libcvd (ExternalProjectTargets\libcvd\libcvd)	D:\thirdParty\mrpt-master\otherlibs\libcvd\cvd\internal\image_ref_implementation.hh	86	
[... cut ...]
Error	C3249	illegal statement or sub-expression for 'constexpr' function [D:\thirdParty\mrpt-master\build\libcvd-prefix\src\libcvd-build\CVD.vcxproj]	libcvd (ExternalProjectTargets\libcvd\libcvd)	D:\thirdParty\mrpt-master\otherlibs\libcvd\cvd\internal\scalar_convert.h	106	
Error	C2131	expression did not evaluate to a constant [D:\thirdParty\mrpt-master\build\libcvd-prefix\src\libcvd-build\CVD.vcxproj]	libcvd (ExternalProjectTargets\libcvd\libcvd)	D:\thirdParty\mrpt-master\otherlibs\libcvd\cvd\internal\scalar_convert.h	112	
```

jolting, 2018-07-17:
Are you sure you're invoking the 2017 compiler?

antithing, 2018-07-17:
yes, vs2017(141).

jlblancoc, 2018-07-30:
Hi, 
I'll close this one for lack of reproducibility and feedback. If the error persists with a clean CMake build directory and an updated version of VS, please, reopen with a copy of CMakeCache.txt.

closed 2018-07-30 by jlblancoc