MRPT/mrpt #811: Build failure with Visual C++ 2017

JackHunt, 2018-08-04:
Hi all,
When building on Windows with Visual Studio/C++ 2017 I encounter the following link error:

```
Link:

  C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.14.26428\bin\HostX86\x64\link.exe /ERRORREPORT:QUEUE /OUT:"C:\Users\Jack Hunt\Desktop\mrpt-1.5.6\build\CMakeFiles\CMakeTmp\Debug\cmTC_00a14.exe" /INCREMENTAL /NOLOGO kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST /MANIFESTUAC:"level='asInvoker' uiAccess='false'" /manifest:embed /DEBUG /PDB:"C:/Users/Jack Hunt/Desktop/mrpt-1.5.6/build/CMakeFiles/CMakeTmp/Debug/cmTC_00a14.pdb" /SUBSYSTEM:CONSOLE /TLBID:1 /DYNAMICBASE /NXCOMPAT /IMPLIB:"C:/Users/Jack Hunt/Desktop/mrpt-1.5.6/build/CMakeFiles/CMakeTmp/Debug/cmTC_00a14.lib" /MACHINE:X64  /machine:x64 cmTC_00a14.dir\Debug\CheckFunctionExists.obj

CheckFunctionExists.obj : error LNK2019: unresolved external symbol timegm referenced in function main [C:\Users\Jack Hunt\Desktop\mrpt-1.5.6\build\CMakeFiles\CMakeTmp\cmTC_00a14.vcxproj]

C:\Users\Jack Hunt\Desktop\mrpt-1.5.6\build\CMakeFiles\CMakeTmp\Debug\cmTC_00a14.exe : fatal error LNK1120: 1 unresolved externals [C:\Users\Jack Hunt\Desktop\mrpt-1.5.6\build\CMakeFiles\CMakeTmp\cmTC_00a14.vcxproj]

Done Building Project "C:\Users\Jack Hunt\Desktop\mrpt-1.5.6\build\CMakeFiles\CMakeTmp\cmTC_00a14.vcxproj" (default targets) -- FAILED.

Build FAILED.
`

The symbol in question in the above error is `timegm`, although others such as `alloca` and `gettid` also cause linkage errors. 

From a little research online, it appears that these symbols would be present on a POSIX system, but on Windows they are prefixed with an underscore. This would suggest a missing `#if defined(WIN32) && !defined(UNIX)` or the like.

Best,
Jack

JackHunt, 2018-08-05:
Ok. At present it is still an issue on master. At some point next week I'll try and get it working and submit a PR if it works. I thought I'd flag it in case it is something obvious to the current contributors.

Best,
Jack

jlblancoc, 2018-08-05:
Anyway... I could swear I recently built the branch mrpt-1.5 with VS2017 without problems... Could you please update CMake to one of the latest versions, cloning/downloading mrpt-1.5 branch, and give it another try?

I say this because the error in `CheckFunctionExists` looks more like a problem of CMake itself than of MRPT (!)... 

Cheers.

JackHunt, 2018-08-07:
Hi, 
Sorry for the delay. I can confirm that the mrpt-1.5 branch builds without fail on my system. I did not need to update CMake as I already run v3.12.0

I first built with just wxWidgets and OpenCV and that was fine. I then rebuilt with PCL, VTK, OpenNI2 and Qt also without fail.