MRPT/mrpt #825: ICP Slam live view is blank over X11 server

andrewda, 2018-09-09:
Hi there!

Whenever I attempt to connect to my robot (a Jetson TX1 mounted on a roomba with a lidar) over X11 with MobaXterm, I get a strange blank window. When running with an HDMI monitor connected to the Jetson the map shows up fine. Not sure what the issue is here. Does anyone have any experience with anything similar? How do you all view your SLAM maps remotely?

[image]

andrewda, 2018-09-09:
A similar issue occurs when trying to use `SceneViewer3D` over X11, after having the following errors pop up:

```
nvidia@tegra-ubuntu:~/Desktop/dunk/build$ SceneViewer3D LOG_ICP-SLAM/buildingmap_04550.3Dscene
08:55:46: Warning: Mismatch between the program and library build versions detected.
The library used 3.0 (wchar_t,compiler with C++ ABI 1009,wx containers,compatible with 2.8),
and your program used 3.0 (wchar_t,compiler with C++ ABI 1011,wx containers,compatible with 2.8).
../src/gtk/window.cpp(2963): assert "(m_widget != __null)" failed in DoGetClientSize(): invalid window
../src/gtk/window.cpp(2831): assert "m_widget" failed in DoSetSize(): invalid window
../src/gtk/window.cpp(2831): assert "m_widget" failed in DoSetSize(): invalid window
```

The errors seem unrelated, but at least worth mentioning just in case.

[image]

jlblancoc, 2018-09-11:
Thanks for reporting! 

I'm not sure about this one (I think I have only tested X11 from another Ubuntu machine, not cross OS), but it **might** be a limitation of the X client on Windows? Just my wild guess without much further research. 

One solution, which has been used in some products (not in the released code for MRPT, though) is to render the map to images and send the images for displaying. One could even render the full 3D scene with [mrpt::opengl::CFBORender](http://mrpt.ual.es/reference/devel/classmrpt_1_1opengl_1_1_c_f_b_o_render.html).

Sorry, I don't know any better solution...

andrewda, 2018-09-11:
I'll take a go at it with a MacBook and see if that solves it. Otherwise the approach you suggested seems like the best solution.

jlblancoc, 2019-07-30:
Not sure it's related to Mac, but just in case:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=900678

It seems one has to set the env var `GDK_BACKEND=x11` for OpenGL to work in some cases...