MRPT/mrpt #850: reactive nav: fix potential "call to pure virtual method" at dtor

jlblancoc, 2018-10-23:
Description: 
At least it happens once at `CReactiveNavigationSystem::~CReactiveNavigationSystem()`, where it ends up calling the virtual method `stop()`. 

This breaks Scott Meyers' "Effective C++": 
```
Item 9: Never call virtual functions during construction or destruction
```

To-do: 
* [ ] fix in mrpt-1.5 (adding issue to next 1.5 milestone)
* [ ] fix into master. 

PS: Does anyone know about any clang/gcc flag to automatically detect these issues? It seems GCC's `-Weffc++` doesn't include "item 9"...  (cc: @jolting )

jolting, 2018-10-24:
optin.cplusplus.VirtualCall 
https://clang-analyzer.llvm.org/available_checks.html

jolting, 2018-10-24:
install clang-tools
scan-build -enable-checker optin.cplusplus.VirtualCall make -j3

jlblancoc, 2018-10-24:
Great! Thanks a lot for the pointer.

Unfortunately, it seems that the static analyzer is not able to detect indirect cases (as the one reported above) where the virtual method call is done a few calls down a chain of function calls at the dtor... (sigh)

However, it found *other* places where this bad practice was used, so it's worth!

PS: For it to work I had to use `scan-build` during the first call to `cmake` instead of `make` for the CXX variable to get caught. For future reference:
```
rm -fr build
mkdir build  && cd build
scan-build -enable-checker optin.cplusplus.VirtualCall cmake ..
make ...
```

jolting, 2018-10-24:
This one might be fun: alpha.clone.CloneChecker

closed 2018-10-29 by jlblancoc