MRPT/mrpt #858: run error

hitzhang, 2018-11-06:
(MRPT:12213): Gtk-ERROR **: GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported

hitzhang, 2018-11-06:
./plslam_dataset /kitti/00 

Initializing PL-SLAM....1

(MRPT:12213): Gtk-ERROR **: GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported
追踪与中断点陷阱

hitzhang, 2018-11-06:
sudo apt-get install libwxgtk2.8-dev  is not helpful when Compile mrpt and plslam. so i compile GtkWidget from source. and there is a problem like
 (MRPT:12213): Gtk-ERROR **: GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported.
and i find the 2.88 of GtkWidget can not link to the  GTK+ 3.
what should i do?

jlblancoc, 2018-11-06:
I think we saw that in the past, and was caused by another lib (opencv?)
being build against wx3.
Do you have ros installed by chance?
Please attach your CMakeCache.txt.

Probably the solution is to build mrpt against wx3?

jolting, 2018-11-06:
same issue as #627

jolting, 2018-11-06:
Forgot to mention.
run `update-alternatives --config wx-config` and pick the one that matches your opencv build before running MRPT cmake.

jlblancoc, 2018-11-07:
Exactly, it's a dup of #627 . 
@hitzhang : Please see [this](https://github.com/MRPT/mrpt/issues/627#issuecomment-404176189) in case it helps.

hitzhang, 2018-11-07:
@jlblancoc 
thank you so much.i solve it through by compiling opencv3 with GTK2.
I try to compile wxWidget  with gtk3 by . ./configure --with-gtk=3. but it can not work.

jlblancoc, 2018-11-12:
For the records: before cmake-configuring MRPT, one can select the `gtk3-unicode-3.0` configuration of wxWidgets with: 

```
sudo update-alternatives --config wx-config 
There are 3 choices for the alternative wx-config (providing /usr/bin/wx-config).

  Selection    Path                                                  Priority   Status
------------------------------------------------------------
* 0            /usr/lib/x86_64-linux-gnu/wx/config/gtk2-unicode-3.0   308       auto mode
  1            /usr/lib/x86_64-linux-gnu/wx/config/base-unicode-3.0   306       manual mode
  2            /usr/lib/x86_64-linux-gnu/wx/config/gtk2-unicode-3.0   308       manual mode
  3            /usr/lib/x86_64-linux-gnu/wx/config/gtk3-unicode-3.0   307       manual mode

Press <enter> to keep the current choice[*], or type selection number: 3
update-alternatives: using /usr/lib/x86_64-linux-gnu/wx/config/gtk3-unicode-3.0 to provide /usr/bin/wx-config (wx-config) in manual mode
```

jlblancoc, 2018-11-13:
For the records-bis 😆 : 

It turns out that there exists an `WX_CONFIG` env var that CMake will honor... 
We should try it for the PPA and Debian builds, otherwise all GUIs are useless since they crash before starting up (!). 

Reopening not to forget it.

jlblancoc, 2018-11-18:
This is finally solved with a more elegant solution: 7e4eb8d03439fad43697b36d9ce826b4cb7b824d

closed 2018-11-18 by jlblancoc

HeyLoong, 2023-12-18:
I solved this problem by compiling and setting OpenCV and wxWidgets to support GTK 3. The specific steps are as follows:
1. libgtk-3-dev installation
2. OpenCV compilation supports GTK 3
3. wxWidgets is compiled to support GTK 3
4. MRPT recompilation

For the specific process, please refer to the demo I wrote:
https://blog.csdn.net/HeyLoong/article/details/135072345

commit 7e4eb8d0 (2018-11-18): Using wxWidgets GTK3 version if available.

Not that in Ubuntu this requires installing `libwxgtk3.0-gtk3-dev`.

The motivation for this is that opencv3 uses gtk3, so using opencv3 implies
that no GUI could be open in MRPT if using gtk2.

Closes #858.