MRPT/mrpt #862: Corrupt particles with RBPF after resampling

Wehrdo, 2018-11-12:
When running the rbpf-slam application with `gridmapping_optimal_sampling.ini` as config file and `2006-01ENE-21-SENA_Telecom Faculty_one_loop_only.rawlog` as dataset, when changing the particle filter algorithm to one that includes resampling (pfStandardProposal or pfOptimalProposal), after the resampling step, a memory access exception is thrown.

Many particles in `CMultiMetricMapPDF::m_particles` seem to have corrupt data.

Reproducing this is very consistent, especially if you increase BETA enough so that resampling happens.

**Version**
This is with latest master compiled on MSVC 15.8.7

jlblancoc, 2018-11-13:
Good one! I could reproduce in GCC, so the problem must be in the new resampling algorithm for the "mrpt2" branch. BTW, it's a regression, since in mrpt-1.5 this works nice. 

Will look at it asap. Thanks for the detailed report.

PS: And this is why #714  is so important!

jlblancoc, 2018-11-16:
Fixed now. 
Let us know if you find further issues.

commit a4bcb080 (2018-11-16): Fix PF resampling memory corruption

Fixes #862
closed 2018-11-16 by jlblancoc