MRPT/mrpt #870: CMake ERROR "mrpt-core" references to file not found

JGMonroy, 2018-11-29:
After updating the find_package(MRPT modules) to the new find_package(mrpt-module1) as described in #859 , I get the CMake error:

**CMake Error** at /usr/share/mrpt-core/mrpt-core-targets.cmake:66 (message):
  The imported target "mrpt-core" references the file:

     "/build/mrpt-4oAlri/mrpt-1.9.9~snapshot20181128-1105-git-68311865-xenial/debian/libmrpt-core1.9/usr/lib/x86_64-linux-gnu/libmrpt-core.so.1.9.9"

**but this file does not exist.** 

The MRPT part in my CMakeLists.txt reads:

find_package(mrpt-obs)
find_package(mrpt-maps)
find_package(mrpt-slam)
find_package(mrpt-poses)
#find_package(mrpt-core)
...
...
target_link_libraries(node_name
   mrpt::mrpt-obs
   mrpt::mrpt-maps
   mrpt::mrpt-slam
   mrpt::mrpt-poses
   #mrpt::mrpt-core
)

bzdfzfer, 2018-11-29:
I also met this error ....

jlblancoc, 2018-11-30:
Progress: I traced this to be caused by a different behavior in CMake when exporting targets depending on whether the install dir is absolute or relative: 
https://gitlab.kitware.com/cmake/cmake/commit/dd089e08b578f20b7dc7d2ce658e3df05e346e35#122970eae82ce10a19e9f72da0d5d40e815cb92b_76_77

[image]

jlblancoc, 2018-11-30:
a407bdc0970488737e297a1123c4519b85e8866b may fix the problem... Let's wait for the PPA builders!

bzdfzfer, 2018-11-30:
Is there any chance we can modify the .cmake file manually?

jlblancoc, 2018-12-01:
@bzdfzfer I guess the easiest and safest would be for you to manually build from sources until this one is fixed...

I'm was trying a `git bisect` trying to find the origin of the issue, but wasn't successful so far (the "test" in this case takes 1-2 hours, sigh...)

closed 2018-12-03 by jlblancoc

jlblancoc, 2018-12-03:
@JGMonroy , @bzdfzfer : Please, stay tuned to the ongoing builds at [the PPA](https://launchpad.net/~joseluisblancoc/+archive/ubuntu/mrpt/+packages) and let me know if that one works fine... It does on a local `debuild`, so it should be fixed now.

PR #874 (2018-12-03): Fix CMake targets in Debian pkgs
Closes: #870 
Closes: #184
commit 4908c90f (2018-12-03): Fix CMake targets in Debian pkgs (#874)

Closes: #870
Closes: #184