MRPT/mrpt #871: Use c++17 aligned new

jolting, 2018-11-29:
See 3
https://en.cppreference.com/w/cpp/memory/new/operator_new

All classes must properly specify alignment requirements.

This probably makes the aligned*.h code unnecessary.

jlblancoc, 2018-12-04:
Still, it doesn't allow us to get rid of the Eigen-inherited `MRPT_MAKE_ALIGNED_OPERATOR_NEW` macro, right? 
It could, however, be modified to use some per-class alignment (from a `constexpr std::size_t required_alignment = XXX;` tag?) or something alike... 
Just thinking aloud.

jlblancoc, 2018-12-05:
wow, thanks! I was aware of that paper but I missed a key point: that `new T;` automatically calls the `new(size_t, align_t)` version of `new`!
Sure, then we can finally get rid of more home-made code... good ;-)

jolting, 2018-12-05:
iOS doesn't implement the aligned alloc function until iOS 11, so if you try to compile c++17 on iOS 10 it becomes obvious because you get a bunch aligned `new` errors. It is a bit of a surprise when you aren't aware that you are using classes all along that need to be over-aligned. Those are hard bugs to find. It's very easy to accidentally call new to construct objects that need to be over-aligned, so I'm glad they fixed this. Instead of silently calling the wrong new without a compiler error it now just works. It's a great change to the language.

PR #924 (2019-05-31): Remove custom aligned STL containers
Fixed-size overaligned structures do not longer require special
allocators since C++17.

See, for example: http://eigen.tuxfamily.org/bz/show_bug.cgi?id=1409

Closes #871
commit 931250ad (2019-05-31): Remove custom aligned STL containers

Fixed-size overaligned structures do not longer require special
allocators since C++17.

See, for example: http://eigen.tuxfamily.org/bz/show_bug.cgi?id=1409

Closes #871
closed 2019-05-31 by jlblancoc