MRPT/mrpt #873: Cmake error

mengqingyu123, 2018-12-03:
I encountered such a problem when I was executing cmake, may I ask how to solve it? Thank you very much!
[image]

jlblancoc, 2018-12-03:
Is that a git commit version from the mrpt-1.5 branch by chance? If yes, please try with the latest version at branch mrpt-1.5...

jlblancoc, 2018-12-03:
PS: Duplicate of #833... Please, continue the discussion there. Thanks!

closed 2018-12-03 by jlblancoc

mengqingyu123, 2018-12-03:
Thank you very much for your reply. However, I had some problems when I used the version you mentioned. 

/usr/bin/ld: /usr/local/lib/libwx_gtk2u_core-3.1.a(corelib_image.o): relocation R_X86_64_32S against `_ZTV8wxObject' can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libwx_gtk2u_core-3.1.a: 无法添加符号: 错误的值
collect2: error: ld returned 1 exit status
libs/base/CMakeFiles/mrpt-base.dir/build.make:4164: recipe for target 'lib/libmrpt-base.so.1.5.7' failed
make[2]: *** [lib/libmrpt-base.so.1.5.7] Error 1
CMakeFiles/Makefile2:1798: recipe for target 'libs/base/CMakeFiles/mrpt-base.dir/all' failed
make[1]: *** [libs/base/CMakeFiles/mrpt-base.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

Please don't mind if Chinese appears on it，Thank you!

ccccwb, 2021-02-10:
> 非常感谢你的回复。但是，当我使用您提到的版本时，我有一些问题。
> 
> /usr/bin/ld： /usr/local/lib/libwx_gtk2u_core-3.1.a（corelib_image.o）： 在创建共享对象时不能对 _ZTV8wxObject R_X86_64_32S""进行重新定位";重新编译与 -fpic
> /usr/本地/lib/libwx_gtk2u_core-3.1.a： *：
> [收集2：错误：ld返回 1 退出
> 状态 libs/base/Cmakefiles/mrpt - base. dir / build. make：4164： 目标 'lib / libmrpt - base.
> so. 1.5.7' 失败的配方 [2]： *** [lib / libmrpt - base. 所以.1.5.7] 错误 1
> CMakefiles/Makefile2：1798：
> 目标 "libs/base/makekefiles/mrpt-base.dir/all" 的配方失败 {1}： *** [libs/base/Cmakefiles/mrpt-base.dir/all] 错误 2
> Makefile：160： 目标"所有"失败的配方
> ： *** [所有] 错误 2
> 
> 请不要介意中文出现，谢谢！

请问您解决这个问题了吗？