MRPT/mrpt #875: Segfault on Returning CPose2D by Value in Python2.7 Bindings in GCC-7

ancantus, 2018-12-06:
MRPT Version: 1.5.6 tag checkout (with some patches on top for my project)
GCC Version: 7.3.0 (Ubuntu Bionic default install)

Came across this bug as I was migrating from Ubuntu Xenial to Bionic. When calling functions within the python wrapper that return the `CPose2D` object by value (example below will be the function: `CMonteCarloLocalization2D::getMean()`) the python code will segfault _sometimes_.

End of a GDB backtrace of the segfault:
```
Thread 1 "python" received signal SIGSEGV, Segmentation fault.
#0  0x00007fffcd2af0ed in boost::python::converter::as_to_python_function<mrpt::poses::CPose2D, boost::python::objects::class_cref_wrapper<mrpt::poses::CPose2D, boost::python::objects::make_instance<mrpt::poses::CPose2D, boost::python::objects::value_holder<mrpt::poses::CPose2D> > > >::convert(void const*) ()
    at $VENV_PATH/python2.7/site-packages/pymrpt.so
#1  0x00007fffcd316b2f in boost::python::objects::caller_py_function_impl<boost::python::detail::caller<mrpt::poses::CPose2D (*)(mrpt::slam::CMonteCarloLocalization2D&), boost::python::default_call_policies, boost::mpl::vector2<mrpt::poses::CPose2D, mrpt::slam::CMonteCarloLocalization2D&> > >::operator()(_object*, _object*) ()
    at $VENV_PATH/local/lib/python2.7/site-packages/pymrpt.so
#2  0x00007ffff0814e55 in boost::python::objects::function::call(_object*, _object*) const ()
    at /usr/lib/x86_64-linux-gnu/libboost_python-py27.so.1.65.1
#3  0x00007ffff0815008 in  () at /usr/lib/x86_64-linux-gnu/libboost_python-py27.so.1.65.1
#4  0x00007ffff081b9a3 in boost::python::handle_exception_impl(boost::function0<void>) ()
    at /usr/lib/x86_64-linux-gnu/libboost_python-py27.so.1.65.1
#5  0x00007ffff08126c9 in  () at /usr/lib/x86_64-linux-gnu/libboost_python-py27.so.1.65.1
#6  0x000055555564f420 in PyObject_Call (kw=0x0, arg=(<CMonteCarloLocalization2D at remote 0x7fff5a631d60>,), func=<optimized out>) at ../Objects/abstract.c:2547
#7  0x000055555564f420 in do_call (nk=<optimized out>, na=<optimized out>, pp_stack=0x7fffffffdc70, func=<optimized out>) at ../Python/ceval.c:4589
#8  0x000055555564f420 in call_function (oparg=<optimized out>, pp_stack=0x7fffffffdc70) at ../Python/ceval.c:4394
```

While debugging this I found the segfault also didn't trigger when I changed the python bindings allocate the `CPose2D` on the heap and let Python manage the memory.
_In pose_bindings.cpp_
```
CPose2D* CMonteCarloLocalization2D_getMean(CMonteCarloLocalization2D& self)
{
    CPose2D* mean = new CPose2D();
    self.getMean(*mean);
    return mean;
}

// In CMonteCarloLocalization2D class bindings
.def("getMean", make_function(CMonteCarloLocalization2D_getMean, return_value_policy<manage_new_object>()), 
       "Returns an estimate of the pose, (the mean, or mathematical expectation of the PDF).")
```

The solution I've arrived at has been to drop GCC from version 7.3.0 to version 5.5.0, which has worked for the time being.

jlblancoc, 2018-12-06:
Interesting.... Thanks for the detailed report!

Though, I can't think of any obvious explanation for this, since ABI changes, I think, shouldn't affect python bindings if everything is recompiled... (?)

Just for curiosity, have you tried the python binding for master? We have a large lack of unit testing of python bindings... so any help testing or fixing them would be greatly appreciated!

ancantus, 2018-12-06:
I've not tried current master, I'll give it a go next week.

I'm also quite puzzled by the break, the MPRT code seems to be doing the _right thing_. I've come across one [bug report to boost-python](https://github.com/boostorg/python/issues/232) that may be related. The reporter over there is unsure where the problem lies as well (could be in boost-python, eigen, custom code, or a relationship between any of them).

jolting, 2018-12-06:
Hmmm. Perhaps it's time to switch to pybind11. It's header only, so at least that eliminates the possibility of an ABI mismatch.

ancantus, 2018-12-12:
Was not able to rebase enough of my local changes on master to run the tests I have. It seems recent changes to serialization have broken how I've been reading gridmaps from files in python.

Will respond again if I get it working / have any more info. But not highest priority at the moment.

jolting, 2018-12-12:
Thanks for the effort.
The 2.0 branch made quite a lot of breaking changes. Hopefully, it's cleaner. If you have any suggestions on how to make that transition easier I'd appreciate the feedback.

jlblancoc, 2019-11-11:
Something similar is still happening, even after upgrading the whole python bindings area to use python3. Example, running the `rbpf-slam.py` demo:

``` 
Thread 1 "python3" received signal SIGSEGV, Segmentation fault.
0x00007ffff302f640 in mrpt::poses::CPose3D::CPose3D() () from /home/jlblanco/code/mrpt/build/lib/libmrpt-poses.so.1.9
(gdb) bt
#0  0x00007ffff302f640 in mrpt::poses::CPose3D::CPose3D() () at /home/jlblanco/code/mrpt/build/lib/libmrpt-poses.so.1.9
#1  0x00007ffff308f648 in mrpt::poses::CPoseRandomSampler::CPoseRandomSampler() () at /home/jlblanco/code/mrpt/build/lib/libmrpt-poses.so.1.9
#2  0x00007ffff4ca0c74 in mrpt::maps::CMultiMetricMapPDF::CMultiMetricMapPDF(mrpt::bayes::CParticleFilter::TParticleFilterOptions const&, mrpt::maps::TSetOfMetricMapInitializers const&, mrpt::maps::CMultiMetricMapPDF::TPredictionParams const&) () at /home/jlblanco/code/mrpt/build/lib/libmrpt-slam.so.1.9
#3  0x00007ffff4c5d472 in mrpt::slam::CMetricMapBuilderRBPF::CMetricMapBuilderRBPF(mrpt::slam::CMetricMapBuilderRBPF::TConstructionOptions const&) () at /home/jlblanco/code/mrpt/build/lib/libmrpt-slam.so.1.9
#4  0x00007ffff5d15df0 in boost::python::objects::make_holder<1>::apply<boost::python::objects::value_holder<mrpt::slam::CMetricMapBuilderRBPF>, boost::mpl::vector1<mrpt::slam::CMetricMapBuilderRBPF::TConstructionOptions> >::execute(_object*, mrpt::slam::CMetricMapBuilderRBPF::TConstructionOptions) () at /home/jlblanco/code/mrpt/build/lib/pymrpt.so
#5  0x00007ffff5d35737 in boost::python::detail::caller_arity<2u>::impl<void (*)(_object*, mrpt::slam::CMetricMapBuilderRBPF::TConstructionOptions), boost::python::default_call_policies, boost::mpl::vector3<void, _object*, mrpt::slam::CMetricMapBuilderRBPF::TConstructionOptions> >::operator()(_object*, _object*) () at /home/jlblanco/code/mrpt/build/lib/pymrpt.so
#6  0x00007ffff5632855 in boost::python::objects::function::call(_object*, _object*) const () at /usr/lib/x86_64-linux-gnu/libboost_python3-py36.so.1.65.1
#7  0x00007ffff5632a18 in  () at /usr/lib/x86_64-linux-gnu/libboost_python3-py36.so.1.65.1
#8  0x00007ffff56393b3 in boost::python::handle_exception_impl(boost::function0<void>) () at /usr/lib/x86_64-linux-gnu/libboost_python3-py36.so.1.65.1
#9  0x00007ffff5630119 in  () at /usr/lib/x86_64-linux-gnu/libboost_python3-py36.so.1.65.1
#10 0x0000000000594f4c in  ()
#11 0x000000000054a2c5 in  ()
#12 0x0000000000551761 in  ()
#13 0x00000000005aa69c in _PyObject_FastCallKeywords ()
#14 0x000000000050ab53 in  ()
#15 0x000000000050c549 in _PyEval_EvalFrameDefault ()
#16 0x00000000005081d5 in  ()
#17 0x000000000050b3a3 in PyEval_EvalCode ()
#18 0x0000000000635082 in  ()
#19 0x0000000000635137 in PyRun_FileExFlags ()
#20 0x00000000006388ef in PyRun_SimpleFileExFlags ()
#21 0x0000000000639491 in Py_Main ()
#22 0x00000000004b0f60 in main ()
``` 

Perhaps the best thing to do (someday) is to drop the current wrappers and start from scratch, and go for cython or any other more-maintainable alternative?

jolting, 2019-11-11:
pybind11 is probably better given the C++.

jlblancoc, 2023-07-03:
The port to pybind11 was finished a few weeks ago so this bug is now obsolete.

closed 2023-07-03 by jlblancoc