MRPT/mrpt #880: CMAKE error

Adlirrahaman, 2018-12-12:


jolting, 2018-12-12:
Are you missing the build-essential package?
Try `sudo apt install build-essential`

Adlirrahaman, 2018-12-13:
> Are you missing the build-essential package?
> Try `sudo apt install build-essential`

i've install that package but still same

Adlirrahaman, 2018-12-13:
maybe i missing that package , how can i found that

Adlirrahaman, 2018-12-13:
maybe i missing that package , how can i found that?

jolting, 2018-12-13:
Normally on Ubuntu this is managed by alternatives, so it should look something like this:
```
$ readlink /usr/bin/c++
/etc/alternatives/c++
$ readlink /etc/alternatives/c++
/usr/bin/g++
```

jolting, 2018-12-13:
If you want to set your c++ alternative you should run
```
sudo update-alternatives --config c++
```
and for consistency
```
sudo update-alternatives --config cc
```
It is certainly is not recommended on an Ubuntu machine to manually edit your symlinks.

Adlirrahaman, 2018-12-14:
> Normally on Ubuntu this is managed by alternatives, so it should look something like this:
> 
> ```
> $ readlink /usr/bin/c++
> /etc/alternatives/c++
> $ readlink /etc/alternatives/c++
> /usr/bin/g++
> ```
then if its same ? i allready check it then its same but still didnt work

jlblancoc, 2018-12-16:
From [this](https://stackoverflow.com/questions/45140273/cmake-c-compiler-is-not-a-full-path-to-an-existing-compiler-tool), try: `apt-get install --reinstall gcc`. 

Anyway, I'm afraid the problem is not directly related to MRPT, so I'll close the ticket, ok? You probably could find a better guess of how to solve your problem by asking in StackOverflow or some other more general forum. In any case, remember to tell what has taken you to this point: you probably manually set/installed/... some compiler packages, and that must be the key to what was broken...

Cheers.

closed 2018-12-16 by jlblancoc