MRPT/mrpt #882: compile error of “could not find module FindMRPT.cmake ”

hamburgerBear, 2018-12-15:
i use ubuntu12.04 
        mrpt1.2.1

i make source code is ok,
but try to make some example code (as: a_starAlgorithm ) is error,
please help to me, thanks advantage
error info is follow:

CMake Error at CMakeLists.txt:34 (FIND_PACKAGE):
Could not find module FindMRPT.cmake or a configuration file for package
MRPT.
Adjust CMAKE_MODULE_PATH to find FindMRPT.cmake or set MRPT_DIR to the
directory containing a CMake configuration file for MRPT.  The file will
have one of the following names:

MRPTConfig.cmake
mrpt-config.cmake

-- Configuring incomplete, errors occurred!
ubuntu12@ubuntu:~/mrpt-1.2.1/samples/a_starAlgorithm/build$ ^C

jlblancoc, 2018-12-16:
Try opening `cmake-gui` and set the variable `MRPT_DIR` to the directory in which you built MRPT, that is, the one that must have: a `Makefile` file, and also a `MRPTConfig.cmake` file.

jlblancoc, 2019-01-03:
Closing due to lack of feedback. Please, reopen if needed.

closed 2019-01-03 by jlblancoc