MRPT/mrpt #892: CMake ERROR on INTERFACE_INCLUDE_DIRECTORIES

MariotoA, 2019-01-16:
Greetings,

Working on Ubuntu 16.04/Xenial
Been trying to build ROS package mrpt_bridge to work with MRPT 1.9.9 fixing CMakelists.txt as indicated [here (mrpt from cmake link)](http://mrpt.ual.es/reference/master/mrpt_from_cmake.html).
MRPT installed from ppa, having gcc7.

Encountered this error:
```
CMake Error in mrpt_bridge/CMakeLists.txt:
  Imported target "mrpt-obs" includes non-existent path

    "/build/mrpt-XSHrkz/mrpt-1.9.9~snapshot20190115-1920-git-e44b10ec-xenial/debian/usr/libmrpt-common-dev/usr/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.
```

Feel free to close this issue if you think is unrelated, but maybe it has to do with the discussion on #870 ?

jlblancoc, 2019-01-16:
Hi! 
Yes, it seems a related problem, though I believed it was solved... somehow might be back again. 
Anyway, just earlier today I proposed one more change to the way user programs (e.g. ROS packages) can find MRPT, via #891. 
After all tests pass and it's merged, we should revisit the way ROS nodes find MRPT, since now it should be much easier: a unique version like: 

```
# One may mix now mrpt 1.5 and 2.0 module names together 
# thanks to the "OPTIONAL_COMPONENTS" part:
find_package(MRPT REQUIRED COMPONENTS nav slam OPTIONAL_COMPONENTS core base)

target_link_libraries(my_app ${MRPT_LIBRARIES})
```

PRs to ROS repos will be welcome! ;-)

PS: Look a the version of the docs you linked, but for the PR branch [here](http://mrpt.ual.es/reference/cmake-mrpt-lib-namespace/mrpt_from_cmake.html).

MariotoA, 2019-01-17:
Ok, I will work with version 1.5 then and will try the upgrade when the time comes.

Really grateful for your response.

jlblancoc, 2019-01-18:
Should be fixed now after https://github.com/MRPT/mrpt/commit/8923218d7302e829f92e42c7488da4de0d04decb 

Let's wait for the PPA builds... and please, reopen if it's not fixed! ;-)

MariotoA, 2019-01-21:
Hi,

For now, it seems like the files mrpt-{$dep}-config.cmake are no longer included in ppa installation.
I'm no cmake expert but I think this causes cmake to raise an error when processing mrpt-config.cmake (since this one calls find_dependency for components specified)

```
include(CMakeFindDependencyMacro)

set(MRPT_LIBRARIES "")
foreach(_comp ${MRPT_FIND_COMPONENTS})
  if (MRPT_FIND_REQUIRED_${_comp})
    find_dependency(mrpt-${_comp}) # This one
  else()
    find_package(mrpt-${_comp} CONFIG QUIET)
  endif()
  if (mrpt-${_comp}_FOUND)
    list(APPEND MRPT_LIBRARIES mrpt::${_comp})
  endif()
endforeach()

# backwards-compatibility var name:
set(MRPT_LIBS ${MRPT_LIBRARIES})
```

jlblancoc, 2019-01-23:
Hi, 

It seems a recent change was causing this failure... it has just been undone with 46f5e92df12cc07fd0f51b3f6b960a68a4ee5523

Let's see if after the PPA processes the update (will take just a few hours) things get back to a working!

jlblancoc, 2019-02-14:
Is anyone still having problems with PPA builds and CMake configurations?

jlblancoc, 2019-02-20:
Seems to be fixed.

closed 2019-02-20 by jlblancoc