MRPT/mrpt #897: Dataset Download link broken

speed8928, 2019-01-30:
Hi 
The [dataset](https://www.mrpt.org/MalagaUrbanDataset) seems broken, could anyone provide other repository to download? Thank you in advance!

jlblancoc, 2019-01-30:
Yep, we are having problems with our server in the data center, and won't be fixed until next week. 

In the meanwhile, I'm uploading the entire dataset to an alternative location: 
https://drive.google.com/drive/folders/0B5HPWlmxIrvFOGFfUkIxVkhyWEE?usp=sharing

The upload will end by a couple of hours from now...

Cheers!

jlblancoc, 2019-01-30:
Added a note on [the web](https://www.mrpt.org/MalagaUrbanDataset) for the alternative download site for use during the next few days while we provide a permanent fix. 

Cheers.

closed 2019-01-30 by jlblancoc