MRPT/mrpt #898: CMake Error at cmakemodules/DeclareMRPTLib.cmake:187 (add_library)

jiapei100, 2019-02-01:
Please any suggestions?

I've got this issue:

```c++
# JLBC Dec 2018: On special install paths for Debian pkg prefixes
# After many hours debugging how to correctly export multiple
# CMake targets in one source package, each to a different debian pkg, I found
# this to be the only working solution:
# * Do NOT add module prefixes to installation paths. For example, this used
#   to be:
#  if(CMAKE_MRPT_USE_DEB_POSTFIXS)
#  	set(this_lib_dev_INSTALL_PREFIX "libmrpt-${name}-dev/usr/")
#   set(this_lib_bin_INSTALL_PREFIX "libmrpt-${name}${CMAKE_MRPT_VERSION_NUMBER_MAJOR}.${CMAKE_MRPT_VERSION_NUMBER_MINOR}/usr/")
#  ...
#  and it was great. But... then CMake xxx-target.cmake files would include
#  "libxxx-dev/" prefix and the user's cmake find_package() would be folish
#  would fail to find the packages.
# * Add, to debian/rules, bash commands to distribute each .so* to its proper
#   place. Even if it seems easier to directly install them to their place
#   from this script, the note above explains why that's incompatible with
#   cmake-exported targets.
```

```console
CMake Error at cmakemodules/DeclareMRPTLib.cmake:187 (add_library):
   Target "maps" links to target "hdf5::hdf5" but the target was not found.
   Perhaps a find_package() call is missing for an IMPORTED target, or an
   ALIAS target is missing?
 Call Stack (most recent call first):
   cmakemodules/DeclareMRPTLib.cmake:47 (internal_define_mrpt_lib)
   libs/maps/CMakeLists.txt:21 (define_mrpt_lib)

 CMake Error at cmakemodules/DeclareMRPTLib.cmake:187 (add_library):
   Target "maps" links to target "hdf5::hdf5_hl" but the target was not found.
   Perhaps a find_package() call is missing for an IMPORTED target, or an
   ALIAS target is missing?
 Call Stack (most recent call first):
   cmakemodules/DeclareMRPTLib.cmake:47 (internal_define_mrpt_lib)
   libs/maps/CMakeLists.txt:21 (define_mrpt_lib)

 CMake Error at cmakemodules/DeclareMRPTLib.cmake:187 (add_library):
   Target "maps" links to target "hdf5::hdf5" but the target was not found.
   Perhaps a find_package() call is missing for an IMPORTED target, or an
   ALIAS target is missing?
 Call Stack (most recent call first):
   cmakemodules/DeclareMRPTLib.cmake:47 (internal_define_mrpt_lib)
   libs/maps/CMakeLists.txt:21 (define_mrpt_lib)

 CMake Error at cmakemodules/DeclareMRPTLib.cmake:187 (add_library):
   Target "maps" links to target "hdf5::hdf5_hl" but the target was not found.
   Perhaps a find_package() call is missing for an IMPORTED target, or an
   ALIAS target is missing?
 Call Stack (most recent call first):
   cmakemodules/DeclareMRPTLib.cmake:47 (internal_define_mrpt_lib)
   libs/maps/CMakeLists.txt:21 (define_mrpt_lib)
```

jlblancoc, 2019-02-08:
I'm curious about where does hdf5 comes from :-) 
Please, attach your CMakeCache.txt and/or the output of `VERBOSE=1 cmake .` on your build directory.

jlblancoc, 2019-02-14:
Closing due to lack of feedback... please reopen if still needed!

closed 2019-02-14 by jlblancoc