MRPT/mrpt #906: Possible bug using CReactiveNavigationSystem3D

MariotoA, 2019-03-18:
Hi,
Just wanted to point out a bug I keep finding lately.

It seems when starting 3D reactive navigation and [CReactiveNavigationSystem3D::checkCollisionWithLatestObstacles](https://github.com/MRPT/mrpt/blob/master/libs/nav/src/reactive/CReactiveNavigationSystem3D.cpp#L343) is called (this not always happens) it causes a segFault. 

Looking at my gdb backtrace, this seems to happen when the goal is seemingly surrounded by obstacles (I can tell looking at my maps that this may be the case). The previous method is called because[ this line](https://github.com/MRPT/mrpt/blob/b37747e347a3e754664813369330f2cf9e583cfa/libs/nav/src/reactive/CAbstractNavigator.cpp#L620). 

I tried to change the parameter **dist_check_target_is_blocked** in the ".ini" files to be a negative number and this solves the issue for me. However, I'm not sure if this will cause more problems down the line. Just to be clear, for some reason, **this only happens when the first goal is sent to the system**.

Anyway, I'm running a ROS wrapper for the reactive 3d nav similar to [(you can find it here)](https://github.com/MariotoA/mrpt_reactivenav), but I do not think this is the cause since I followed the 2d navigation wrapper on mrpt_navigation stack.

I have encountered this on both mrpt 1.5 and master branch.

Greetings.

jlblancoc, 2019-03-18:
Interesting... I intended to release v1.5.7 this week to focus on the launch of 2.0.0-alpha, but we must look at this first now! 
Adding this one as a blocker to 1.5.7....

jlblancoc, 2019-04-24:
This should be fixed now after the two patches (mrpt-1.5 & master) referenced above. 
If the error persists, please reopen this. 
Cheers.

closed 2019-04-24 by jlblancoc