MRPT/mrpt #909: mrpt_ekf_slam_2d does not associate landmarks with yaw angle outside [90, -90]

kwahl1, 2019-03-25:
I am running mrpt_ekf_slam_2d on an indoor robot equipped with an RPLidar. Landmarks are fed to the ekf by running a clustering algorithm on the lidar pointcloud.

I am having a problem that the ekf does not seem to associate observations with landmarks located behind the robot, that is with a yaw angle outside the interval [90, -90]. Those observation are simply just added as new landmarks. Landmarks are mapped and associated correctly when in front of the robot, but as soon as they end up behind it, the ekf spams out multiple landmarks: (https://user-images.githubusercontent.com/32065960/54906042-b0f60780-4ee2-11e9-8000-fa7a31918f33.png)

Here is the result of the same run when filtering the lidar scan to only include points within [90,-90]: (https://user-images.githubusercontent.com/32065960/54906043-b18e9e00-4ee2-11e9-9a7f-ca519aa52bc1.png)

The model settings I use are:
std_sensor_range = 0.05
std_sensor_yaw_deg = 0.5
data_assoc_method  = assocNN 
data_assoc_metric = metricMaha
data_assoc_IC_chi2_thres  = 0.98 
data_assoc_IC_metric  = metricMaha
data_assoc_IC_ml_threshold  = 0.0 

I noticed that if I heavily increase the sensor noise (to around 0.5 m and 10 deg), the interval where landmarks are associated correctly is extended, but still spams out landmarks close to +-180, and with the cost of a lot higher uncertainties.

Here you can find two ROS bagfiles for testing, one containing landmarks from the full laser scan and one with landmarks from the filtered scan, along with transforms from odom->base_link:
https://drive.google.com/drive/folders/1y8817WdiMivd0N3qiur3MVloTvpS3nqW?usp=sharing

jlblancoc, 2019-03-28:
Hi, 

I checked this issue first by running the standalone app `2d-slam-demo` with a simulated sensor with a FOV of 330 degrees: 

[image]

[image]

and it seems that the problem does not lie within the general mrpt-slam class, since it (almost 100%) correctly solves data-association... 

So, the next thing to check is whether there's something wrong in the handling of messages / data structures in the ROS wrapper...

jlblancoc, 2019-03-29:
@kwahl1 Have you tried with mrpt_ekf_slam_3d instead of the 2d version? (just a hunch...)

jlblancoc, 2019-03-29:
Also, from the rosbag: 

`
...
min_sensor_distance: 0.3
max_sensor_distance: 5.0
sensor_std_range: 0.0005
sensor_std_yaw: 10.0
sensor_std_pitch: 0.0
sensed_data: 
...
`

This `sensor_std_range: 0.0005` is just too small! Is that an error? Try increasing it and things will most likely work much better...

kwahl1, 2019-03-29:
@jlblancoc  I have not tried mrpt_ekf_slam_3d but I could try that.

The sensor noise sensor_std_range etc is set in the .ini file under the [RangeBearingKFSLAM] section, to the values I wrote in my first post. As I understand it the ekf only takes in those values from the ini file, not from the ObservationRangeBearing message. Nothing seems to change regardless of what a use for noise settings on the ObservationRangeBearing message.

Logrus, 2019-03-30:
@kwahl1, do you actually use a ROS package or mrpt directly? I have recently started to rework some of the packages, so I could have a look into that, but I didn't realize it might be related since it's posted in mrpt repo, not mrpt_slam with ros wrappers.

jlblancoc, 2019-03-30:
@Logrus yes, the problem happens in the ros package, but following his post in stackoverflow I recommended him to open this issue here, just in case the problem lied within mrpt-slam itself... it **seems** that it's not the case, although until the problem is 100% solved we won't know for sure.
It would be great if you could re-check the related code within the ekf2d ros package, add some traces, etc. just to make sure that there is nothing weird. 
Feel free of coming back here to ask any doubt about the internals of the EKF implementation, if needed.

jlblancoc, 2019-10-06:
@kwahl1 I tried to reproduce this bug again today. It happens in the ROS node with your supplied bag files, but it doesn't in the 2d-slam-demo app in simulation, no matter where landmarks are placed.

After playing a little bit with uncertainties and parameters, I think I might have reached a possible explanation: you should enlarge your odometry and sensor uncertainties (i.e. how uncertain EKF **thinks** measurements are) with respect to the real noise. Use from 120% to 200% of what you think is the real noise as EKF uncertainty for both, odometry and sensor readings. 
I tested it and even with your data, one has way less false negatives. 

Cheers.

closed 2019-10-06 by jlblancoc

kwahl1, 2019-10-12:
Hi @jlblancoc. Yes heavily increased uncertainties kind of solves the problem. I found the best performance around std_sensor_range =0.4 and std_sensor_yaw_deg=10, which I think are unreasonably large..
Anyway I ended up implementing my own EKF SLAM following the algorithm in Probabilistic Robotics in which I did not encounter this problem anymore. The project was part of my masters thesis which is now finished. 

Thanks for looking into this.

jlblancoc, 2019-10-12:
Cool! Glad you made it. 
Cheers