MRPT/mrpt #910: fix wx-config --toolkit=gtk3  problem

RubyLai, 2019-04-08:
* problem:
$ cmake ..
-- Kernel name (uname -s): Linux
Warning: No config found to match: /usr/bin/wx-config --selected-config --toolkit=gtk3                                                                                                            
                   in /usr/lib/x86_64-linux-gnu/wx/config                                                                                                                                                   
          If you require this configuration, please install the desired                                                                                                                                     
          library build.  If this is part of an automated configuration                                                                                                                                     
          test and no other errors occur, you may safely ignore it.                                                                                                                                         
          You may use wx-config --list to see all configs available in                                                                                                                                      
          the default prefix.  
          ...

$ wx-config --list
    Default config is gtk2-unicode-3.0
  Default config will be used for output
  Alternate matches:
    base-unicode-3.0

* slove:
$ sudo apt install libwxgtk3.0-gtk3-dev
$ wx-config --list
    Default config is gtk2-unicode-3.0
  Default config will be used for output
  Alternate matches:
    base-unicode-3.0
    gtk3-unicode-3.0

$ cmake ..                                                                                               
-- Kernel name (uname -s): Linux                                 
-- wxWidgets: Found gtk3 version, using it.            
-- Found wxWidgets: -L/usr/lib/x86_64-linux-gnu;-pthread;;;-lwx_baseu-3.0;-lwx_gtk3u_core-3.0;-lwx_gtk3u_gl-3.0;-lwx_gtk3u_adv-3.0;-lwx_gtk3u_aui-3.0;-lwx_gtk3u_html-3.0 (found version "3.0.4")

RubyLai, 2019-04-08:
https://github.com/MRPT/mrpt#32-build-from-sources
README.md should add
```bash
$ sudo apt install libwxgtk3.0-gtk3-dev
```

jlblancoc, 2019-04-09:
Done, thanks! 

If you want to propose further improvements, feel free of directly editing + proposing a pull-request.

closed 2019-04-09 by jlblancoc

dibet, 2019-08-19:
Hi, i have here this error

jolting, 2019-08-19:
I think you need libwxgtk3.0-dev for xenial.

brunoeducsantos, 2020-08-25:
@jolting  I install that package and keep having the same error. Any suggestions?
thanks

jlblancoc, 2020-08-25:
@brunoeducsantos this was (and keeps being for the current `develop` version) only a warning, mrpt should builds against older versions of wxWidgets such as Xenial's... let us know if you actually find an error related to this issue.

commit 9cf1d484 (2019-04-09): Add wx-gtk3 apt instructions

Closes #910