MRPT/mrpt #913: this project has asked CMake to find a package configuration file provided by "mrpt-base", but CMake did not find one.

TheodoreT, 2019-04-17:
Hi guys,

I am trying to compile this project here https://github.com/raluca-scona/staticfusion which has as a dependency the mrpt library. I have installed the mrpt lib based on your instructions and by adding the corresponding repository:

```
    sudo add-apt-repository ppa:joseluisblancoc/mrpt
    sudo apt-get update
    sudo apt-get install libmrpt-dev mrpt-apps
```
then I had the problem that it couldn't locate the `libmrpt-base` which I solved by installing the `libmrpt-base1.9` package. However, now I am getting the following error:
```

cmake ..
-- The C compiler identification is GNU 4.9.3
-- The CXX compiler identification is GNU 4.9.3
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_MODULE_PATH:   /home/ciates-dev07/theo/staticfusion/cmake_modules
CMake Error at /usr/share/cmake-3.5/Modules/CMakeFindDependencyMacro.cmake:65 (find_package):
  By not providing "Findmrpt-base.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "mrpt-base", but CMake did not find one.

  Could not find a package configuration file provided by "mrpt-base" with
  any of the following names:

    mrpt-baseConfig.cmake
    mrpt-base-config.cmake

  Add the installation prefix of "mrpt-base" to CMAKE_PREFIX_PATH or set
  "mrpt-base_DIR" to a directory containing one of the above files.  If
  "mrpt-base" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  /usr/share/mrpt/mrpt-config.cmake:26 (find_dependency)
  CMakeLists.txt:16 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/ciates-dev07/theo/staticfusion/build/CMakeFiles/CMakeOutput.log".
```

where it tries to find the `mrpt-baseConfig.cmake` or `mrpt-base-config.cmake` files but these files are not installed anywhere. Thus, any hints how to solve this? I believe that it is mrpt related issue that's why I ask here but please correct me if I am wrong.

My system is ubuntu 16.04.

Thanks.

jlblancoc, 2019-04-17:
Hi, 
It seems that the main problem is that your CMakeLists.txt script is mixing two MRPT library conventions: those for mrpt-1.5.x and mrpt >= 1.9.9. The PPA repo installs the latest version of mrpt >= 1.9.9, where mrpt-base does no longer exists. 
It has been split into many smaller libraries. 
Please, see: http://mrpt.ual.es/reference/devel/mrpt_from_cmake.html
And this list of existing modules in mrpt >= 1.9.9 here: http://mrpt.ual.es/reference/devel/index.html

Cheers.

closed 2019-04-17 by jlblancoc

TheodoreT, 2019-04-18:
thanks @jlblancoc yes indeed using the 1.5.x version instead of the latest 1.9.9 solved the linking problem

OscarMind, 2019-11-17:
> thanks @jlblancoc yes indeed using the 1.5.x version instead of the latest 1.9.9 solved the linking problem

but...  what's your Specific steps for "  using the 1.5.x version instead of the latest 1.9.9 "
thx.

OscarMind, 2019-11-17:
> Hi,
> It seems that the main problem is that your CMakeLists.txt script is mixing two MRPT library conventions: those for mrpt-1.5.x and mrpt >= 1.9.9. The PPA repo installs the latest version of mrpt [...]

How can I change the  mrpt >= 1.9.9 by mrpt-1.5.x ?

TheodoreT, 2019-11-18:
check [here ](https://github.com/MRPT/mrpt/tree/mrpt-1.5#31-ubuntu-ppa)