MRPT/mrpt #914: Fix ROS melodic package for mrpt1

jlblancoc, 2019-05-01:
It seems there's a problem with dpkg-source: 

```
dpkg-source: warning: ignoring deletion of file doc/html/pnp_perf_comp.html, use --include-removal to override
```

mrpt1 is [going to be removed](https://github.com/ros/rosdistro/pull/21041#issuecomment-488327929) from ros melodic for now because of this.

jolting, 2019-05-01:
Is this an Artful thing?
https://github.com/mrpt-ros-pkg-release/mrpt1-release/blob/debian/melodic/artful/mrpt1/doc/html/pnp_perf_comp.html

jolting, 2019-05-01:
http://build.ros.org/job/Msrc_uB__mrpt1__ubuntu_bionic__source/lastFailedBuild/console
```
01:13:20 dpkg-source: error: cannot represent change to tests/test-3d-obs-ground.rawlog: binary file contents changed
01:13:20 dpkg-source: error: add tests/test-3d-obs-ground.rawlog in debian/source/include-binaries if you want to store the modified binary in the debian tarball
01:13:20 dpkg-source: error: unrepresentable changes to source
01:13:20 dpkg-buildpackage: error: dpkg-source -i -I -b source subprocess returned exit status 2
```

jolting, 2019-05-01:
The orig tarball was generated the first time 1.5.7 was built.
I guess that was a pre-release.
http://repositories.ros.org/ubuntu/building/pool/main/r/ros-melodic-mrpt1/ros-melodic-mrpt1_1.5.7.orig.tar.gz

The new builds are still fetching this original tarball.

jlblancoc, 2019-05-03:
Thanks for taking a look! 
Hmm... if the orig tarball is the one generated by bloom-release, it's probably "correct" despite the name: it always asks whether it should add "--force" when doing "git push --tags" after each new release, probably because of the weird configuration of version numbers.

But it's been working like that w/o problems for 1+ year, and the same version seems to work in Kinetic...  Needs further investigation, but have no time for this now! To the queue... :-)

jolting, 2019-05-03:
The orig tarball was generated in this build
http://build.ros.org/job/Msrc_uB__mrpt1__ubuntu_bionic__source/1/

It will be used by all subsequent builds. Debian is quite picky about not modifying the orig tarball. Any changes there will have to be applied as patches.

My guess is that it was created during the first 1.5.7 build from the ros branch.
https://github.com/MRPT/mrpt/tree/1.5.7-2018-06-11-ros-mrpt1

It treated that as the one and only canonical release of 1.5.7 when it did that first build and generated the orig file which now appears to be immutable. Clearly this wasn't intended to be an official release of 1.5.7, but the ROS build server doesn't know that. It will continue to use that as the original version of the tarball for 1.5.7 builds. I'm not sure if there is a way to clear that out. Probably best to just bump the MRPT version number to 1.5.8, so it's not conflicting any more. Perhaps it's worth burning an MRPT version number for consistency.

jlblancoc, 2019-05-04:
Great idea! Thanks :+1:

Done that... bumped to 1.5.8 and PR against rosdistro. 
https://github.com/ros/rosdistro/pull/21081

closed 2019-05-04 by jlblancoc