MRPT/mrpt #918: Blank webcam image on Windows 10

ahtur, 2019-05-09:
Hi,
I have a Logitech webcam that I read regularly in VLC, VirtualDub and in my own code via OpenCV/DSHOW. Whenever I try to open it in the mrpt's calib-camera, I see its LED lighting up, proof that the software has taken control of it, but the UI frame meant to show the images is always completely black. This seems to occur with both x64 and win32 builds of mrpt 1.5.5 to 1.5.7. I didn't really test older versions. Are you sure that calib-camera is able to open local cameras on Windows 10 ?

jlblancoc, 2019-05-10:
Hi, 

I debugged into this and also find problems opening the laptop camera from OpenCV under Windows 10 (it works well in Linux). 
I nailed down to this issue, at least in my case (I cannot even get to a "blank webcam image" as you): 
https://stackoverflow.com/questions/38878249/opencv-camera-cant-get-a-video-decompresser?rq=1#

If possible, give it a try to the master branch of MRPT, by building it from sources.

ahtur, 2019-05-10:
Hi Jose Luis. I'm in a office without Internet access (I'm answering from my mobile now) so I can't build mrpt from source. Could you possibly issue a x64 build that works for you ?

ahtur, 2019-05-14:
Hi Jose Luis. I've rebuilt camera-calib on Windows 10, linked against opencv 2.4 and it has been able to load the webcam out of the box, without any code modification. Do you think you can do the same and release a working build ?

jlblancoc, 2019-05-14:
Good! Hmm... Some doubts: 
* what version of mrpt did you rebuild? 1.5.8, right?
* Did you also build opencv from sources or did you installed them from a binary download?
* Could you please try rebuilding against opencv 3.x just to see if it's a regression introduced by using a newer opencv version?

ahtur, 2019-05-14:
>     * what version of mrpt did you rebuild? 1.5.8, right?
> 
1.5.7
>     * Did you also build opencv from sources or did you installed them from a binary download?
>
I had to rebuild it because their build didn't have all the libraries requested by mrpt.
 
>     * Could you please try rebuilding against opencv 3.x just to see if it's a regression introduced by using a newer opencv version?
camera-calib does build against opencv 3.4 but, after that, I think opencv 4 loses the C API.

jlblancoc, 2019-10-10:
Grabbing from a camera (for example from camera-calib) is verified to work against opencv 4.1.1 with mrpt "develop", so I'll mark this one as "fixed". If the problem still exists for mrpt-1.5, it would be "won't fix" since we'll focus on mrpt2.

closed 2019-10-10 by jlblancoc