MRPT/mrpt #923: MacOS link errors

jlblancoc, 2019-05-29:
It FTBFS in CircleCI. 

Things to try: 
- https://stackoverflow.com/a/53040873/1631514
- https://stackoverflow.com/a/16353112/1631514

jolting, 2019-05-30:
Weird. The compiler should treat that template as ODR-used.
```
template<class _CharT, class _Traits, class _Allocator>
basic_ostream<_CharT, _Traits>&
operator<<(basic_ostream<_CharT, _Traits>& __os,
           const basic_string<_CharT, _Traits, _Allocator>& __str)
{
    return _VSTD::__put_character_sequence(__os, __str.data(), __str.size());
}
```
I'm a bit surprised it's not inlining that function and calling `__put_character_sequence` directly.

jlblancoc, 2019-06-21:
This is still the only failure on CircleCI.
Probably, the best way to fix it is to find the exact commit that started to fail and carefully analyze it (sigh)...

jolting, 2019-06-21:
https://github.com/MRPT/mrpt/commit/0cbd317d8b1aeec30c525710321ef1657f22b563

jolting, 2019-06-21:
This is probably the culprit.
https://github.com/MRPT/mrpt/commit/e2f5f387bbd7bc4e59d272f9d762e926bdfa7bef

jolting, 2019-06-21:
I'm wondering if the `mrpt::aligned_std_basicstring<T>` in CMatrixDynamic is causing the problem.
Why isn't that just a vector?

jlblancoc, 2019-06-21:
Thanks for searching it! So it was the big Eigen refactoring... 

> Why isn't that just a vector?

I know it should be proven by benchmarking but... it's to exploit the small string optimization. Small dynamic matrices / vectors should still fit on it so only stack is used, one less malloc/free pair.

jlblancoc, 2019-06-22:
Tried to explicitly specify -stdlib in #930 ... no luck.

jolting, 2019-06-22:
See if this works, https://github.com/MRPT/mrpt/tree/small_vector_optimization

jlblancoc, 2019-06-22:
Wow, just brilliant! :-)

I guess that the OSX problem is not related at all with this type, but anyway if your patch builds, we could factor it out to its own header and use it. It's better to know for sure **what** is the small size, and not being "C++ library implementation dependant".

jolting, 2019-06-22:
Still worth benchmarking.

jolting, 2019-06-22:
Didn't fix the mac error.

jolting, 2019-06-23:
Closed with #932

closed 2019-06-23 by jolting