MRPT/mrpt #926: /mrpt-0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3/libs/vision/src/sift-hess/imgfeatures.c:342:19: error:

dengzhiwei12, 2019-06-07:
when I compile the mrpt ,there are some problems I can solve, can someone can tell me how solve it?
here is the message:

[ 41%] Building CXX object apps/carmen2rawlog/CMakeFiles/carmen2rawlog.dir/carmen2rawlog_main.cpp.o
[ 41%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/COccupancyGridMap2D_io.cpp.o
[ 42%] Linking CXX executable ../../bin/carmen2rawlog
[ 42%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/CReflectivityGridMap2D.cpp.o
[ 42%] Built target carmen2rawlog
[ 42%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/CSimplePointsMap.cpp.o
[ 42%] Building CXX object libs/vision/CMakeFiles/mrpt-vision.dir/src/obs/CObservationVisualLandmarks.cpp.o
[ 42%] Building C object libs/vision/CMakeFiles/mrpt-vision.dir/src/sift-hess/imgfeatures.c.o
/home/dzw/plslam/mrpt-0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3/libs/vision/src/sift-hess/imgfeatures.c: In function ‘draw_oxfd_features’:
/home/dzw/plslam/mrpt-0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3/libs/vision/src/sift-hess/imgfeatures.c:342:19: warning: implicit declaration of function ‘CV_RGB’ [-Wimplicit-function-declaration]
  CvScalar color = CV_RGB( 255, 255, 255 );
                   ^
/home/dzw/plslam/mrpt-0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3/libs/vision/src/sift-hess/imgfeatures.c:342:19: error: invalid initializer
/home/dzw/plslam/mrpt-0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3/libs/vision/src/sift-hess/imgfeatures.c:346:9: error: incompatible types when assigning to type ‘CvScalar {aka struct CvScalar}’ from type ‘int’
   color = FEATURE_OXFD_COLOR;
         ^
/home/dzw/plslam/mrpt-0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3/libs/vision/src/sift-hess/imgfeatures.c: In function ‘draw_oxfd_feature’:
/home/dzw/plslam/mrpt-0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3/libs/vision/src/sift-hess/imgfeatures.c:383:13: error: incompatible type for argument 7 of ‘cvEllipse’
     0, 360, CV_RGB(0,0,0), 3, 8, 0 );
             ^
In file included from /usr/local/include/opencv2/highgui/highgui_c.h:46:0,
                 from /home/dzw/plslam/mrpt-0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3/libs/base/include/mrpt/otherlibs/do_opencv_includes.h:28,
                 from /home/dzw/plslam/mrpt-0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3/libs/vision/src/sift-hess/utils.h:15,
                 from /home/dzw/plslam/mrpt-0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3/libs/vision/src/sift-hess/imgfeatures.c:10:
/usr/local/include/opencv2/imgproc/imgproc_c.h:1030:14: note: expected ‘CvScalar {aka struct CvScalar}’ but argument is of type ‘int’
 CVAPI(void)  cvEllipse( CvArr* img, CvPoint center, CvSize axes,
              ^
/home/dzw/plslam/mrpt-0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3/libs/vision/src/sift-hess/imgfeatures.c: In function ‘draw_lowe_features’:
/home/dzw/plslam/mrpt-0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3/libs/vision/src/sift-hess/imgfeatures.c:553:19: error: invalid initializer
  CvScalar color = CV_RGB( 255, 255, 255 );
                   ^
/home/dzw/plslam/mrpt-0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3/libs/vision/src/sift-hess/imgfeatures.c:557:9: error: incompatible types when assigning to type ‘CvScalar {aka struct CvScalar}’ from type ‘int’
   color = FEATURE_LOWE_COLOR;
         ^
libs/vision/CMakeFiles/mrpt-vision.dir/build.make:134: recipe for target 'libs/vision/CMakeFiles/mrpt-vision.dir/src/sift-hess/imgfeatures.c.o' failed
make[2]: *** [libs/vision/CMakeFiles/mrpt-vision.dir/src/sift-hess/imgfeatures.c.o] Error 1
make[2]: *** 正在等待未完成的任务....
Scanning dependencies of target carmen2simplemap
[ 42%] Building CXX object apps/carmen2simplemap/CMakeFiles/carmen2simplemap.dir/carmen2simplemap_main.cpp.o
[ 42%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/CColouredOctoMap.cpp.o
CMakeFiles/Makefile2:496: recipe for target 'libs/vision/CMakeFiles/mrpt-vision.dir/all' failed
make[1]: *** [libs/vision/CMakeFiles/mrpt-vision.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
[ 43%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/CPointsMap.cpp.o
[ 43%] Linking CXX executable ../../bin/carmen2simplemap
[ 43%] Built target carmen2simplemap
[ 43%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/COccupancyGridMap2D_insert.cpp.o
[ 43%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/CBeaconMap.cpp.o
[ 43%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/CHeightGridMap2D_MRF.cpp.o
/home/dzw/plslam/mrpt-0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3/libs/maps/src/maps/CHeightGridMap2D_MRF.cpp: In member function ‘virtual bool mrpt::maps::CHeightGridMap2D_MRF::insertIndividualPoint(double, double, double, const mrpt::maps::CHeightGridMap2D_Base::TPointInsertParams&)’:
/home/dzw/plslam/mrpt-0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3/libs/maps/src/maps/CHeightGridMap2D_MRF.cpp:102:70: note: #pragma message: TODO: insertIndividualPoint: allow different std values
[ 43%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/CColouredPointsMap.cpp.o
/home/dzw/plslam/mrpt-0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3/libs/maps/src/maps/CPointsMap.cpp:66:18: warning: ‘dumm’ defined but not used [-Wunused-variable]
 static const int dumm = mrpt_maps_class_reg.do_nothing(); // Avoid compiler removing this class in static linking
                  ^
[ 43%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/CBeacon.cpp.o
[ 43%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/CRandomFieldGridMap2D.cpp.o
/home/dzw/plslam/mrpt-0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3/libs/maps/src/maps/CRandomFieldGridMap2D.cpp: In member function ‘void mrpt::maps::CRandomFieldGridMap2D::updateMapEstimation_GMRF()’:
/home/dzw/plslam/mrpt-0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3/libs/maps/src/maps/CRandomFieldGridMap2D.cpp:2562:67: note: #pragma message: TODO: Use compressed access instead of coeff() below
[ 43%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/CHeightGridMap2D_Base.cpp.o
[ 43%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/COccupancyGridMap2D_simulate.cpp.o
[ 44%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/CGasConcentrationGridMap2D.cpp.o
[ 44%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/CHeightGridMap2D.cpp.o
[ 44%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/COccupancyGridMap2D_getAs.cpp.o
[ 44%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/COccupancyGridMap2D_common.cpp.o
[ 44%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/opengl/CAngularObservationMesh.cpp.o
[ 44%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/opengl/CPlanarLaserScan.cpp.o
[ 44%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/__/__/otherlibs/octomap/src/CountingOcTree.cpp.o
[ 44%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/__/__/otherlibs/octomap/src/OcTreeLUT.cpp.o
[ 44%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/__/__/otherlibs/octomap/src/OcTreeStamped.cpp.o
[ 44%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/__/__/otherlibs/octomap/src/OcTreeNode.cpp.o
[ 45%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/__/__/otherlibs/octomap/src/ColorOcTree.cpp.o
[ 45%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/__/__/otherlibs/octomap/src/Pointcloud.cpp.o
[ 45%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/__/__/otherlibs/octomap/src/AbstractOcTree.cpp.o
[ 45%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/__/__/otherlibs/octomap/src/AbstractOccupancyOcTree.cpp.o
[ 45%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/__/__/otherlibs/octomap/src/OcTree.cpp.o
[ 45%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/__/__/otherlibs/octomap/src/ScanGraph.cpp.o
[ 45%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/__/__/otherlibs/octomap/src/math/Quaternion.cpp.o
[ 45%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/__/__/otherlibs/octomap/src/math/Vector3.cpp.o
[ 45%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/__/__/otherlibs/octomap/src/math/Pose6D.cpp.o
[ 45%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/maps-precomp.cpp.o
[ 46%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/registerAllClasses.cpp.o
[ 46%] Linking CXX shared library ../../lib/libmrpt-maps.so
[ 46%] Built target mrpt-maps
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

jolting, 2019-06-07:
What version of OpenCV are you using? My best guess is that OpenCV wasn't installed correctly.

dengzhiwei12, 2019-06-08:
> What version of OpenCV are you using? My best guess is that OpenCV wasn't installed correctly.

I installed the OpenCV 3.4.6 and I also installed the opencv_contrib modules .The OpenCV was installed and I tested it, it works well, but I did't test the opencv_contrib, I just need it recently. I don't know if they are related.
I tried to compile the mrpt 1.5 and I was successful. Now I want to know the mrpt 1.5 are the version before mrpt--0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3 or after it. I want to run the PL_SLAM and its author recommended mrpt-0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3

jlblancoc, 2019-06-20:
May it be that you have two opencv versions? By "I installed" you really mean that you ran "sudo make install"? 
If yes, there are chances that your system have a mix of versions under /usr and /usr/local that is known to be problematic if not extremely careful with your CMake variables to ensure it detects the version you intend to use...

jlblancoc, 2019-06-27:
Closing, please reopen if needed.

closed 2019-06-27 by jlblancoc

SSAIW, 2022-10-10:
Could you tell me which version of OpenCV can be installed?

jlblancoc, 2022-10-11:
MRPT should build without issues against any version of OpenCV in the ranges 3.x and 4.x. It probably also still builds against v2.x. Please, open an issue ticket if you find any issue building against a modern version of opencv!