MRPT/mrpt #927: fix armhf builds

jlblancoc, 2019-06-14:
See, for example: 
https://launchpadlibrarian.net/428169778/buildlog_ubuntu-bionic-arm64.mrpt_1%3A1.9.9~snapshot20190614-0127-git-1ea03cd5-bionic-1~ppa1~bionic_BUILDING.txt.gz

To-do: observe `std::max_align_t` and use it as a limit to MRPT_MAX_STATIC_ALIGN_BYTES.

jolting, 2019-06-15:
Reminds me of this:
https://github.com/MRPT/mrpt/commit/8c633ad2416666b401ae9c8cec10c998eba796f8

jlblancoc, 2019-06-20:
> Reminds me of this:

Yep! But shouldn't apply here, after all the c++17 "automagically aligned" stuff... 

Make your bet: Will [this tiny b*st*rd](https://github.com/MRPT/mrpt/commit/8e0b0dd9cae6c8d32899f52cc64907a97118f549) fix it? :-)

jolting, 2019-06-20:
If it incidentally works then there are other issues with alignment with the copy constructor. The copies *should* be aligned.

jolting, 2019-06-20:
```For a Composite Type, the alignment of the copy will have 4-byte alignment if its natural alignment is
<= 4 and 8-byte alignment if its natural alignment is >= 8```
https://static.docs.arm.com/ihi0042/f/IHI0042F_aapcs.pdf
I think it only supports 8 byte alignment. I don't think there is a reason the have stricter alignment on arm64.

jolting, 2019-06-20:
That's an arm64 build.

jlblancoc, 2019-06-20:
Thanks for searching for the docs. 

So... it *did* work (!). [log](https://launchpad.net/~joseluisblancoc/+archive/ubuntu/mrpt/+build/17175662). 

One thing is: I don't know why passing by value (object on the stack) does not respect the alignment... Probably the reason is what you posted: for arm64, max_align_t seems to be 8. That explains these warnings: 

```
/<<PKGBUILDDIR>>/libs/math/src/TPoseOrPoint.cpp:320:63: warning: requested alignment 32 is larger than 8 [-Wattributes]
   alignas(MRPT_MAX_STATIC_ALIGN_BYTES) const double nums[4 * 3] = {
```

However, as part of the tests, I peek into Eigen's choice for alignment, and here's the values for arm64:

```
[ RUN      ] EigenAlignment.PrintAlignment
MRPT_MAX_ALIGN_BYTES         : 32
MRPT_MAX_STATIC_ALIGN_BYTES  : 32
EIGEN_MAX_ALIGN_BYTES        : 16
EIGEN_MAX_STATIC_ALIGN_BYTES : 16
[       OK ] EigenAlignment.PrintAlignment (0 ms)
```
Which is larger than 8 anyway (?).

In any case, I *think* all alignment issues could be considered as "fixed" now... perhaps not in the most efficient ways (due to some extra bytes lost for the 16->32 or 8->32 alignment in MRPT), but "good enough for now"...

closed 2019-06-20 by jlblancoc

jolting, 2019-06-21:
All the ARM and ARM64 vector instructions use registers. The only reason to overalign is to support the double word load/store instructions. In arm64 I guess that's 8 bytes and arm32 thats 4 bytes.

Intel over-aligns because those instructions have to align with cache boundaries.