MRPT/mrpt #940: RFC (Request for comments): New generic observation to unify 2D and 3D rotating LiDARs

jlblancoc, 2019-07-22:
To be merged before mrpt 2.0.0-beta. 
The WIP code is in this branch: https://github.com/MRPT/mrpt/commits/add-rotating-scan-obs

This issue is intended to be a public place of discussion for ideas on how to achieve having a container for raw range data (**not** point clouds) for (mainly) 3D and (also) 2D scans.
Right now we have a good container for 2D scans, and a 3D container for Velodyne scans only, but it would be strongly desirable to have a unified container for 3D scans from different vendors, especially thinking of LIDAR modules in the MOLA architecture. 

Please, check the code in the PR: XX where each data field in `mrpt::obs::CObservationRotatingScan.h ` has comments explaining the design criteria.
Please, comment here, or review the PR  with suggestions. 

Some main decisions explained:
- We must support having at least one "main" (strongest) range for each direction and each lidar ("different heights").
- Ranges are stored as structured images (matrices). For efficiency, it would be desired to store ranges as integers, to enable fast filters to run on "range images" as if they were regular images. However, using 8bit or 16 bit is convenient for some models (u8: good old SICK lidars; 16u: velodyne) but may render insufficient for future models, so I propose using a variant of matrix types.
- Apart of this range image, there may be an optional intensity image, which is assumed to always be of type u8 (I think it will never get short??).
- An optional std::map of alternative range images, e.g. first returns, N-th return, etc.
- Having GPS-based precise timing, 
- etc.

The area of calibration parameters is still open. I'll check the current ones for Velodyne, and those of Ouster, and try to reach to a set of parameters that fit, at least, those two models.

Thoughts?

(cc: @jolting )

jolting, 2019-07-22:
Posting here since it's more appropriate...

I'd prefer templating CObservationRotatingScan instead of using CRangeImageVariant. Variants can be inefficient. It violates the zero-overhead principle.

jolting, 2019-07-22:
Consider the following
```c++
#include <variant>
#include <iostream>

struct VectorFloat {
    float x;
    float y;
    float z;
};

struct VectorDouble {
    double x;
    double y;
    double z;
};
int main(int argc, char* argv[])
{
    auto print =[](auto &result){
        std::cout << result.x << " " << result.y << " " << result.z << std::endl; 
    } ;
#ifdef VARIANT
    using MyVariant = std::variant <VectorFloat, VectorDouble>;
    auto someRandomFunc = [](const auto &v)->MyVariant {
        return decltype(v){v.x+v.z,v.x*v.y*v.z,v.x+v.y+v.y};
    };
    MyVariant test1 = VectorFloat{1,2,3};
    MyVariant test2 = VectorDouble{1,2,3};
    auto result1 = std::visit(someRandomFunc, test1);
    auto result2 = std::visit(someRandomFunc, test2);
    std::visit(print, result1);
    std::visit(print, result2);
#else
    auto someRandomFunc = [](const auto &v) {
        return decltype(v){v.x+v.z,v.x*v.y*v.z,v.x+v.y+v.y};
    };
    auto test1 = VectorFloat{1,2,3};
    auto test2 = VectorDouble{1,2,3};
    auto result1 = someRandomFunc(test1);
    auto result2 = someRandomFunc(test2);
    print(result1);
    print(result2);
#endif
    return 0;
}
```
I prefer the non-variant version since it allows for better optimization and avoids the dynamic polymorphism. Finally, it's a bit easier to read, requiring fewer type conversions and avoiding the std::visit function.

If you go the MyVariant route you will be using MyVariant in many unexpected places.

jlblancoc, 2019-07-24:
:+1: convinced me ;-)

It's much simpler now. Let's rely on uint16_t only for range data (times a float multiplier for "resolution") and uint8_t for intensity.

jlblancoc, 2019-10-08:
Closed after #941

closed 2019-10-08 by jlblancoc