MRPT/mrpt #943: CMake No such file or directory

MariaRamos89, 2019-07-29:
Hi,

I built the library from source the past 25th of July. Everything compiled and I used the command `sudo make install`.

The problem is when I try to link it in my program. I use cmake for that and after reading [this](https://docs.mrpt.org/reference/devel/mrpt_from_cmake.html) my cmake file is in this way:

```
message(STATUS "Searching for MRPT")
find_package(MRPT REQUIRED COMPONENTS core nav maps slam poses)
if (NOT MRPT_LIBRARIES)
    message(FATAL_ERROR "MRPT not found")
else()
    message(STATUS "MRPT found")
endif()

target_link_libraries(myapp ${MRPT_LIBRARIES})
```

I've only added the `mrpt` part. 
When I run `cmake .. ` the library is found. Even if I use a message for showing `${MRPT_LIBRARIES}` the result is 
```
mrpt::core;mrpt::nav;mrpt::maps;mrpt::slam;mrpt::poses
```

However, when I do `make` the files are not found:
```
fatal error: mrpt/poses/CPose3D.h: No such file or directory
 #include <mrpt/poses/CPose3D.h>
fatal error: mrpt/slam/CMetricMapBuilderICP.h: No such file or directory
```

I've been looking if any `mrpt-xxx.config` was missing, but it seems that they are all in `/usr/local/share` like it should be.

I tried to do `target_link_libraries(myapp mrpt::poses)` for testing but the result is the same.
In previous versions I used `#include_directories(${MRPT_INCLUDE_DIRS})`, but it is suppossed is not neccessary any more.

Do you know what I am doing bad?

Thanks in advance

jlblancoc, 2019-07-30:
Hi María!

What's your OS and, especially, your version of cmake?

Also: show us the result of `VERBOSE=1 make`. 

Final advice: why are you using `make install`? Try to avoid that as much as possible, really!! 
Try to clean it up with `sudo make uninstall`, then build your program and set MRPT_DIR in cmake (using cmake-gui for example) pointing to the build directory of MRPT. It's much safer, otherwise it's pretty easy to end up running into library inconsistencies, old files left behind under `/usr`, etc. 

Cheers

MariaRamos89, 2019-07-30:
Hi jlblanco,

My OS is `Ubuntu 16.04.6 LTS`.
My cmake version is `cmake version 3.13.3`
Before doing `sudo make uninstall` I did `VERBOSE=1 make` and this is the result:

```
/usr/local/lib/python3.5/dist-packages/cmake/data/bin/cmake -E cmake_progress_start /home/ubuntu/Dev/robot/build/CMakeFiles /home/ubuntu/Dev/robot/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/ubuntu/Dev/robot/build'
make -f CMakeFiles/cloud.dir/build.make CMakeFiles/cloud.dir/depend
make[2]: Entering directory '/home/ubuntu/Dev/robot/build'
cd /home/ubuntu/Dev/robot/build && /usr/local/lib/python3.5/dist-packages/cmake/data/bin/cmake -E cmake_depends "Unix Makefiles" /home/ubuntu/Dev/robot /home/ubuntu/Dev/robot /home/ubuntu/Dev/robot_platform/build /home/ubuntu/Dev/robot_platform/build /home/ubuntu/Dev/robot/build/CMakeFiles/cloud.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/ubuntu/Dev/robot/build'
make -f CMakeFiles/cloud.dir/build.make CMakeFiles/cloud.dir/build
make[2]: Entering directory '/home/ubuntu/Dev/robot/build'
[  1%] Building CXX object CMakeFiles/cloud.dir/source/common/computer_vision.cpp.o
/usr/bin/c++  -DCAFFE_CUDA -I/home/ubuntu/Dev/robot/source -I/usr/local/cuda/include -I/usr/local/include -I/usr/local/include/opencv -I/home/ubuntu/Dev/robot/platform/src/vendors/httpxx/httpxx -I/usr/local/include/mongocxx/v_noabi -I/usr/local/include/bsoncxx/v_noabi -I/usr/include/eigen3  -std=c++17 -O3 -DNDEBUG -s -O3   -o CMakeFiles/cloud.dir/source/common/computer_vision.cpp.o -c /home/ubuntu/Dev/robot/source/common/computer_vision.cpp
In file included from /home/ubuntu/Dev/robot/source/common/computer_vision.hpp:3:0,
                 from /home/ubuntu/Dev/robot/source/common/computer_vision.cpp:1:
/home/ubuntu/Dev/robot/source/common/includes.ihh:20:10: fatal error: mrpt/poses/CPose3D.h: No such file or directory
 #include <mrpt/poses/CPose3D.h>
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/cloud.dir/build.make:62: recipe for target 'CMakeFiles/cloud.dir/source/common/computer_vision.cpp.o' failed
make[2]: *** [CMakeFiles/cloud.dir/source/common/computer_vision.cpp.o] Error 1
make[2]: Leaving directory '/home/ubuntu/Dev/robot/build'
CMakeFiles/Makefile2:368: recipe for target 'CMakeFiles/cloud.dir/all' failed
make[1]: *** [CMakeFiles/cloud.dir/all] Error 2
make[1]: Leaving directory '/home/ubuntu/Dev/robot/build'
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2
```

I am not using interface so I can't use `cmake-gui`.
So the first thing I've tried  is to export MRPT_DIR : `export MRPT_DIR=/my_path/mrpt/build`

```
CMake Error at /usr/local/lib/python3.5/dist-packages/cmake/data/share/cmake-3.13/Modules/CMakeFindDependencyMacro.cmake:48 (find_package):
  By not providing "Findmrpt-core.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "mrpt-core", but CMake did not find one.

  Could not find a package configuration file provided by "mrpt-core" with
  any of the following names:

    mrpt-coreConfig.cmake
    mrpt-core-config.cmake

  Add the installation prefix of "mrpt-core" to CMAKE_PREFIX_PATH or set
  "mrpt-core_DIR" to a directory containing one of the above files.  If
  "mrpt-core" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  /home/ubuntu/.libs/mrpt/build/mrpt-config.cmake:26 (find_dependency)
  cmake/dependencies.cmake:172 (find_package)
  CMakeLists.txt:35 (include)

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/Dev/robot_platform/build/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/Dev/robot_platform/build/CMakeFiles/CMakeError.log".
```

The second thing I tried is modifying my CMakeLists.txt in the following way:

```
find_package(MRPT REQUIRED COMPONENTS core nav maps slam poses PATHS ${MRPT_DIR})
```
and doing `cmake .. -DMRPT_DIR=/my_path/mrpt/build`

But the result is the same, it is not finding the core config even if I see the file in `/my_path/mrpt/build`

I am missing something, but I am not sure what.

Sorry for the inconvenience and thanks for your reply :)

jlblancoc, 2019-07-31:
It's weird... you don't have any `-I` flag for mrpt header directories... probably the problem is your CMakeLists file. 

Could you create the smallest possible set of: CMakeLists.txt and main.cpp that fails, and upload it here in a ZIP? 

You can also use ccmake, no gui, ncurses gui, good enough though ;-)
In there, try setting mrpt-core_DIR to your build directory, etc. 

Also, you could use mrpt from our PPA server, so you don't have to build it manually, and it will be easier for cmake to find all the stuff.

jolting, 2019-07-31:
`make install` will put those files in `/usr/local/include` unless you changed the install prefix. If the PPA works and the `make install` doesn't that would explain the difference. `/usr/include` is part of the default include path for gcc/clang.

MariaRamos89, 2019-08-01:
I am going to try to create the smallest example to see what I am doing wrong. If after that it's still failing I will share it and try the option of the PPA :)

@jolting thanks for your reply. I didn't change the prefix of the folders. In that case there is no flag indicating that the includes directories are in `/usr/local/include`. The last test was without doing the `make install`, like jblancoc suggested.

jlblancoc, 2019-08-01:
Only, remember clearing the /usr/local installation first, with "sudo
make uninstall" or after installing mrpt from PPA you'd have 2
versions and that might make be problematic in the future. If you
forgot uninstalling it, just go and manually delete anything under
/usr/local/*, assuming you don't have other required libraries
there...
Cheers

MariaRamos89, 2019-08-01:
Ok. I did a little piece of code using the `ICP` algorithm. The `main` is not going to do anything. Like I was expecting, if I forget to set up `MRPT_DIR` the error is that is not finding `MRPTConfig.cmake` or `mrpt-config.cmake`. After fixing it, it is not finding `mrpt-core_DIR`. As @jlblancoc suggested, I installed `ccmake` and after setting `mrpt-core_DIR` it compiled without any problem. The rest of the components of the library `mrpt-xxx_DIR` use the same address as `mrpt-core_DIR` internally? I mean, I was expecting to have to fill the rest of the directories of the other components. However, after filling the path of the `core` component, the rest had the same path automatically.

_Note_: I used `ccmake` to fill the variables, but I had to use `cmake` for compiling due to the first one is using the version `3.5.1` of `cmake`, so it doesn't allow c++17 features (from version`3.8`)

[small_piece_code.zip](https://github.com/MRPT/mrpt/files/3456745/small_piece_code.zip)

----

After that, I tried PPA version and it worked like a charm for the `small_piece_code`. 
I just removed the option `PATHS ${MRPT_DIR}` from my CMakeLists.txt and the result is: 

```
-- Searching for MRPT
mrpt-core_DIR was: /usr/share/mrpt-core
-- MRPT found
```

However, I am still having the same error in my main application.

```
/home/ubuntu/Dev/robot/source/common/includes.ihh:20:10: fatal error: mrpt/poses/CPose3D.h: No such file or directory
 #include <mrpt/poses/CPose3D.h>
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

This was weird. I continue modifying the small_piece_code until I found the same error:
[small_piece_code2.zip](https://github.com/MRPT/mrpt/files/3457233/small_piece_code2.zip)

The problem is related with the multiple use of CMakeLists that I am using. Because the original project has many files the more elegant solution was to use an intermediate CMakeLists which add automatically the `cpp` files and  __creates__ a new library with those files.

In that moment it looses the reference with `MRPT`. What should I do in this case?
Like I mentioned in a previous comment, when I used the `MRPT_INCLUDE_DIRS` in an old version of the library, I didn't have this problem, so what is the difference with your current version where `include_directories` is not needed. 

Thanks :)

jlblancoc, 2019-08-08:
Hi!

I think this is your problem:

```
target_link_libraries(robot tmp
                            ${MRPT_LIBRARIES})
```

that command says: for the robot executable, link it against: 1) tmp, and 2) MRPT. 

But you missed another one: 
```
target_link_libraries(tmp
                            ${MRPT_LIBRARIES})
```

On this: 

> I installed ccmake and after setting mrpt-core_DIR it compiled without any problem. The rest of the components of the library mrpt-xxx_DIR use the same address as mrpt-core_DIR internally?

Yes, it is a feature implemented into mrpt cmake scripts, to make things easier to users! ;-)

MariaRamos89, 2019-08-08:
Thanks!

Due to the previous versions of my library it worked in that way I never thought about it. 
In the `small_piece_code` it worked. However, in my main project not (even after removing `build` folder and rebuilding again).

I have just figured out that instead of doing the `target_link_libraries` to my sources, I just added sources in this way, removing the intermidiate library:

```
add_executable(robot ${INCLUDE_DIR}/main.cpp
                                     ${SOURCES})
```

And it seems it is working.

Thank you very much for your help! :D

closed 2019-08-08 by MariaRamos89