MRPT/mrpt #949: New Debian issues after merging ROS 1 stuff

jlblancoc, 2019-08-20:
- [ ] Build in i386 is broken due to failing unit tests, apparently in the conversion of gridmap values between ROS and MRPT maps: [log](https://launchpadlibrarian.net/438118104/buildlog_ubuntu-eoan-i386.mrpt_1%3A1.9.9~snapshot20190820-0956-git-eb3744a2-eoan-1_BUILDING.txt.gz)

```
[ RUN      ] Map.check_ros2mrpt_and_back
/<<PKGBUILDDIR>>/libs/ros1bridge/src/map_unittest.cpp:92: Failure
The difference between 1.0f - 0.01f * i and desMrpt.getCell(i, 0) is 0.99964302935404703, which exceeds 0.03f, where
1.0f - 0.01f * i evaluates to 1,
desMrpt.getCell(i, 0) evaluates to 0.00035697064595296979, and
0.03f evaluates to 0.029999999329447746.
ros to mprti=0
/<<PKGBUILDDIR>>/libs/ros1bridge/src/map_unittest.cpp:95: Failure
The difference between srcRos.data[i] and desRos.data[i] is 100, which exceeds 1, where
srcRos.data[i] evaluates to 0,
desRos.data[i] evaluates to 100, and
1 evaluates to 1.
ros to mprt to ros
[  FAILED  ] Map.check_ros2mrpt_and_back (3 ms)
```

After a quick inspection, I didn't see any obvious source of this problem... postponing. (Anyone?)

- [ ] Xenial packages are broken due to a bug in vtk6 / PCL packaging. See:
https://github.com/PointCloudLibrary/pcl/issues/1594#issuecomment-283873617
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=819741

Our error is: [log](https://launchpad.net/~joseluisblancoc/+archive/ubuntu/mrpt/+build/17443160)
```
/usr/bin/ld: cannot find -lvtkproj4
```
(vtkproj4 is not a real lib, but something internal to vtk)

@jolting : Didn't you have a VTK PPA for Xenial with a more modern version of vtk? I could add it as a PPA dependency to MRPT2 PPA.

PS: Good news is... `mrpt-ros1bridge` and the app `rosbag2rawlog` are already available from the PPA, and will be available from the official Ubuntu repos when mrpt 2 gets released :tada:

jolting, 2019-08-20:
VTK and PCL need to be rebuilt
Fetch your PCL and VTK packages from my backport PPA.
`sudo add-apt-repository -y ppa:jolting/backport-mrpt`

jlblancoc, 2019-08-21:
Brilliant, thanks! Added to the PPA and build retried, let's see...

jlblancoc, 2019-09-07:
> Build in i386 ...

might be fixed after 95ea836214f4d5d3f6158e50ec84a74963377937 . Let's keep an eye on PPA builds to check whether this one can be closed...

PR #983 (2019-11-04): Fix unsafe float == in logodds LUT; add unit tests
Finally fixes the 1st bug reported in #949, which was still present in PPA 32bit builds.

Closes #949
closed 2019-09-07 by jlblancoc