MRPT/mrpt #950: ARM64 not disabling sse

jolting, 2019-08-21:
```
[  0%] Built target DocumentationFiles
[  0%] Creating directories for 'libcvd'
Scanning dependencies of target man_page_robotic-arm-kinematics
[  0%] Generating MRPT man pages
[  0%] Building CXX object otherlibs/wxThings/CMakeFiles/mrptwxthings.dir/src/toggle.cpp.o
c++: error: unrecognized command line option ‘-msse2’
c++: error: unrecognized command line option ‘-mfpmath=sse’
c++: error: unrecognized command line option ‘-msse3’
c++: error: unrecognized command line option ‘-mssse3’
c++: error: unrecognized command line option ‘-msse4.1’
c++: error: unrecognized command line option ‘-msse4.2’
c++: error: unrecognized command line option ‘-msse4a’
otherlibs/wxThings/CMakeFiles/mrptwxthings.dir/build.make:62: recipe for target 'otherlibs/wxThings/CMakeFiles/mrptwxthings.dir/src/toggle.cpp.o' failed
make[2]: *** [otherlibs/wxThings/CMakeFiles/mrptwxthings.dir/src/toggle.cpp.o] Error 1
CMakeFiles/Makefile2:320: recipe for target 'otherlibs/wxThings/CMakeFiles/mrptwxthings.dir/all' failed
make[1]: *** [otherlibs/wxThings/CMakeFiles/mrptwxthings.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

I think these should be options
```
option(CMAKE_MRPT_HAS_SSE2 "SSE2" "OFF")
option(CMAKE_MRPT_HAS_SSE3 "SSE3" "OFF")
option(CMAKE_MRPT_HAS_SSE4_1 "SSE4_1" "OFF")
option(CMAKE_MRPT_HAS_SSE4_2 "SSE4_2" "OFF")
option(CMAKE_MRPT_HAS_SSE4_A "SSE4a" "OFF")
```

jlblancoc, 2019-08-21:
:+1:  We've been seeing this error for ages in the ROS build farm for the arm64 build of mrpt2, but being too lazy to create the issue ticket XD

We used to have an "AUTODETECT" feature in cmake for all those SIMD flags... should check it too.

jolting, 2019-08-21:
Ok I just spotted the Auto detect. I think if AutoDetect is off then is should default to disable everything. If you turn off autodetect everything is on by default.

jolting, 2019-08-21:
I think this will sort of fix it if you're lazy.
```
-set("DISABLE_${_setname}" OFF CACHE BOOL "Forces compilation WITHOUT ${_setname} extensions")

+set("DISABLE_${_setname}" ON CACHE BOOL "Forces compilation WITHOUT ${_setname} extensions")
```

jlblancoc, 2019-08-21:
Reopen if it's actually not fixed.

PR #952 (2019-08-21): refactor SIMD auto detection
Fixes #950 

And should also fix: http://build.ros2.org/job/Cbin_ubv8_uBv8__mrpt2__ubuntu_bionic_arm64__binary/133/consoleFull

This change is [[image]](https://reviewable.io/reviews/mrpt/mrpt/952)
closed 2019-08-21 by jlblancoc