MRPT/mrpt #951: Segmentation fault in mrpt-performance on Jetson Xavier

jolting, 2019-08-21:
```
feature_extraction [640x480]: ORB (OpenCV)                  (1000 found)
[17:24:22.2232|INFO |CTimeLogger] dumpAllStats:
--------------------------- MRPT CTimeLogger report ----------------------------
           FUNCTION                         #CALLS  MIN.T  MEAN.T MAX.T TOTAL 
--------------------------------------------------------------------------------
detectFeatures                               10   7.8ms  10.3ms  20.0ms 102.8ms
extractFeaturesORB                           10   7.8ms  10.3ms  20.0ms 102.8ms
+-> .fillFeatsStruct                         10 657.0us 904.2us   1.4ms   9.0ms
+-> .openCV_detectAndCompute                 10   6.9ms   9.1ms  18.4ms  91.5ms
--------------------------- MRPT CTimeLogger report ----------------------------
10.28 ms
feature_extraction [640x480]: FAST (OpenCV)                 Segmentation fault (core dumped)
```

jlblancoc, 2019-08-21:
Hmm... Xavier is based on arm64, right? It routinely passes all unit tests for all archs in the PPA build farm. 
Could you please check if `make test_legacy` is ok? If it is, then it means unit tests should be expanded ;-)

jolting, 2019-08-21:
Rebuilding with some more debug symbols.
```
0x0000007fb613f704 in cv::Mat::create(int, int const*, int) () from /usr/lib/libopencv_core.so.3.3
(gdb) bt
#0  0x0000007fb613f704 in cv::Mat::create(int, int const*, int) () at /usr/lib/libopencv_core.so.3.3
#1  0x0000007fb615e754 in cv::_OutputArray::create(int, int, int, int, bool, int) const ()
    at /usr/lib/libopencv_core.so.3.3
#2  0x0000007fb60ab2f8 in cv::Mat::copyTo(cv::_OutputArray const&) const ()
    at /usr/lib/libopencv_core.so.3.3
#3  0x0000007fb775edd8 in mrpt::img::CImage::extract_patch(mrpt::img::CImage&, unsigned int, unsigned int, unsigned int, unsigned int) const () at /home/ubuntu/mrpt/build/lib/libmrpt-img.so.1.9
#4  0x0000007fb7d8ec44 in mrpt::vision::CFeatureExtraction::extractFeaturesFAST(mrpt::img::CImage const&, mrpt::vision::CFeatureList&, unsigned int, unsigned int) ()
    at /home/ubuntu/mrpt/build/lib/libmrpt-vision.so.1.9
#5  0x0000007fb7d9745c in mrpt::vision::CFeatureExtraction::detectFeatures(mrpt::img::CImage const&, mrpt::vision::CFeatureList&, unsigned int, unsigned int, mrpt::vision::TImageROI const&) ()
    at /home/ubuntu/mrpt/build/lib/libmrpt-vision.so.1.9
#6  0x0000005555598698 in double benchmark_detectFeatures<(mrpt::vision::TKeyPointMethod)6>(int, int) ()
#7  0x000000555557e678 in main ()
```

jolting, 2019-08-21:
No issues with the tests.

Here's the line numbers for the mrpt-performance.
```
#0  0x0000007fb6145704 in cv::Mat::create(int, int const*, int) () at /usr/lib/libopencv_core.so.3.3
#1  0x0000007fb6164754 in cv::_OutputArray::create(int, int, int, int, bool, int) const () at /usr/lib/libopencv_core.so.3.3
#2  0x0000007fb60b12f8 in cv::Mat::copyTo(cv::_OutputArray const&) const () at /usr/lib/libopencv_core.so.3.3
#3  0x0000007fb7761de8 in mrpt::img::CImage::extract_patch(mrpt::img::CImage&, unsigned int, unsigned int, unsigned int, unsigned int) const (this=this@entry=0x7ffff81248, patch=..., col_=376, col_@entry=2357758996, row_=239, col_num=21, row_num=21)
    at /home/ubuntu/mrpt/libs/img/src/CImage.cpp:1201
#4  0x0000007fb7d90c74 in mrpt::vision::CFeatureExtraction::extractFeaturesFAST(mrpt::img::CImage const&, mrpt::vision::CFeatureList&, unsigned int, unsigned int) (this=0x7fb5d27858 <jpc_ns_invlift_colgrp+2304>, 
    this@entry=0x7ffff81350, inImg=..., feats=..., init_ID=init_ID@entry=0, nDesiredFeatures=nDesiredFeatures@entry=0)
    at /home/ubuntu/mrpt/libs/vision/src/CFeatureExtraction_FAST.cpp:176
#5  0x0000007fb7d9948c in mrpt::vision::CFeatureExtraction::detectFeatures(mrpt::img::CImage const&, mrpt::vision::CFeatureList&, unsigned int, unsigned int, mrpt::vision::TImageROI const&) (this=this@entry=0x7ffff81350, img=..., feats=..., init_ID=init_ID@entry=0, nDesiredFeatures=nDesiredFeatures@entry=0, ROI=...)
    at /home/ubuntu/mrpt/libs/vision/src/CFeatureExtraction_common.cpp:62
#6  0x00000055555986a8 in benchmark_detectFeatures<(mrpt::vision::TKeyPointMethod)6>(int, int) (N=100, h=<optimized out>)
    at /home/ubuntu/mrpt/apps/mrpt-performance/perf-feature_extraction.cpp:35
#7  0x000000555557e688 in std::function<double (int, int)>::operator()(int, int) const (__args#1=<optimized out>, __args#0=100, this=0x5555690c28) at /usr/include/c++/7/bits/std_function.h:706
#8  0x000000555557e688 in main(int, char**) (argc=<optimized out>, argv=<optimized out>)
    at /home/ubuntu/mrpt/apps/mrpt-performance/perf-main.cpp:168
```

jolting, 2019-08-21:
https://github.com/MRPT/mrpt/blob/16c1ac351c3188550f10723bf9e14d5d87846470/libs/vision/src/CFeatureExtraction_FAST.cpp#L164-L180
`*ft.patch` does not cause an exception. One of the quirks of optional. Also CFeature constructor is defaulted, so ft.patch isn't initialized anywhere. Maybe CFeature should construct a new CImage, so that ft.patch is always valid?

jlblancoc, 2019-08-22:
Thanks! Yes, there was a few `ft.patch.emplace()` calls in many other feature types, but not in that one...

commit a3295ccf (2019-08-22): BUGFIX: Missing emplace() in optional descriptors

Fixes #951
closed 2019-08-22 by jlblancoc