MRPT/mrpt #957: Problem building on windows

DaddyWesker, 2019-09-13:
Hello, so i've tried to build mrpt for windows. CMake configured and generated project properly (i've added opencv path as it was strongly recommended), but whem i'm trying to build BUILD_ALL using visual studio i'm getting errors (see picture). What could be done to build it properly?

[image]

jlblancoc, 2019-09-13:
Hi, 
Please, attach your CMakeCache.txt file, and also report what's the absolute path of your OpenCV **build** directory, as well as what version of OpenCV are you using.

DaddyWesker, 2019-09-14:
Hello,

Here is full path to opencv installation D:\workspace\libs\Install\opencv

[CMakeCache.txt](https://github.com/MRPT/mrpt/files/3612398/CMakeCache.txt)

jlblancoc, 2019-09-14:
Ok, so it might be a compatibility problem with opencv 4... will check it.

jlblancoc, 2019-09-14:
I just tested it and mrpt at present builds cleanly against opencv4. 

It looks like a misconfiguration of your MRPT's CMake build directory... Most `opencv_*` variables in cmakeCache are empty. Perhaps the OpenCV_DIR should point to somewhere else? Try setting OpenCV_DIR directly to the BUILD directory where you compiled opencv...

DaddyWesker, 2019-09-14:
Well, i've put exactly build location of opencv... Oh well, i'll try again maybe in the monday and will get back to you. Probably will try  to rebuild opencv if this will not works

jlblancoc, 2019-09-14:
OpenCV_DIR:PATH=D:/workspace/libs/Install/opencv

Hmm... that is what you currently have, maybe that's the problem, it's probably safer to use the build directory instead.

DaddyWesker, 2019-09-17:
Alright, i've rebuilt opencv (downloaded 3.4.7 with contribs, built it from sources) and tried to rebuilt mrpt from scratch (deleted build dir and used cmake to generate code). And still...
[image]

[CMakeCache.txt](https://github.com/MRPT/mrpt/files/3620161/CMakeCache.txt)

DaddyWesker, 2019-09-17:
Current path for opencv is D:\workspace\libs\opencv\opencv\build

jlblancoc, 2019-09-20:
It's weird... today I rebuilt MRPT in Windows with opencv 3.4.5 and it builds cleanly. AppVeyor CI [is also happy](https://ci.appveyor.com/project/jlblancoc/mrpt/history), so... I can't reproduce your problem (!)

What version of MRPT are you building? git master?

DaddyWesker, 2019-09-25:
Not master. Actually, i'm trying to build this one https://github.com/pangfumin/gf-pl-slam and it says i need to build not master, but https://github.com/MRPT/mrpt/tree/0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3

jlblancoc, 2019-09-26:
Ah! So, bottom line is we must address https://github.com/rubengooj/pl-slam/issues/30
:-)

DaddyWesker, 2019-09-26:
SO, i guess, nothing can be done until your issue for pl-slam is addressed?

jlblancoc, 2019-09-26:
Well, if you want to start giving a hand with this, build MRPT master on Windows, if that's the OS you will be using, and then try to build (GF-)-PL-SLAM against it, and let's see what needs to be updated in PL-SLAM. 
There's a "how to port guide" here: https://docs.mrpt.org/reference/master/porting_mrpt2.html

richard5635, 2019-10-08:
I faced this problem as well. 
What I did was to comment out  `#include <opencv2/video/tracking_c.h>` in 
libs\img\include\mrpt\otherlibs\do_opencv_includes.h (it is at line 36 in my file)

I'm not sure if this is the right way to tackle this issue, but after that, I managed to build most of the projects. Now I am facing this problem [here](https://github.com/MRPT/mrpt/issues/962) though..

jlblancoc, 2019-12-07:
Closing, as this has been fixed upstream (though, the PR is still in the wait for merging): 
https://github.com/rubengooj/pl-slam/pull/40

closed 2019-12-07 by jlblancoc