MRPT/mrpt #958: Visualising depths in rawlog

hoonkai, 2019-09-18:
Hi

I wonder if there's any tool available for visualising depths in a rawlog file as depth images. I have converted the TUM dataset RGBD images to a rawlog file using the rgbd_dataset2rawlog tool, but I'd like to verify that the depths are preserved. Is there a tool for that?

Thanks

jolting, 2019-09-18:
Did you try the RawLogViewer?

hoonkai, 2019-09-18:
> Did you try the RawLogViewer?

Yes, I can see point clouds but not depth images.

jolting, 2019-09-18:
I could be wrong, but I don't think that feature exists yet. I think it would be a great feature request for RawLogViewer.

Take a look here for examples on how to add it:
https://github.com/MRPT/mrpt/blob/1de0e027c16e094e1fe6e5327731db5148ba8bbe/apps/kinect-3d-view/kinect-3d-view_main.cpp#L217
https://github.com/MRPT/mrpt/blob/1de0e027c16e094e1fe6e5327731db5148ba8bbe/apps/kinect-3d-view/kinect-3d-view_main.cpp#L264

hoonkai, 2019-09-18:
Right. Thanks for that.

jlblancoc, 2019-09-18:
Hmm... Do you mean a visualization of the depth image/matrix? It has been already there in RawLogViewer for ages! :-) You must just select the "range" tab, to see the range instead of the pointcloud. I'm assuming you have `CObservation3DRangeScan` objects, right?

See the attached screenshot:

[image]

You can right click on the image to save it to disk, or use the tool `rawlog-edit` to ensure that range images are already all saved as external files, if that's needed.

hoonkai, 2019-09-19:
> Hmm... Do you mean a visualization of the depth image/matrix? It has been already there in RawLogViewer for ages! :-) You must just select the "range" tab, to see the range instead of the [...]

You're right! Thanks so much.

closed 2019-09-19 by hoonkai