MRPT/mrpt #960: libs/pbmap/CMakeFiles/mrpt-pbmap.dir/build.make:63: recipe for target 'libs/pbmap/cotire/mrpt-pbmap_CXX_prefix.hxx.gch' failed

why-freedom, 2019-09-21:
CMake Error at /home/why/Documents/SLAMlibs/mrpt-1.5.5/cmakemodules/cotire.cmake:1848 (message):
  cotire: error 1 precompiling
  /home/why/Documents/SLAMlibs/mrpt-1.5.5/build/libs/pbmap/cotire/mrpt-pbmap_CXX_prefix.hxx.
Call Stack (most recent call first):
  /home/why/Documents/SLAMlibs/mrpt-1.5.5/cmakemodules/cotire.cmake:3512 (cotire_precompile_prefix_header)

libs/pbmap/CMakeFiles/mrpt-pbmap.dir/build.make:63: recipe for target 'libs/pbmap/cotire/mrpt-pbmap_CXX_prefix.hxx.gch' failed
make[2]: *** [libs/pbmap/cotire/mrpt-pbmap_CXX_prefix.hxx.gch] Error 1
CMakeFiles/Makefile2:1204: recipe for target 'libs/pbmap/CMakeFiles/mrpt-pbmap.dir/all' failed
make[1]: *** [libs/pbmap/CMakeFiles/mrpt-pbmap.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

I meet this problem ，can you help me？

jlblancoc, 2019-09-21:
Try setting `MRPT_ENABLE_PRECOMPILED_HDRS` to OFF from ccmake or cmake-gui...

jlblancoc, 2019-09-29:
Any update on this problem? Is it reproducible?

why-freedom, 2019-09-29:
yeah。I try to make it by your methd，but failed。Then I delete pbmap file，and make successful！Thank you

| |
温焕宇
|
|
邮箱：why_17535184010@163.com
|

签名由 网易邮箱大师 定制

On 09/29/2019 17:39, Jose Luis Blanco-Claraco wrote:

Any update on this problem? Is it reproducible?

jlblancoc, 2019-09-30:
Ok, great!

closed 2019-09-30 by jlblancoc