MRPT/mrpt #962: Pbmap build error: RegisterAllClasses.cpp and SubgraphMatcher.cpp

richard5635, 2019-10-08:
Hi, I tried to build MRPT using CMake for MSVC2017 with C++17 on my windows.
I managed to build everything except for pbmap, which gave me a list of errors shown below.

I assume the error rooted from registerAllClasses.cpp in the code here
```cpp
MRPT_INITIALIZER(registerAllClasses_mrpt_pbmap)
{
#if !defined(DISABLE_MRPT_AUTO_CLASS_REGISTRATION)
	registerClass(CLASS_ID(Plane));
	registerClass(CLASS_ID(PbMap));
#endif
}
```
Where the compiler says `registerAllClasses_mrpt_pbmap` is undeclared.
I am still not sure about what causes this, but if anyone's familiar with this any help would be appreciated!

I have installed OpenCV 4.0.1, PCL 1.9 on my pc as well.

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	MSB6006	"cmd.exe" exited with code -1073741515.	run_tests_mrpt_topography	C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	209	
Error	MSB6006	"cmd.exe" exited with code -1073741515.	run_tests_mrpt_hwdrivers	C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	209	
Error	C2065	'registerAllClasses_mrpt_pbmap': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\registerAllClasses.cpp	16	
Error	C4430	missing type specifier - int assumed. Note: C++ does not support default-int	pbmap	D:\lib\mrpt\src\libs\pbmap\src\registerAllClasses.cpp	16	
Error	C2448	'MRPT_INITIALIZER': function-style initializer appears to be a function definition	pbmap	D:\lib\mrpt\src\libs\pbmap\src\registerAllClasses.cpp	17	
Error	C2653	'SubgraphMatcher': is not a class or namespace name	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	40	
Error	C4430	missing type specifier - int assumed. Note: C++ does not support default-int	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	41	
Error	C2653	'SubgraphMatcher': is not a class or namespace name	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	48	
Error	C2065	'Plane': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	49	
Error	C2065	'plane1': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	49	
Error	C2065	'plane2': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	49	
Error	C2065	'PbMap': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	49	
Error	C2065	'trgPbMap': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	49	
Error	C2062	type 'bool' unexpected	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	49	
Error	C2143	syntax error: missing ';' before '{'	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	50	
Error	C2447	'{': missing function header (old-style formal list?)	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	50	
Error	C2653	'SubgraphMatcher': is not a class or namespace name	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	210	
Error	C2065	'Plane': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	211	
Error	C2065	'plane1': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	211	
Error	C2065	'plane2': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	211	
Error	C2065	'PbMap': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	211	
Error	C2065	'trgPbMap': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	211	
Error	C2062	type 'bool' unexpected	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	211	
Error	C2143	syntax error: missing ';' before '{'	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	212	
Error	C2447	'{': missing function header (old-style formal list?)	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	212	
Error	C2653	'SubgraphMatcher': is not a class or namespace name	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	396	
Error	C2065	'Plane': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	397	
Error	C2065	'plane1': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	397	
Error	C2065	'plane2': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	397	
Error	C2065	'PbMap': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	397	
Error	C2065	'trgPbMap': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	397	
Error	C2062	type 'bool' unexpected	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	397	
Error	C2143	syntax error: missing ';' before '{'	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	398	
Error	C2447	'{': missing function header (old-style formal list?)	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	398	
Error	C2653	'SubgraphMatcher': is not a class or namespace name	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	572	
Error	C2065	'Plane': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	573	
Error	C2065	'plane1': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	573	
Error	C2065	'plane2': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	573	
Error	C2065	'PbMap': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	573	
Error	C2065	'trgPbMap': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	573	
Error	C2062	type 'bool' unexpected	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	573	
Error	C2143	syntax error: missing ';' before '{'	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	574	
Error	C2447	'{': missing function header (old-style formal list?)	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	574	
Error	C2653	'SubgraphMatcher': is not a class or namespace name	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	778	
Error	C2065	'Plane': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	779	
Error	C2065	'Ref': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	779	
Error	C2065	'neigRef': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	779	
Error	C2065	'Check': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	779	
Error	C2065	'neigCheck': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	779	
Error	C2448	'evalBinaryConstraints': function-style initializer appears to be a function definition	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	780	
[... 3 lines cut ...]
Error	C2143	syntax error: missing ';' before '{'	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	1016	
Error	C2447	'{': missing function header (old-style formal list?)	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	1016	
Error	C2653	'SubgraphMatcher': is not a class or namespace name	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	1183	
Error	C2065	'set': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	1184	
Error	C2062	type 'unsigned int' unexpected	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	1184	
Error	C2143	syntax error: missing ';' before '{'	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	1186	
Error	C2447	'{': missing function header (old-style formal list?)	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	1186	
Error	C2653	'SubgraphMatcher': is not a class or namespace name	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	1342	
Error	C2065	'subgraphSrc': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	1347	
Error	C2653	'SubgraphMatcher': is not a class or namespace name	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	1352	
Error	C2039	'set': is not a member of 'std'	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	1352	
Error	C2065	'set': undeclared identifier	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	1352	
Error	C2062	type 'unsigned int' unexpected	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	1352	
Error	C2143	syntax error: missing ';' before '{'	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	1353	
Error	C2447	'{': missing function header (old-style formal list?)	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	1353	
[... cut ...]
Error	C2059	syntax error: 'const'	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	1397	
Error	C2143	syntax error: missing ';' before '{'	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	1398	
Error	C2447	'{': missing function header (old-style formal list?)	pbmap	D:\lib\mrpt\src\libs\pbmap\src\SubgraphMatcher.cpp	1398	
Error	C2143	syntax error: missing ',' before '<' (compiling source file D:\lib\mrpt\src\libs\pbmap\src\PbMapMaker.cpp)	pbmap	D:\lib\FLANN\install\include\flann\util\heap.h	108	

```

jlblancoc, 2019-10-08:
I could reproduce the problem. 
Work in progress solution is in the branch of the PR #963 , give it a try is you want to. 

I fixed many of the problems (see log messages), but the next big problem now is that mrpt requires C++17 and apparently FLANN (inside PCL) uses obsolete C++ classes that have been deprecated in c++17. 
My WIP solution is to split mrpt-specific code and PCL-code in two files, and build PCL stuff with C++14 only... (sigh) Lots of work, not sure it will be worth... 

By the way: if you don't need PbMap, just disable it via CMake: BUILD_mrpt-pbmap  = OFF and you should have no more errors.

richard5635, 2019-10-08:
Thanks a lot!!

True, I just disabled PbMap for the time being. Will refer to this if I ever need it :)

PR #963 (2019-10-10): Fix build PbMap / PCL  in MSVC
## Changed apps/libraries

* mrpt-pbmap

Closes: #962 

WIP...
commit ec286e96 (2019-10-10): Merge pull request #963 from MRPT/fix-build-pbmap-pcl-msvc

Fix build PbMap / PCL  in MSVC

Closes: #962
closed 2019-10-10 by jlblancoc