MRPT/mrpt #976: ro-localisation application fails

antithing, 2019-10-26:
Hi, I have built the ro-localisation app and am running it on the provided data.

`ro-localization.exe UWB_localization_demo.ini`

It fails with this error:
```

Loading the rawlog file...rawlog Entries: 686
OK
Creating directory: LOG_BEACONS_Pc_0.050000
Creating directory: LOG_BEACONS_Pc_0.050000/REP_000
PDF of 15000 particles initialized in 25.386ms

RAWLOG_ENTRY: 0

Executing ParticleFilter on 15000 particles....Caught MRPT exception:
Exception message:  Assert condition failed: std::isfinite(ret)
==== MRPT exception backtrace ====
[4] d:\\mrpt\mrpt-master\libs\vision\src\maps\clandmarksmap.cpp:306: [mrpt::maps::CLandmarksMap::internal_computeObservationLikelihood] Assert condition failed: std::isfinite(ret)
[3] d:\\mrpt\mrpt-master\libs\vision\src\maps\clandmarksmap.cpp:432: [mrpt::maps::CLandmarksMap::internal_computeObservationLikelihood] Called from here.
[2] d:\\mrpt\mrpt-master\libs\maps\src\maps\cmultimetricmap.cpp:188: [mrpt::maps::CMultiMetricMap::internal_computeObservationLikelihood] Called from here.
[1] d:\\mrpt\mrpt-master\apps\ro-localization\cposepdfparticlesextended.cpp:414: [mrpt::poses::CPosePDFParticlesExtended::prediction_and_update_pfStandardProposal] Called from here.
[0] d:\\mrpt\mrpt-master\libs\bayes\src\cparticlefilter.cpp:96: [mrpt::bayes::CParticleFilter::executeOn] Called from here.
```

Do i need to change something to run this application?

Thanks!

jlblancoc, 2019-10-27:
Thanks for reporting! 
It's true: it doesn't fail all the time... but I managed to make it fail at least once, so it's definitively a bug. 
Will look at it.

antithing, 2019-10-28:
Thank you! That PR is running smoothly for me. 

Quick question, if I add more beacons to the data, would it be a simple thing to return a 3d position as xyz instead of xy yaw ?

PR #978 (2019-10-28): Fix/ro localization crash
Closes #976
closed 2019-10-28 by jlblancoc