MRPT/mrpt #984: core dumped after target_link_libraries( ${MRPT_LIBRARIES})

lisilin013, 2019-11-04:
I install mrpt 1.99 by ppa.

when I want to integrate 3d pf-localization function in my ros package, I write the follow simple cpp code for testing whether cod built is ok.
```cpp
#include <ros/ros.h>
#include <string>
using namespace std;

int main(int argc, char **argv) {
    ros::init(argc, argv, "pf_localization");
    ROS_INFO("ini_file is ready ");
    return 0;
}
```
Cmakelists.txt
```cmake
cmake_minimum_required(VERSION 3.5)
project(pf_localization)

## Compile as C++11, supported in ROS Kinetic and newer
add_compile_options(-std=c++11)

# find mrpt
find_package(MRPT REQUIRED COMPONENTS
        slam nav ros1bridge)
message(STATUS "=============================")
message(STATUS "Find MRPT_VERSION: ${MRPT_VERSION}")
message(STATUS "MRPT_LIBRARIES: ${MRPT_LIBRARIES}")
message(STATUS "=============================")
find_package(catkin REQUIRED COMPONENTS
        roscpp tf nav_msgs sensor_msgs pcl_ros pcl_conversions
        )

catkin_package(
        INCLUDE_DIRS include
        #          LIBRARIES pf_localization
        CATKIN_DEPENDS roscpp tf nav_msgs sensor_msgs pcl_ros pcl_conversions
        #          DEPENDS system_lib
)

include_directories(
        include
        ${catkin_INCLUDE_DIRS}
)

## Declare a cpp executable
add_executable(pf_localization
        src/pf_localization_node.cpp
#        src/pf_localization.cpp
        )

target_link_libraries(pf_localization
        ${catkin_LIBRARIES}
        ${PCL_LIBRARIES}
        ${MRPT_LIBRARIES}
        )
```
Then I got the following problem when I run this ros node
```bash
$ rosrun pf_localization pf_localization
[1]    13146 segmentation fault (core dumped)  rosrun pf_localization pf_localization

```

But when I comment this line in CMakeLists.txt
```cmake
 target_link_libraries( ${MRPT_LIBRARIES})
```
This node can run successfully.

I am very confused about this problem.
Very looking forward to your reply!

jolting, 2019-11-04:
Can you get a backtrace from the core dump?

I think on Ubuntu they go in /var/crash.

lisilin013, 2019-11-05:
> Can you get a backtrace from the core dump?
> 
> I think on Ubuntu they go in /var/crash.

I find nothing in the /var/crash.
But I try to use gdb to show the core dump infomation.
```bash
$ gdb devel/lib/pf_localization/pf_localization core 

warning: exec file is newer than core file.
[New LWP 23468]
[New LWP 23477]
[New LWP 23478]
[New LWP 23484]
Core was generated by `/home/nrsl/code/pf_demo_ws/devel/lib/pf_localization/pf_localization'.
Program terminated with signal SIGABRT, Aborted.
#0  0x00007fa48bdb8428 in ?? ()
[Current thread is 1 (LWP 23468)]
(gdb) bt
#0  0x00007fa48bdb8428 in ?? ()
#1  0x00007fa48bdba02a in ?? ()
#2  0x0000000000000020 in ?? ()
#3  0x0000000000000000 in ?? ()
(gdb) bt full
#0  0x00007fa48bdb8428 in ?? ()
No symbol table info available.
#1  0x00007fa48bdba02a in ?? ()
No symbol table info available.
#2  0x0000000000000020 in ?? ()
No symbol table info available.
#3  0x0000000000000000 in ?? ()
No symbol table info available.
(gdb) info threads 
  Id   Target Id         Frame 
* 1    LWP 23468         0x00007fa48bdb8428 in ?? ()
  2    LWP 23477         0x00007fa48be8aa13 in ?? ()
  3    LWP 23478         0x00007fa48be805d3 in ?? ()
  4    LWP 23484         0x00007fa48af17709 in ?? ()
(gdb) thread apply all bt

Thread 4 (LWP 23484):
#0  0x00007fa48af17709 in ?? ()
#1  0x0000000000000000 in ?? ()

Thread 3 (LWP 23478):
#0  0x00007fa48be805d3 in ?? ()
#1  0x0000000000000000 in ?? ()

Thread 2 (LWP 23477):
#0  0x00007fa48be8aa13 in ?? ()
#1  0x0000000000000000 in ?? ()

Thread 1 (LWP 23468):
#0  0x00007fa48bdb8428 in ?? ()
#1  0x00007fa48bdba02a in ?? ()
#2  0x0000000000000020 in ?? ()
#3  0x0000000000000000 in ?? ()
(gdb) thread apply all bt full

Thread 4 (LWP 23484):
#0  0x00007fa48af17709 in ?? ()
No symbol table info available.
#1  0x0000000000000000 in ?? ()
No symbol table info available.

Thread 3 (LWP 23478):
#0  0x00007fa48be805d3 in ?? ()
No symbol table info available.
#1  0x0000000000000000 in ?? ()
No symbol table info available.

Thread 2 (LWP 23477):
#0  0x00007fa48be8aa13 in ?? ()
No symbol table info available.
#1  0x0000000000000000 in ?? ()
No symbol table info available.

Thread 1 (LWP 23468):
#0  0x00007fa48bdb8428 in ?? ()
No symbol table info available.
#1  0x00007fa48bdba02a in ?? ()
No symbol table info available.
#2  0x0000000000000020 in ?? ()
No symbol table info available.
#3  0x0000000000000000 in ?? ()
No symbol table info available.
(gdb) quit

```

when I use gdb to analyze the core file
```bash
(gdb) core-file core 
warning: core file may not match specified executable file.
[New LWP 1031]
Core was generated by `/usr/lib/xorg/Xorg -core :0 -seat seat0 -auth /var/run/lightdm/root/:0 -noliste'.
Program terminated with signal SIGABRT, Aborted.
#0  0x00007fef793a5428 in ?? ()
```

I don't think I caught the error essentially, so anything else I can do for solving the problem?
Thanks for your hints sincerely.

jolting, 2019-11-05:
Did you try building with `catkin_make -DCMAKE_BUILD_TYPE=Debug`?

jlblancoc, 2019-11-05:
Another idea: please report the output for
`uname -a`
and
`grep flags /proc/cpuinfo`
just in case you have an old CPU using the >=SSE3 PPA builds.

jolting, 2019-11-05:
Do you think there are SSE instructions during global init? That could probably be avoided.

jlblancoc, 2019-11-05:
> Do you think there are SSE instructions during global init? That could probably be avoided.

Hmm... there shouldn't be, but perhaps some matrix initialization to zeros or alike is implemented by Eigen with such instructions (just a wild guess). 

Without a stack trace with debug information it's hard to tell what's going on :-)

lisilin013, 2019-11-06:
> Did you try building with `catkin_make -DCMAKE_BUILD_TYPE=Debug`?

Yes, I have tried two kinds of building type, but still not work. 
And I find that if I build my code with MRPT 1.5, everything will be OK! 

>Another idea: please report the output for
```bash
$ uname -a
Linux nrsl 4.15.0-33-generic #36~16.04.1-Ubuntu SMP Wed Aug 15 17:21:05 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

```bash
grep flags /proc/cpuinfo      
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp
[... cut ...] ology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp
```

jolting, 2019-11-06:
I would suggest installing the -dbgsym package, but it looks like those aren't available in the PPA.

Also, I just noticed `warning: exec file is newer than core file.` which means you probably rebuilt since the core was dumped. Your debug symbols won't match the core dump.

jlblancoc, 2019-11-06:
> I would suggest installing the -dbgsym package, but it looks like those aren't available in the PPA.

Ooops... at some point I disabled (why!?) that on the PPA. It's enabled now again, the next build will bring the -dbgsym packages.

jlblancoc, 2019-11-08:
The dbgsym packages are [now published](https://launchpad.net/~joseluisblancoc/+archive/ubuntu/mrpt/+packages) in the PPA... 

I expected them to be available with:

`sudo apt install libmrpt*-dbgsym` 

but they are apparently not available after doing `sudo apt update`. 
@jolting Can you see anything else that I may be missing here??

jolting, 2019-11-08:
https://wiki.ubuntu.com/DebuggingProgramCrash

You need to add
```
deb http://ppa.launchpad.net/TEAM/PPA/ubuntu RELEASE main main/debug
```

jolting, 2019-11-08:
Ok for some reason that didn't work for me, but they were clearly generated.
https://launchpad.net/~joseluisblancoc/+archive/ubuntu/mrpt/+build/18066128

jlblancoc, 2019-11-19:
Any news on this one? 
Will close due to lack of activity and since we couldn't reproduce it. 
Please, reopen if the problem persists.

closed 2019-11-19 by jlblancoc