MRPT/mrpt #986: Possible regression in 3D LiDAR-based PF localization (?)

jlblancoc, 2019-11-07:
Running the global relocalization test for [this experiment](http://ingmec.ual.es/datasets/lidar3d-pf-benchmark/) now (git develop) takes >1 minute for the first iterations, while older versions (e.g. 1f1c05290 ) seems to run in a few seconds.

Also, current version seems to reach much smaller ESS numbers, so it might be a likelihood function issue, perhaps just a backwards incompatible parameter change, or a real bug. 

To be further investigated...

lisilin013, 2019-11-07:
Yeah, this method works well, and the pf algorithm can localize successfully in 25s~30s.

jlblancoc, 2020-08-26:
Indeed it was slower, due to incorrect likelihood parameters being propagated to the actual object running inside the PF.
Fixed with ceec2a34a042838dcde34bdc8c055ecd61661b37

closed 2020-08-26 by jlblancoc