MRPT/mrpt #994: DifOdometry-Datasets not recognizing key press

omar-ogm, 2019-11-16:
Hi,

Ive been trying to make work the DIFODO algorithm with DifOdometry-Datasets app without luck.

**Problem:**
The problem is that, even though the 3D window (GUI) opens and I can freely move around with my mouse, the keys are not recognized ("n", "s", "e").

**What Ive tried:**
Im working on Ubuntu 18.04.3 LTS, kernel 4.15.0-70. Ive download the mrpt-apps within the ubuntu offical repository (MRPT v1.5.5).
Ive also installed throught your ppa (v 1.9.9).
Neither of them seems to work. The same problems arise, the key pressed is unnoticed.

Also ive downloaded and installed the executable for windows with the same problem, no keys recognized.

The way Im working is:

Create a config file with:
`DifOdometry-Datasets --create-config myconfig.txt`

Then change the filename configuration parameter from myconfig.txt to point to the rawlog from 
*filename = /home/omar/workspaces/DIFODO/rawlog/rawlog_rgbd_dataset_freiburg1_360/rgbd_dataset_freiburg1_360.rawlog* 
which I downloaded from your website, and extracted.

And finally I run the program with:
`DifOdometry-Datasets --config myconfig.txt`

Maybe Im missing something but since the key "e" to exit is not working I find that my keyboard is not being recognized for some reason. (USB keyboard).

I will keep trying to make it work, any help will be appreciated, thanks.

Finally, maybe is related to  #13?

jlblancoc, 2019-11-18:
Thanks for reporting, I'll check it...

jlblancoc, 2019-11-18:
The commit above fixed the problem! 

We have @MarianoJT88 's great difodo demo apps back alive ;-)

[image]

closed 2019-11-18 by jlblancoc

omar-ogm, 2019-11-19:
Is working!
Thank you for you quick response.

MarianoJT88, 2019-11-21:
Thanks José Luis :)

commit 9b18308f (2019-11-18): fixes to difodo demo apps (Closes #994)