MRPT/mrpt #996: Eigensolver couldn't find a solution. Pose is not updated

omar-ogm, 2019-11-19:
Hello again, previously the DifOdometry-Datasets and  DifOdometry-Camera were not starting due to #994 .

Now Ive realised that while the grountruth appears, the DifOdo pose estimation is not being done (also not updated on the visor).

On the terminal this message appears 5 times each pose estimation:

**Eigensolver couldn't find a solution. Pose is not updated**

This is due to the method CDifodo::filterLevelSolution() in mrpt/libs/vision/src/CDifodo.cpp.

Specifically the mrpt::math::CMatrixFloat66 est_cov when it calls the method **eig_symmetric()** to get the eigenvalues and vectors doesnt seem to return any value.

Ill keep trying to debug this more while a solution is provided.

Sorry to bother again.

MarianoJT88, 2019-11-21:
I don´t remember using any function called eig_symmetric() but I might have forgotten it... check is there is an alternative to get eigenvalues and eigenvectors with Eigen?

jlblancoc, 2019-11-21:
Fixed in 31e853f504294c1b1ed706b2c3c525c945d63dc0 . 

@MarianoJT88 if you see the commit diff, you will see what a silly error this was. It was me during a huge recent refactoring of mrpt::math classes.

closed 2019-11-21 by jlblancoc