MRPT/mrpt #998: Ubuntu18.04 CMake Error: LIST GET given empty list

wsakobe, 2019-11-23:
The following is the detail in cmake-gui (cmake version 3.10.2)
CMake Error at cmakemodules/script_detect_gcc.cmake:16 (LIST):
  LIST GET given empty list
Call Stack (most recent call first):
  CMakeLists.txt:83 (include)

CMake Error at cmakemodules/script_detect_gcc.cmake:17 (LIST):
  LIST GET given empty list
Call Stack (most recent call first):
  CMakeLists.txt:83 (include)

What's the problem?

wsakobe, 2019-11-23:
BTW I'd like to run pl-slam and the guide said that: 
If working with the most recent versions of the MRPT library you might find some issues due to hard refactoring, for which we recommend to use this version instead (the last one we tested):

https://github.com/MRPT/mrpt/tree/0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3

wsakobe, 2019-11-23:
I have fixed this problem by changing 'dumpversion' in  this line into 'dumpfullversion'
But now when I did 'make -j8' i found another problem:
/home/xxx/Downloads/mrpt-0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3/libs/base/include/mrpt/math/matrix_adaptors.h:507:20: error: expression cannot be used as a function
     return howMany();
How can I fix this problem? i'm using gcc 7.4.0

jlblancoc, 2019-11-23:
Hi, 
This is actually a duplicated of https://github.com/rubengooj/pl-slam/issues/30
Will address that one and post there...

closed 2019-11-23 by jlblancoc

jlblancoc, 2019-11-23:
This has been finally solved via these two PRs (hopefully to be merged soon...): 
1) https://github.com/rubengooj/stvo-pl/pull/15
2) https://github.com/rubengooj/pl-slam/pull/40

Cheers.

wsakobe, 2019-11-24:
Thanks a lot bro! Let me have a look : )

weiningwei, 2021-06-04:
> Thanks a lot bro! Let me have a look : )

Hi, I also met the same problem. Do you solve it?

johnjin404, 2021-09-10:
> Thanks a lot bro! Let me have a look : )

Hi,  have you fixed the problem of make error that shows "error: expression cannot be used as a function  return howMany();"

johnjin404, 2021-09-10:
@weiningwei 
Hi, have you fixed the problem of "make error "that shows
 "mrpt/libs/base/include/mrpt/math/matrix_adaptors.h:507:20:  error: expression cannot be used as a function return howMany( );"

weiningwei, 2021-09-10:
@johnjin404 
Hi, line 507 of file "mrpt/libs/base/include/mrpt/math/matrix_adaptors.h" is `return howMany();`. 
I modified it to `return howMany;`

jlblancoc, 2021-09-14:
It seems you are using an outdated version of mrpt, please try to upgrade to either the latest stable release or the develop branch.

shaoxh, 2022-04-08:
> @johnjin404 Hi, line 507 of file "mrpt/libs/base/include/mrpt/math/matrix_adaptors.h" is `return howMany();`. I modified it to `return howMany;`

This is great! Solved it!