MRPT/mrpt #1017: Set Kinect to /dev/video0 ?

jiapei100, 2020-01-28:
I'm testing **kinect-stereo-calib**, which seems to **ALWAYS** search `/dev/video0` as Kinect. I believe there is a way to search the device SSID, instead of `/dev/video0` as Kinect...

```console
(kinect-stereo-calib:2122): Gtk-CRITICAL **: 20:29:15.882: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton
Could not open camera: -3
Calling CKinect::initialize()...[Kinect thread] Exception: Exception message:  Error opening Kinect sensor with index: 0
==== MRPT exception backtrace ====
[0] ....../mrpt/libs/hwdrivers/src/CKinect.cpp:392: [void mrpt::hwdrivers::CKinect::open()] Error opening Kinect sensor with index: 0

Calling CKinect::initialize()...[Kinect thread] Exception: Exception message:  Error opening Kinect sensor with index: 0
==== MRPT exception backtrace ====
[0] ....../mrpt/libs/hwdrivers/src/CKinect.cpp:392: [void mrpt::hwdrivers::CKinect::open()] Error opening Kinect sensor with index: 0

(kinect-stereo-calib:2122): Gtk-CRITICAL **: 20:29:41.858: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton
```

jlblancoc, 2020-01-28:
Perhaps it's a permissions problem? Try running the program with "sudo" to verify it...

jiapei100, 2020-01-28:
Yes, it is... thank you...

closed 2020-01-28 by jiapei100