MRPT/mrpt #1018: kinect-stereo-calib

jiapei100, 2020-01-28:
Still about **kinect-stereo-calib** .

1)
Where is the video?
[https://www.mrpt.org/tutorials/programming/miscellaneous/kinect-calibration/](https://www.mrpt.org/tutorials/programming/miscellaneous/kinect-calibration/)
It seems the youtube video is a kind of old [https://www.youtube.com/watch?v=9-PWe4vTipo](https://www.youtube.com/watch?v=9-PWe4vTipo) .

2) Step 3: Stereo/RGBD calibration
**Calibration returned an error status.**

During the live calibration, I can clearly tell in **ALL** images, the chessboard is detected correctly. However, **Run optimizer...** gives me the following ERROR message:
**Calibration returned an error status.**

And, it seems it's impossible to backward from step 3 to step 2 to re-capture some new image pairs?

3)  Finally, for the calibration board.

**7x9 checkerboard for camera calibration. 
Squares are: 20x20 mm if printed to 1:1 scale on an A4 paper.**

It should be better to display it on a monitor directly, but how to guarantee this pdf file is display in an A4 with size of 100%? My **qpdfview** doesn't seem to have such an option.

jolting, 2020-03-19:
1. It should automatically show up.

2. Set verbose in TStereoCalibParams
You're probably not capturing any images.

3. The scale isn't all that important as long as you use the same scale throughout the calibration. Don't change sizes of the checker box.

jlblancoc, 2020-04-02:
@jiapei100 I fixed the embedded video in [that page](https://www.mrpt.org/tutorials/programming/miscellaneous/kinect-calibration/ ), thanks for reporting!

When the checkerboard is not shown, it should be because of a failure to detect it by the underlying algorithm, so that image won't be used in the optimization.

closed 2020-10-04 by jlblancoc