MRPT/mrpt #1030: cross build error

znstj, 2020-03-20:
Protocol "https" not supported or disabled in libcurl
ExternalProject_Add() use curl download librarise.
How set curl with host computer not cross build env.

jlblancoc, 2020-03-20:
If the problem happens with ASSIMP or OCTOMAP and you don't need them in your app (likely, if you are cross compiling), then:

cmake . -DBUILD_OCTOMAP=OFF -DBUILD_ASSIMP=OFF

If the error happens because eigen3 is missing, install it somehow in your system (apt?).

znstj, 2020-03-23:
Apt can not use with cross build.
I download package and put it on ./build/EP_assimp-prefix , EP_eigen3-prefix and EP_octomap-prefix.
Also edit MD5 value in CMakeLists.
Now I can build mrpt package,but it has another errors.

closed 2020-03-23 by znstj