MRPT/mrpt #1039: cross compile file

hamburgerBear, 2020-04-02:
I am work on: 
Ubuntu 14.04.5 LTS 
MRPT 1.5.7
Cross compile is : aarch64-linux-gnu

i am only compile mrpt 'base' lib, and i make is ok. by when i make install, have a proble, i can't find mrpt include installed my path /usr/local/include。

below follow my install_manifest.txt
/usr/local/share/mrpt/MRPTConfig.cmake
/usr/local/share/mrpt/MRPTConfig-version.cmake
/usr/local/share/mrpt1/package.xml
/usr/local/include/mrpt/mrpt-config/mrpt/config.h
/usr/local/include/mrpt/mrpt-config/mrpt/version.h
/usr/local/include/mrpt/base/include//Eigen/PaStiXSupport
/usr/local/include/mrpt/base/include//Eigen/Geometry
/usr/local/include/mrpt/base/include//Eigen/SPQRSupport
/usr/local/include/mrpt/base/include//Eigen/SparseLU
/usr/local/include/mrpt/base/include//Eigen/Cholesky
/usr/local/include/mrpt/base/include//Eigen/Eigen
/usr/local/include/mrpt/base/include//Eigen/PardisoSupport
/usr/local/include/mrpt/base/include//Eigen/Dense
/usr/local/include/mrpt/base/include//Eigen/OrderingMethods
/usr/local/include/mrpt/base/include//Eigen/CMakeLists.txt
/usr/local/include/mrpt/base/include//Eigen/Sparse
/usr/local/include/mrpt/base/include//Eigen/StdList
/usr/local/include/mrpt/base/include//Eigen/LU
/usr/local/include/mrpt/base/include//Eigen/src/PaStiXSupport/PaStiXSupport.h
/usr/local/include/mrpt/base/include//Eigen/src/Geometry/Umeyama.h
/usr/local/include/mrpt/base/include//Eigen/src/Geometry/Rotation2D.h
/usr/local/include/mrpt/base/include//Eigen/src/Geometry/OrthoMethods.h
/usr/local/include/mrpt/base/include//Eigen/src/Geometry/AngleAxis.h
/usr/local/include/mrpt/base/include//Eigen/src/Geometry/RotationBase.h
/usr/local/include/mrpt/base/include//Eigen/src/Geometry/Scaling.h
/usr/local/include/mrpt/base/include//Eigen/src/Geometry/AlignedBox.h
/usr/local/include/mrpt/base/include//Eigen/src/Geometry/arch/Geometry_SSE.h
/usr/local/include/mrpt/base/include//Eigen/src/Geometry/Transform.h
/usr/local/include/mrpt/base/include//Eigen/src/Geometry/Translation.h
/usr/local/include/mrpt/base/include//Eigen/src/Geometry/EulerAngles.h
/usr/local/include/mrpt/base/include//Eigen/src/Geometry/Hyperplane.h
/usr/local/include/mrpt/base/include//Eigen/src/Geometry/Homogeneous.h
/usr/local/include/mrpt/base/include//Eigen/src/Geometry/ParametrizedLine.h
/usr/local/include/mrpt/base/include//Eigen/src/Geometry/Quaternion.h
/usr/local/include/mrpt/base/include//Eigen/src/SPQRSupport/SuiteSparseQRSupport.h
/usr/local/include/mrpt/base/include//Eigen/src/SparseLU/SparseLU_relax_snode.h
/usr/local/include/mrpt/base/include//Eigen/src/SparseLU/SparseLUImpl.h
/usr/local/include/mrpt/base/include//Eigen/src/SparseLU/SparseLU_Structs.h
/usr/local/include/mrpt/base/include//Eigen/src/SparseLU/SparseLU_pivotL.h
/usr/local/include/mrpt/base/include//Eigen/src/SparseLU/SparseLU_pruneL.h
/usr/local/include/mrpt/base/include//Eigen/src/SparseLU/SparseLU.h
/usr/local/include/mrpt/base/include//Eigen/src/SparseLU/SparseLU_copy_to_ucol.h
/usr/local/include/mrpt/base/include//Eigen/src/SparseLU/SparseLU_SupernodalMatrix.h
/usr/local/include/mrpt/base/include//Eigen/src/SparseLU/SparseLU_column_dfs.h
/usr/local/include/mrpt/base/include//Eigen/src/SparseLU/SparseLU_heap_relax_snode.h
/usr/local/include/mrpt/base/include//Eigen/src/SparseLU/SparseLU_gemm_kernel.h
/usr/local/include/mrpt/base/include//Eigen/src/SparseLU/SparseLU_Utils.h
/usr/local/include/mrpt/base/include//Eigen/src/SparseLU/SparseLU_panel_bmod.h
/usr/local/include/mrpt/base/include//Eigen/src/SparseLU/SparseLU_kernel_bmod.h
/usr/local/include/mrpt/base/include//Eigen/src/SparseLU/SparseLU_panel_dfs.h
/usr/local/include/mrpt/base/include//Eigen/src/SparseLU/SparseLU_Memory.h
/usr/local/include/mrpt/base/include//Eigen/src/SparseLU/SparseLU_column_bmod.h
/usr/local/include/mrpt/base/include//Eigen/src/StlSupport/details.h
/usr/local/include/mrpt/base/include//Eigen/src/StlSupport/StdVector.h
/usr/local/include/mrpt/base/include//Eigen/src/StlSupport/StdList.h
/usr/local/include/mrpt/base/include//Eigen/src/StlSupport/StdDeque.h
/usr/local/include/mrpt/base/include//Eigen/src/Cholesky/LLT.h
/usr/local/include/mrpt/base/include//Eigen/src/Cholesky/LLT_LAPACKE.h
/usr/local/include/mrpt/base/include//Eigen/src/Cholesky/LDLT.h
/usr/local/include/mrpt/base/include//Eigen/src/PardisoSupport/PardisoSupport.h
/usr/local/include/mrpt/base/include//Eigen/src/misc/Kernel.h
/usr/local/include/mrpt/base/include//Eigen/src/misc/lapack.h
[... 1034 lines cut ...]
/usr/local/share/applications/navlog-viewer.desktop
/usr/local/share/pixmaps/mrpt_icon.ico
/usr/local/share/pixmaps/cameracalibgui.png
/usr/local/share/pixmaps/sceneviewer.png
/usr/local/share/pixmaps/gridmapsimul.png
/usr/local/share/pixmaps/robotic-arm-kinematics.png
/usr/local/share/pixmaps/rawlogviewer.png
/usr/local/share/pixmaps/navlog-viewer.png
/usr/local/share/pixmaps/2d-slam-demo.png
/usr/local/share/pixmaps/reactivenav.png
/usr/local/share/metainfo/sceneviewer.appdata.xml
/usr/local/share/metainfo/navlog-viewer.appdata.xml
/usr/local/share/metainfo/reactivenavdemo.appdata.xml
/usr/local/share/metainfo/robotic-arm-kinematics.appdata.xml
/usr/local/share/metainfo/gridmapnavsimul.appdata.xml
/usr/local/share/metainfo/2dslamdemo.appdata.xml
/usr/local/share/metainfo/cameracalib.appdata.xml
/usr/local/share/metainfo/rawlogviewer.appdata.xml
/usr/local/share/metainfo/kinect-stereo-camera-calib-gui.appdata.xml
/usr/local/share/mime/packages/x-mrpt-hmtmap.xml
/usr/local/share/mime/packages/x-mrpt-gridmap.xml
/usr/local/share/mime/packages/x-mrpt-rawlog.xml
/usr/local/share/mime/packages/x-mrpt-3dscene.xml
/usr/local/share/mime/packages/x-mrpt-simplemap.xml
/usr/local/share/mime/packages/x-mrpt-navlog.xml

jlblancoc, 2020-04-02:
Hi, 

Is a more recent version of mrpt an option for you? You will need [gcc-7](https://gist.github.com/heartonbit/2575ce02ca730559d2f2c65eb8682231), I think. 
Cross-compile instructions [here](https://github.com/MRPT/mrpt/blob/develop/doc/CrossCompilingMRPT.md).

Also, I just realized the [mrpt-1.5 PPA](https://launchpad.net/~joseluisblancoc/+archive/ubuntu/mrpt-1.5/+packages) was not building for ubuntu 14.04 because of missing dependencies. I have fixed it. Expect arm64 packages to be built there within a few hours...

jlblancoc, 2020-05-27:
Closing for lack of activity... feel free of reopening, especially if there are problems with the "master" or "develop" branches. The old mrpt-1.5 branch will not be maintained any more, except for critical small patches.

closed 2020-05-27 by jlblancoc