MRPT/mrpt #1040: .simplemap to point cloud and ground truth pose

fahiraMaken, 2020-04-05:
Hi, I want to use dataset at https://ingmec.ual.es/datasets/lidar3d-pf-benchmark/ in my project. The available map is .simplemap. What I understand is it stores both map and the robot poses as well. I want to get the point cloud representation of this map (which later I can convert into octomap) as well as vehicles ground truth pose in the map. 

I have been able to get the CPose3DPDF from which I obtained CPose3d which I believe is the desired vehicle's ground truth pose. Please correct me if I am wrong. Now I have two problems. First the length of trajectory is just 97 which makes me suspicious about my code to obtain it. Second is about the CSensoryFrame which I obtain along with CPose3DPDF. When I get CObservation by doing CSensorFrame->getObservationByIndex and write to a file, it gives me idea that it stores velodyne readings. But I am unable to recover point cloud from it.  Could anyone please guide me to a tool which can convert a .simplemap into a point cloud or an octomap representation and obtain vehicle's pose out of it as well. Many thanks in advance.

jlblancoc, 2020-04-07:
Hi!

Your assumptions were all correct.

I realized the full UAL campus map was not included into the downloads. It's now available to download inside 2018-02-26-ual-campus-map.zip, at the bottom of [this dataset page](https://ingmec.ual.es/datasets/lidar3d-pf-benchmark/). 

You can also regenerate the pointcloud, octomap from the .simplemap using the app [application-observations2map](https://www.mrpt.org/list-of-mrpt-apps/application-observations2map/).
Example .ini files can be found under `MRPT/share/mrpt/config_files/*`

You can also visually inspect .simplemap files with the [robot-map-gui app](https://www.youtube.com/watch?v=YgGcNvrxZI0).

closed 2020-04-07 by jlblancoc

fafz3958, 2020-04-14:
Thank you so much for your help. It works.
 However this tool does not give me ground truth pose in .txt format. Using config file "icp-slam_demo_with_octomap.ini", I got three files, first octomap.bt, second pointCloud.txt and third .3Dscene. I am unable to inspect the third file because somehow the required app is broken. Is there anyway that I can get ground truth robot pose in txt as well. I can extract it using "mrpt::poses::CPose3DPDF" API function, but I want to obtain through some mrpt standard tool along with the map. 
Thanks